intel/media-driver #1130: iHD driver 20.4.3 onwards fails on Gemini Lake

caiusseverus, 2021-01-22:
I have a J4105 system running ubuntu 20.04. I have the iHD driver version 20.4.1+i435\~u20.04 installed from https://repositories.intel.com/graphics/ubuntu focal/main

This works ok, with vainfo giving output:
```
error: can't connect to X server!
libva info: VA-API version 1.9.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_9
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.9 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.4.1 (19107c1)
vainfo: Supported profile and entrypoints
```
etc

If I update to a later version (currently 20.4.3+i449\~u20.04 or 20.4.5+i455\~u20.04 are available), then vainfo gives an error:
```
error: can't connect to X server!
libva info: VA-API version 1.9.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva error: dlopen of /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so failed: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so: undefined symbol: _ZN6GmmLib16GmmClientContext25CreateCustomResInfoObjectEP32GMM_RESCREATE_CUSTOM_PARAMS__REC
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
iHD driver is set in environment along with the correct path.
ffmpeg debug output shows that it is trying to use the iHD driver falling back to the i965 driver:

```
Applying option vaapi_device (set VAAPI hardware device (DRM path or X11 display name)) with argument /dev/dri/renderD128.
[AVHWDeviceContext @ 0x55be2db56780] libva: VA-API version 1.9.0
[AVHWDeviceContext @ 0x55be2db56780] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55be2db56780] libva: dlopen of /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so failed: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so: undefined symbol: _ZN6GmmLib16GmmClientContext25CreateCustomResInfoObjectEP32GMM_RESCREATE_CUSTOM_PARAMS__REC
[AVHWDeviceContext @ 0x55be2db56780] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55be2db56780] libva: Trying to open /usr/local/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55be2db56780] libva: va_openDriver() returns -1
[AVHWDeviceContext @ 0x55be2db56780] libva: Trying to open /usr/lib/jellyfin-ffmpeg/lib/dri/i965_drv_video.so
[AVHWDeviceContext @ 0x55be2db56780] libva: Found init function __vaDriverInit_1_9
[AVHWDeviceContext @ 0x55be2db56780] libva: va_openDriver() returns 0
.
.
.
[AVHWDeviceContext @ 0x55be2db56780] VAAPI driver: Intel i965 driver for Intel(R) Gemini Lake - 2.4.1.
[AVHWDeviceContext @ 0x55be2db56780] Driver not found in known nonstandard list, using standard behaviour.
```

dvrogozh, 2021-01-23:
```
libva error: dlopen of /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so failed: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so: undefined symbol: _ZN6GmmLib16GmmClientContext25CreateCustomResInfoObjectEP32GMM_RESCREATE_CUSTOM_PARAMS__REC
```

This symbol comes from `libigdgmm11` from which media driver depends. It seems that you somehow updated media driver, but gmmlib was not updated. I am currently running docker ubuntu:20.04 image with configured https://repositories.intel.com/graphics/ubuntu and all works fine for me, but the difference is that I fetched packages anew while you performed an update. Can you, please, try to check version of `libigdgmm11` and update it? something like:
```
apt-get install libigdgmm11
```

Below you can find output from my system for your reference (see commands to check package version on your side). 
```
$ vainfo 2>/dev/null
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.4.5 (74e2f11)
vainfo: Supported profile and entrypoints

$ apt-cache policy intel-media-va-driver-non-free
intel-media-va-driver-non-free:
  Installed: 20.4.5+i455~u20.04
  Candidate: 20.4.5+i455~u20.04
  Version table:
 *** 20.4.5+i455~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
     20.4.3+i449~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.4.1+i435~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.0+i423~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.0+i419~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.0+i414~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.0+i405~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.0+i391~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.0+i374~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.0+i371~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.0+i362~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.0+i10~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.0+i5~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.1.1+ds1-1build1 500
        500 http://archive.ubuntu.com/ubuntu focal/multiverse amd64 Packages

$ apt-cache policy libigdgmm11
libigdgmm11:
  Installed: 20.4.1+i455~u20.04
  Candidate: 20.4.1+i455~u20.04
  Version table:
 *** 20.4.1+i455~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
     20.3.3+i449~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.2+i435~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.2+i423~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.2+i419~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.2+i414~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.2+i405~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.3.2+i391~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.5+i374~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.4+i371~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.3+i362~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.2+i10~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.2.2+i5~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.1.1+ds1-1 500
        500 http://archive.ubuntu.com/ubuntu focal/universe amd64 Packages

```

caiusseverus, 2021-01-23:
Interesting - The latest version of libigdgmm11 I have available is 20.4.1 according to apt:

The driver is manually set to the last working one at the moment.
```
$ apt-cache policy intel-media-va-driver-non-free

intel-media-va-driver-non-free:
  Installed: 20.4.1+i435~u20.04
  Candidate: 20.4.5+i455~u20.04
  Version table:
     20.4.5+i455~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
     20.4.3+i449~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
 *** 20.4.1+i435~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
.
.
. etc
```
and 

```
$ apt-cache policy libigdgmm11

libigdgmm11:
  Installed: 20.4.1+i455~u20.04
  Candidate: 20.4.1+i455~u20.04
  Version table:
 *** 20.4.1+i455~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status
     20.3.3+i449~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
.
.
etc
```

So that explains why the driver isn't working, but I don't know why I don't see the more up to date libigdgmm11 package in the list.

The repo I have in apt sources is `deb [arch=amd64] https://repositories.intel.com/graphics/ubuntu focal main`

dvrogozh, 2021-01-23:
> So that explains why the driver isn't working

No, I afraid it does not. I have the same libigdgmm11 as you installed (`20.4.1+i455~u20.04`), but newer driver (`20.4.5+i455~u20.04`) and I don't see the same issue as you do. Don't be tricked by different version for gmmlib and driver - they have different enumeration. 20.4.1 is actually the latest available gmmlib (see https://github.com/intel/gmmlib/tags). And here is the symbol on my side:
```
$ nm --extern-only --defined-only --dynamic /usr/lib/x86_64-linux-gnu/libigdgmm.so.11 | grep _ZN6GmmLib16GmmClientContext25CreateCustomResInfoObjectEP32GMM_RESCREATE_CUSTOM_PARAMS__REC
000000000006e2b0 T _ZN6GmmLib16GmmClientContext25CreateCustomResInfoObjectEP32GMM_RESCREATE_CUSTOM_PARAMS__REC
```

I wonder, did you ever rebuilt libigdgmm11 from sources? if not, then did you attempt to alter it's installation in any other way? For example, maybe you tried to install some other driver, like opencl driver who could bring some other version of the libigdgmm11?

Can you, please, try to verify libigdgmm.so.11 symbolic link and compare checksum. Here it is on my side:
```
user@b4bcf18de97d:~$ ls -al /usr/lib/x86_64-linux-gnu/libigdgmm.so.11
lrwxrwxrwx. 1 root root 21 Jan  5 18:44 /usr/lib/x86_64-linux-gnu/libigdgmm.so.11 -> libigdgmm.so.11.2.905

user@b4bcf18de97d:~$ sha1sum /usr/lib/x86_64-linux-gnu/libigdgmm.so.11.2.905
5355ab53cbf13421531cdb5dbb63f0a5d4261570  /usr/lib/x86_64-linux-gnu/libigdgmm.so.11.2.905
```

caiusseverus, 2021-01-23:
> No, I afraid it does not. I have the same libigdgmm11 as you installed (`20.4.1+i455~u20.04`), but newer driver (`20.4.5+i455~u20.04`) 

OK noted. 

I’ve not compiled libigdgmm1 from source previously. I do have opencl installed, but the symbolic link is identical to your example down to the timestamp and the sha1sum is also identical.

dvrogozh, 2021-01-28:
Sorry for delay with response. I got distracted.

Basically, you have correct driver+gmmlib installed, i.e. driver of `20.4.5+i455~u20.04` + gmm of `20.4.1+i455~u20.04` is valid combination and should work as it does on my side. If you see that gmm library library is there and checksum is the same, I would suggest to verify whether it was actually tried by dynamic loader? So, the only other idea I have is that you might somehow have other gmmlib in your path which gets picked with higher priority and it is of older version w/o symbol you mentioned above.

Can you, please, collect strace and attach it and/or analyze on your side? Something like:
```
strace /your/app 2>strace.log
```

We need to understand which gmm library got actually loaded. Check that it's expected one (the one you've verified w/ checksum).

caiusseverus, 2021-01-28:
> So, the only other idea I have is that you might somehow have other gmmlib in your path which gets picked with higher priority and it is of older version w/o symbol you mentioned above.

This was the problem. It looks like there was another version of gmmlib being tried. From strace:

``openat(AT_FDCWD, "/usr/local/lib/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 4``

I removed that version and vainfo now picks up the correct version:

``openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libigdgmm.so.11", O_RDONLY|O_CLOEXEC) = 4``

which is the version that matches your sha1sum above.

That did not completely fix the issue as it seems vainfo was still trying to load some other old libraries in /usr/local, so I used ldd to identify which ones and remove them.

Everything is now working as it should be:

```
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.4.5 (74e2f11)
vainfo: Supported profile and entrypoints
```

Thanks very much for your help, I appreciate you taking the time. I'm not overly familiar with the way linux does drivers, so it would have taken me a long to work out what was going on otherwise.

closed 2021-01-28 by caiusseverus

dvrogozh, 2021-01-28:
You are welcome. Do you know what you have installed which landed artifacts into /usr/local? Which other libs/drivers you see here, maybe this will give a clue. But taking packages from https://repositories.intel.com/graphics/ubuntu you should not have anything landed into /usr/local (unless there is some bug in our packaging I am not aware of). So, my assumption would be you installed something from other location. What was that? some opencl PPA? or openvino? Just throwing something which sparks in my mind...

caiusseverus, 2021-01-28:
I'm not sure what would have installed to /usr/local. This particular machine is a bit a jack of all trades and has had quite a lot of things added and removed over time, mostly to do with media serving. I don't think it would have been from the intel ubuntu repo - I added that relatively recently. It's possible that there was a version of this driver installed outside of the repo at some point which could have done it. The extra libraries I removed were old versions of libva-x11 libva-drm libva-wayland and libva.so.2. 

I do have opencl installed from https://github.com/intel/compute-runtime/releases but I don't think that puts anything in /usr/local either.

Anyway I don't think there should be any further problem with it, and now I know where to look if something crops up again.