intel/media-driver #1131: parole: /usr/lib/dri/iHD_drv_video.so init failed - Unable to play videos

MartinX3, 2021-01-27:
Intel Skylake 6700HQ with Nvidia GTX 940M
Arch Linux / EndeavourOS
intel-media-sdk 20.5.1
intel-media-driver 20.4.5
libva 2.10.0
```
$ sudo cat /sys/kernel/debug/dri/128/name
i915 dev=0000:00:02.0 master=pci:0000:00:02.0 unique=0000:00:02.0
$ sudo cat /sys/kernel/debug/dri/129/name
nvidia-drm dev=0000:02:00.0 unique=0000:02:00.0
```

```
$ GST_DEBUG=2 parole '/run/media/martin/extLinux/simplescreenrecorder-2020-11-24_18.17.54.mp4' 
0:00:00.021679755 61762 0x55d4d8d2a290 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:467:gst_element_factory_make: no such element factory "autoimagesink"!
0:00:00.309384106 61762 0x55d4d8d2a290 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.309617733 61762 0x55d4d8d2a290 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.339973354 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux_types.c:244:qtdemux_type_get: unknown QuickTime node type pasp
0:00:00.340034060 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:3066:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 1
0:00:00.340923386 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:3066:qtdemux_parse_trex:<qtdemux0> failed to find fragment defaults for stream 2
DRM_IOCTL_I915_GEM_APERTURE failed: Invalid argument
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
0:00:00.354372669 61762 0x55d4d95565e0 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.506574788 61762 0x55d4d959f300 ERROR          vaapipostproc gstvaapipostproc.c:1087:gst_vaapipostproc_process_vpp:<vaapipostproc0> failed to apply VPP filters (error 2)
0:00:00.520345335 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:6545:gst_qtdemux_loop:<qtdemux0> error: Internal data stream error.
0:00:00.520363716 61762 0x55d4d95565e0 WARN                 qtdemux qtdemux.c:6545:gst_qtdemux_loop:<qtdemux0> error: streaming stopped, reason error (-5)
0:00:02.342438216 38521 0x562180796290 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:02.342751001 38521 0x562180796290 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:02.363220947 38521 0x5621810174c0 WARN                 qtdemux qtdemux_types.c:244:qtdemux_type_get: unknown QuickTime node type sgpd
0:00:02.363246269 38521 0x5621810174c0 WARN                 qtdemux qtdemux_types.c:244:qtdemux_type_get: unknown QuickTime node type sbgp
0:00:02.363296433 38521 0x5621810174c0 WARN                 qtdemux qtdemux.c:3066:qtdemux_parse_trex:<qtdemux1> failed to find fragment defaults for stream 1
0:00:02.365385119 38521 0x5621810174c0 WARN                 qtdemux qtdemux.c:3066:qtdemux_parse_trex:<qtdemux1> failed to find fragment defaults for stream 2
0:00:02.443886367 38521 0x5621810251e0 ERROR          vaapipostproc gstvaapipostproc.c:1087:gst_vaapipostproc_process_vpp:<vaapipostproc1> failed to apply VPP filters (error 2)
0:00:02.455318479 38521 0x5621810174c0 WARN                 qtdemux qtdemux.c:6545:gst_qtdemux_loop:<qtdemux1> error: Internal data stream error.
0:00:02.455337324 38521 0x5621810174c0 WARN                 qtdemux qtdemux.c:6545:gst_qtdemux_loop:<qtdemux1> error: streaming stopped, reason error (-5)
```

```
Complete name                            : /run/media/martin/extLinux/simplescreenrecorder-2020-11-24_18.17.54.mp4
Format                                   : MPEG-4
Format profile                           : Base Media
Codec ID                                 : isom (isom/iso2/avc1/mp41)
File size                                : 1.16 GiB
Duration                                 : 2 h 21 min
Overall bit rate mode                    : Variable
Overall bit rate                         : 1 176 kb/s
Writing application                      : Lavf58.45.100

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Baseline@L4
Format settings                          : 1 Ref Frames
Format settings, CABAC                   : No
Format settings, Reference frames        : 1 frame
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 2 h 21 min
Bit rate                                 : 1 046 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Variable
Frame rate                               : 30.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.017
Stream size                              : 1.03 GiB (89%)
Writing library                          : x264 core 160 r3011 cde9a93
Encoding settings                        : cabac=0 / ref=1 / deblock=0:0:0 / analyse=0:0 / me=dia / subme=0 / psy=1 / psy_rd=1,00:0,00 / mixed_ref=0 / me_range=16 / chroma_me=1 / trellis=0 / 8x8dct=0 / cqm=0 / deadzone=21,11 / fast_pskip=1 / chroma_qp_offset=0 / threads=8 / lookahead_threads=1 / sliced_threads=0 / nr=0 / decimate=1 / interlaced=0 / bluray_compat=0 / constrained_intra=0 / bframes=0 / weightp=0 / keyint=250 / keyint_min=25 / scenecut=0 / intra_refresh=0 / rc=crf / mbtree=0 / crf=23,0 / qcomp=0,60 / qpmin=0 / qpmax=69 / qpstep=4 / ip_ratio=1,40 / aq=0
Color range                              : Limited
Color primaries                          : BT.709
Transfer characteristics                 : BT.709
Matrix coefficients                      : BT.709
Codec configuration box                  : avcC

Audio
ID                                       : 2
Format                                   : Ogg
Codec ID                                 : mp4a-DD
Duration                                 : 2 h 21 min
Bit rate mode                            : Variable
Bit rate                                 : 121 kb/s
Maximum bit rate                         : 128 kb/s
Stream size                              : 122 MiB (10%)
Default                                  : Yes
Alternate group                          : 1
```

XinfengZhang, 2021-01-29:
hi @MartinX3 the log is enough, it just show a VPP filter failure, could you help to enable LIBVA_LOG?

MartinX3, 2021-01-29:
Hi and thank you very much for your response @XinfengZhang  

I tried `GST_DEBUG=2 LIBVA_LOG=1 parole ` but the log is the same.
Do I use it wrong?

XinfengZhang, 2021-03-15:
you should add LIBVA_TRACE=./filename. then it will a libva trace generated. and from this log:
"get chip id failed: -1 [22]", 
it should failed at https://github.com/intel/media-driver/blob/master/media_driver/linux/common/os/i915/mos_bufmgr.c#L3479
so, driver could not get the information related with this device ID
I suppose you should already meet some problem to open /dri/renderD128
does vainfo work?

MartinX3, 2021-03-15:
@XinfengZhang Thank you for your response
It generated 4 log files on a single run.

[va.log.zip](https://github.com/intel/media-driver/files/6140169/va.log.zip)

```
$ vainfo 
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.4.5 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
```

XinfengZhang, 2021-03-26:
from the log from https://github.com/intel/media-driver/files/6140169/va.log.zip ， it is a vpp issue. 
[9824.030478][ctx       none]=========vaRenderPicture ret = VA_STATUS_SUCCESS, success (no error) 
[9824.030479][ctx 0x40000000]==========va_TraceEndPicture
[9824.030480][ctx 0x40000000]	context = 0x40000000
[9824.030480][ctx 0x40000000]	render_targets = 0x00000008
[9824.030502][ctx       none]=========vaEndPicture ret = VA_STATUS_ERROR_OPERATION_FAILED, operation failed 
[9824.030504][ctx       none]=========vaDestroyBuffer ret = VA_STATUS_SUCCESS, success (no error) 

change the assignee  and the issue label

TheGamer1002, 2021-04-20:
Any update? I've also been having issues with this error.

XinfengZhang, 2021-04-21:
hi @TheGamer1002 you also have another NV card on your machine, also have two issues, 1. get deviceID faild, get chip id failed: -1 [22] (suppose it is because the device is not opened correct) 2. vaEncPicutre failed on the context with id 0x400000xx? the highest bit could be used to check the context type: https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva_common.h#L97

LhGu, 2022-04-22:
I have found the root cause of this issue in vpp. If you use parole app in the  Video Output Automatic mode, it will create the surface with **1x1** size : vaCreateSurfaces (dpy=0x7fffd406d7f0, format=1, width=1, height=1, surfaces=0x7fffbf96a1dc, num_surfaces=1, attrib_list=0x7fffbf96a1e0, num_attribs=1), and the size of image is also 1x1.

The format of the image is NV12, and its align unit is 2.  If our image size is 1*1, we will  floored the source rectangle to 0*0 to get rid of invalid data.  We will check this size in  vaEndPicture function, and we will return error because of 0,0,0,0 rectangle size.

LhGu, 2022-04-22:
In this case, you need to change the Video Output mode to X Window System(X11/Xshm/Xv).  When I playback 1080P  avc 
 mp4, it won't create 1*1 surface, and everything works well for me. By reading the guideline book in parole app, we can know  that X Window System(X11/Xshm/Xv) mode makes use of hardware video acceleration, and the other mode No Xv  uses (the usually slower) software acceleration. 
We can choose  X Window System(X11/Xshm/Xv) mode directly for hardware video acceleration instead of Automatic mode.

LhGu, 2022-05-16:
This issue needs feedback, but no feedback from submitter yet. I will close this issue if no more information provided from submitter in 1 month. Please feel free to re-open this issue if it still exist.

MartinX3, 2022-05-16:
What feedback do you need?

LhGu, 2022-05-16:
> What feedback do you need?

Use automatic mode to playback video, Maybe everything will be ok

MartinX3, 2022-05-17:
I'm just a regular user opening a video with a player who knows how to gather logs.
I sadly don't know how I can achieve what you want.

LhGu, 2022-05-24:
Hi, @MartinX3 . I root-caused that it is an app issue. No any update info and could not reproduce it now, maybe we can close this issue.

MartinX3, 2022-05-24:
Currently it works and I can't or don't know how to reproduce it.
I also don't use windows

LhGu, 2022-05-25:
Could you close this issue now?  Pls Feel free to re-open this issue if you can reproduce it and want to discuss any more things.

MartinX3, 2022-05-25:
Okay :)

MartinX3, 2022-06-09:
Damn, the issue is back with self made screen record libvorbis h265 videos via the software OBS Studio.

```
0:00:00.026902760 213748 0x55dd1189fc90 WARN     GST_ELEMENT_FACTORY gstelementfactory.c:754:gst_element_factory_make_valist: no such element factory "autoimagesink"!
0:00:00.235098653 213748 0x55dd1189fc90 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
0:00:00.235353274 213748 0x55dd1189fc90 WARN                 basesrc gstbasesrc.c:3688:gst_base_src_start_complete:<source> pad not activated yet
DRM_IOCTL_I915_GEM_APERTURE failed: Das Argument ist ungültig
Assuming 131072kB available aperture size.
May lead to reduced performance or incorrect rendering.
get chip id failed: -1 [22]
param: 4, val: 0
0:00:00.286379366 213748 0x55dd12143cc0 WARN                   vaapi gstvaapiutils.c:78:gst_vaapi_warning: /usr/lib/dri/iHD_drv_video.so init failed
0:00:00.424049093 213748 0x55dd12143cc0 WARN      matroskareadcommon matroska-read-common.c:713:gst_matroska_read_common_parse_skip:<matroskademux0:sink> Unknown CueTrackPositions subelement 0xf0 - ignoring
0:00:00.432235410 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.535: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.432329526 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.432469211 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.535: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.432500377 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.432508265 213748 0x55dd12149aa0 WARN                GST_PADS gstpad.c:4351:gst_pad_peer_query:<capsfilter1:src> could not send sticky events
0:00:00.433044733 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.536: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.433099672 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.433242166 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.536: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.433275200 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.433405784 213748 0x55dd12149aa0 WARN           vaapipostproc gstvaapipostprocutil.c:750:_get_preferred_caps:<vaapipostproc0> Could not fixate src caps

(parole:213748): GStreamer-Base-CRITICAL **: 15:30:38.536: basetransform: second attempt to fixate caps returned invalid (NULL) caps on pad vaapipostproc0:sink
0:00:00.433434913 213748 0x55dd12149aa0 WARN           basetransform gstbasetransform.c:1370:gst_base_transform_setcaps:<vaapipostproc0> transform could not transform video/x-raw(memory:VASurface), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string) [... cut ...] ight-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, colorimetry=(string)bt709, framerate=(fraction)0/1 in anything we support
0:00:00.435905909 213748 0x55dd12143cc0 WARN           matroskademux matroska-demux.c:6067:gst_matroska_demux_loop:<matroskademux0> error: Internal data stream error.
0:00:00.435925252 213748 0x55dd12143cc0 WARN           matroskademux matroska-demux.c:6067:gst_matroska_demux_loop:<matroskademux0> error: streaming stopped, reason not-negotiated (-4)
```

MartinX3, 2022-06-09:
> HI, It seems not the same cause related to invalid (NULL) caps . Maybe you could raise another PR

PR === Pull Request?
Sadly I'm not a C(++) developer.

LhGu, 2022-06-09:
yes, It means Pull Request.

MartinX3, 2022-06-09:
I sadly have no clue how to develop in this project and in this programming language.

LhGu, 2022-06-09:
You could raise it in other PR becuase of not the same issue,  and we will solve it

MartinX3, 2022-06-09:
Do you mean issue ticket instead of pull request?

MartinX3, 2022-06-09:
Hihihi :D
Done https://github.com/intel/media-driver/issues/1435

closed 2022-06-09 by MartinX3