intel/media-driver #1132: vaPutSurface crashes with default driver from Ubuntu20.04

dmitryermilov, 2021-01-29:
On Gen9 calls to vaPutSurface cause crashes. 
Repro steps from freshly installed Ubuntu20.04
install packages:
```
apt install libva2
apt install intel-media-va-driver-dbgsym
apt install mvp
```
Run mpv:
```
mvp -vo vaapi /home/user/Downloads/Big_Buck_Bunny_1080_10s_1MB.mp4
The legacy option syntax ('-vo value') is deprecated and dangerous.
Please use '--vo=value'.
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
File tags:
 Artist: Blender Foundation 2008, Janus Bager Kristensen 2013
 Comment: Creative Commons Attribution 3.0 - http://bbb3d.renderfarming.net
 Composer: Sacha Goedegebure
 Genre: Animation
 Title: Big Buck Bunny, Sunflower version
[vo/vaapi] OSD format not supported. Disabling OSD.
[vo/vaapi] Warning: this compatibility VO is low quality and may have issues with OSD, scaling, screenshots and more.
[vo/vaapi] vo=gpu is the preferred choice in any case and includes VA-API support via hwdec=vaapi or vaapi-copy.
Using hardware decoding (vaapi).
VO: [vaapi] 1920x1080 vaapi[nv12]
Segmentation fault (core dumped)
```
The same run under gdb:
```
hread 10 "mpv/vo" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe4958700 (LWP 85980)]
KernelDll_AllocateStates (pKernelBin=pKernelBin@entry=0x7fffc8093110, uKernelSize=0, pFcPatchCache=pFcPatchCache@entry=0x0, 
    uFcPatchCacheSize=<optimized out>, pDefaultRules=0x0, ModifyFunctionPointers=0x0) at ./media_driver/agnostic/common/vp/kdll/hal_kerneldll.c:3349
3349	./media_driver/agnostic/common/vp/kdll/hal_kerneldll.c: No such file or directory.
(gdb) bt
#0  KernelDll_AllocateStates (pKernelBin=pKernelBin@entry=0x7fffc8093110, uKernelSize=0, pFcPatchCache=pFcPatchCache@entry=0x0, 
    uFcPatchCacheSize=<optimized out>, pDefaultRules=0x0, ModifyFunctionPointers=0x0) at ./media_driver/agnostic/common/vp/kdll/hal_kerneldll.c:3349
#1  0x00007fffcf2676d4 in VphalRenderer::Initialize (this=0x7fffc80a33c0, pSettings=0x7fffe4956fc0)
    at ./media_driver/agnostic/common/vp/hal/vphal_renderer.cpp:1385
#2  0x00007fffcf249cea in VphalState::Allocate (pVpHalSettings=0x7fffe4956fc0, this=0x7fffc808ec20)
    at ./media_driver/agnostic/common/vp/hal/vphal.cpp:146
#3  VphalState::Allocate (this=0x7fffc808ec20, pVpHalSettings=0x7fffe4956fc0) at ./media_driver/agnostic/common/vp/hal/vphal.cpp:75
#4  0x00007fffcf46ed83 in DdiVp_InitVpHal (pVpCtx=pVpCtx@entry=0x7fffc8076010) at ./media_driver/linux/common/vp/ddi/media_libva_vp.c:1769
#5  0x00007fffcf472727 in DdiVp_InitCtx (pVaDrvCtx=<optimized out>, pVpCtx=0x7fffc8076010) at ./media_driver/linux/common/vp/ddi/media_libva_vp.c:1629
#6  DdiVp_InitCtx (pVaDrvCtx=<optimized out>, pVpCtx=0x7fffc8076010) at ./media_driver/linux/common/vp/ddi/media_libva_vp.c:1580
#7  0x00007fffcf4729aa in DdiVp_CreateContext (pVaDrvCtx=pVaDrvCtx@entry=0x7fffc8023000, vaConfigID=vaConfigID@entry=0, iWidth=iWidth@entry=0, 
    iHeight=iHeight@entry=0, iFlag=iFlag@entry=0, vaSurfIDs=vaSurfIDs@entry=0x0, iNumSurfs=0, pVaCtxID=0x7fffe49570ec)
    at ./media_driver/linux/common/vp/ddi/media_libva_vp.c:3096
#8  0x00007fffcf43ad5b in DdiMedia_PutSurface (ctx=<optimized out>, surface=18, draw=0x5400002, srcx=<optimized out>, srcy=<optimized out>, 
    srcw=<optimized out>, srch=1080, destx=0, desty=20, destw=1848, desth=1039, cliprects=0x0, number_cliprects=0, flags=32)
    at ./media_driver/linux/common/ddi/media_libva.cpp:4030
#9  0x000055555567e0c7 in ?? ()
#10 0x0000555555676697 in ?? ()
#11 0x00007ffff736f609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#12 0x00007ffff4f9f293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) q
```

dmitryermilov, 2021-01-29:
vaPutSurface internally depends on color space conversion to RGB. And this CSC is enabled via shaders. CSC is missing in intel-media-va-driver. Due to a bug in code (missing nullptr check in code related to CSC) vaPutSurface crashes.
“apt install intel-media-va-driver-non-free” can workaround the crash.
The issue is seen in 20.1.1 media-driver release which landed in Ubuntu20.04, and it’s still there is latest code.

XinfengZhang, 2021-01-31:
"Due to a bug in code (missing nullptr check in code related to CSC) vaPutSurface crashes" @FurongZhang ,have we fixed it? it is about free driver which maybe disable some kernel path. 
@dmitryermilov suppose , it is a gen9 issue, because we open source VP shaders from gen11, right?

FurongZhang, 2021-02-01:
@XinfengZhang , let me check that. Will update ASAP.

sebastinas, 2021-03-02:
Any news here? We've got a bug report in Debian with the same [issue](https://bugs.debian.org/983860) where the driver is built with `-DENABLE_KERNELS=ON -DENABLE_NONFREE_KERNELS=OFF`.

FurongZhang, 2021-04-16:
https://github.com/intel/media-driver/pull/1182 fixed the crash. 

BTW, Intel-media-va-driver-non-free build contains full feature. Intel-media-va-driver doesn’t support VP shaders in Gen9 platform.

dmitryermilov, 2021-04-16:
> @dmitryermilov suppose , it is a gen9 issue, because we open source VP shaders from gen11, right?

Yes, Carl.

FurongZhang, 2021-06-14:
I have fixed the crash. Will update the PR and further information.

FurongZhang, 2021-06-26:
https://github.com/intel/media-driver/commit/66fa14da59ba6a81f1d604d1c71fa93502884599

FurongZhang, 2021-06-26:
Fixed the crash issue.

FurongZhang, 2021-06-26:
Please feel free to let me know if any opens and questions.

closed 2021-06-26 by FurongZhang