intel/media-driver #1133: Possible memory leak in media_libva.cpp

wizkid057, 2021-01-31:
https://github.com/intel/media-driver/blob/085cb68483d26edc06b389ba6569bd5b1f649198/media_driver/linux/common/ddi/media_libva.cpp#L5249

This memory allocation does not appear to be free'd properly and causes an ongoing memory leak of approximately 208 bytes per frame when using VAAPI decoding.

I've done my best to rule out this being an application or issue with another library, and everything else appears to behave correctly all the way through libva. Everything seems to be getting called appropriately, but this library never releases the ~208 bytes allocated after vaUnmapBuffer and vaDestroyImage are called as expected after copying a decoded frame to system memory.

Some valgrind output (using libav)

```
==30145== 471,536 bytes in 2,267 blocks are definitely lost in loss record 1,271 of 1,296
==30145==    at 0x483AB65: calloc (vg_replace_malloc.c:760)
==30145==    by 0x13831F43: MosUtilities::MosAllocAndZeroMemory(unsigned long) (mos_utilities_next.cpp:391)
==30145==    by 0x137B4888: DdiMedia_GetImage(VADriverContext*, unsigned int, int, int, unsigned int, unsigned int, unsigned int) (media_libva.cpp:5244)
==30145==    by 0x49BE057: vaGetImage (va.c:1842)
==30145==    by 0x82AD44: vaapi_map_frame (hwcontext_vaapi.c:762)
==30145==    by 0x82B019: vaapi_transfer_data_from (hwcontext_vaapi.c:830)
==30145==    by 0x829E4E: av_hwframe_transfer_data (hwcontext.c:434)
```

Was using a HEAD of e280504bb3ccc429aa43f30aa76e0c990df6ccbf when doing this testing.

I've confirmed the libav call to av_hwframe_transfer_data calls vaUnmapBuffer and vaDestroyImage on the surface as expected, and libva passes this to the intel media driver as expected.

Any help appreciated.  Unfortunately I don't have a ton of time to dedicate to debugging this further.

wizkid057, 2021-01-31:
As a local hack, this fixes this issue for me.  I'm actually unsure what's going on, as the check in this code only appears to attempt to free the pSurfDesc memory if it is an external surface? Seems backwards... but I'm not super familiar with the code.

```
diff --git a/media_driver/linux/common/ddi/media_libva_util.cpp b/media_driver/linux/common/ddi/media_libva_util.cpp
index 8d963933..88839853 100755
--- a/media_driver/linux/common/ddi/media_libva_util.cpp
+++ b/media_driver/linux/common/ddi/media_libva_util.cpp
@@ -1406,6 +1406,10 @@ void DdiMediaUtil_FreeSurface(DDI_MEDIA_SURFACE *surface)
             DdiMediaUtil_UnlockSurface(surface);
             DDI_VERBOSEMESSAGE("DDI: try to free a locked surface.");
         }
+       if (surface->pSurfDesc) {
+               MOS_FreeMemory(surface->pSurfDesc);
+               surface->pSurfDesc = nullptr;
+       }
         mos_bo_unreference(surface->bo);
         surface->bo = nullptr;
     }
```

Again, on HEAD of e280504bb3ccc429aa43f30aa76e0c990df6ccbf

I confirmed that DdiMediaUtil_IsExternalSurface(surface) returns false.  Since pSurfDesc is allocated by DdiMedia_GetImage, seems like DdiMediaUtil_FreeSurface should free it, no?

Anyway, this hack works for me.  Hopefully helps someone else avoid this leak.

XinfengZhang, 2021-02-01:
the rootcause is : surfaceDescriptor was created when there are VASurfaceAttribExternalBufferDescriptor. but the DdiMediaUtil_IsExternalSurface just checked memory type. will file a PR to fix it ASAP

XinfengZhang, 2021-02-03:
@wizkid057 does #1136  fix the issue?

wizkid057, 2021-02-03:
> @wizkid057 does #1136 fix the issue?

This appears to have done the trick!  Thanks for the quick work. :)

PR #1136 (2021-02-08): fix memory leak caused by external surface descriptor
Fixes #1133
commit 100dd43d (2021-02-08): fix memory leak caused by external surface descriptor

Fixes #1133
closed 2021-02-08 by intel-mediadev