intel/media-driver #1134: Test Failure: MediaCapsDdiTest.DecodeEncodeProfile

uartie, 2021-02-01:
Since 86c1dc2dcc03833eca74a55cc18efc7e99978e8b, the MediaCapsDdiTest.DecodeEncodeProfile test case fails after compilation.

```
2021-02-01T20:37:07.9663327Z [ RUN      ] MediaCapsDdiTest.DecodeEncodeProfile
2021-02-01T20:37:07.9687804Z /home/runner/work/media-driver/media-driver/media/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:106: Failure
2021-02-01T20:37:07.9689074Z Value of: (CompareFeatureIDTable(queriedFeatureIDTable, refFeatureIDTable))
2021-02-01T20:37:07.9689899Z   Actual: false
2021-02-01T20:37:07.9690275Z Expected: true
2021-02-01T20:37:07.9690900Z Platform = SKL, Failed function = CompareFeatureIDTable
2021-02-01T20:37:07.9691383Z 
2021-02-01T20:37:07.9713080Z /home/runner/work/media-driver/media-driver/media/media_driver/linux/ult/ult_app/ddi_test_caps.cpp:106: Failure
2021-02-01T20:37:07.9714409Z Value of: (CompareFeatureIDTable(queriedFeatureIDTable, refFeatureIDTable))
2021-02-01T20:37:07.9715227Z   Actual: false
2021-02-01T20:37:07.9715622Z Expected: true
2021-02-01T20:37:07.9716229Z Platform = BXT, Failed function = CompareFeatureIDTable
2021-02-01T20:37:07.9716713Z 
2021-02-01T20:37:07.9739318Z [  FAILED  ] MediaCapsDdiTest.DecodeEncodeProfile (8 ms)
2021-02-01T20:37:07.9740633Z [----------] 1 test from MediaCapsDdiTest (8 ms total)
```

cc: @ryanthua

uartie, 2021-02-01:
This breaks "make install"

dvrogozh, 2021-02-01:
Yep. And this was caught by media-driver own CI as well: https://github.com/intel/media-driver/runs/1809680772.

ryanthua, 2021-02-01:
Revert this change in https://github.com/intel-innersource/drivers.gpu.unified/pull/10687

uartie, 2021-02-02:
@ryanthua how long before the revert is sync'd here?

ryanthua, 2021-02-02:
> @ryanthua how long before the revert is sync'd here?

It's currently under internal CI build. I think it still needs around 1 hour to finish. After CI build, the sync will start and will take another few minutes I think.

uartie, 2021-02-02:
Reverted in 2990a19101421ef0973b767d296d448345ac9a20

closed 2021-02-02 by uartie