intel/media-driver #1138: VPP PROAMP not effect with 3840x2160 and above

SiewHoon, 2021-02-05:
This issue not able to reproduce by using video clip in resolution :  2560x1440, 1920x1080, 1280x720, 800x600
It only happen with 3840x2160 resolution and above. It can playback but not effect on PROAMP setting at here on this 4k resolution.

Command:
gst-launch-1.0 filesrc location= /mnt/media/Puppies_3840x2160_30mbps_60fps_Baseline@L5.2.mp4 ! qtdemux ! vaapih264dec ! vaapipostproc brightness=-1 contrast=1 saturation=1 hue=-180 ! glimagesink

gst-launch-1.0 filesrc location= /mnt/media/Puppies_3840x2160_30mbps_60fps_Baseline@L5.2.mp4 ! qtdemux ! vaapih264dec ! vaapipostproc brightness=-1 ! glimagesink

media driver: intel-media-20.4.5
libva: 2.10.0
gstreamer framework master:
gstreamer-vaapi: 5e3fde85
Platform: TGL-U
OS: Ubuntu 18.04

This issue been report at here:
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/302

SiewHoon, 2021-02-05:
This issue able to reproduce in gstreamer-vaapi side after revert this WR patch due to media-driver side already fixed  it:

[a354c14e58f9afc6548668f5d6ed9bcc1a1d35f7] vaapipostproc: Remove YUV to/from RGB color primary quirk since iHD driver has fixed in https://github.com/intel/media-driver/commit/a39fe9bc051a8c3efa8f35122a1585981ec7f816.

I re-test with combination  by using :
media-driver (intel-media-20.3.0) and libva 2.9.0 version environment.

SiewHoon, 2021-02-05:
Another issue report in gst-vaapi side: https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/301 after remove the WR done in gstreamer-vaapi.

On MJPEG 4k Video playback failing if go thru video postprocessing. 

Gst Command:  
gst-launch-1.0 filesrc location=/mnt/media/Elecard1_3840x2160_8mbps_30fps_444_1.avi ! avidemux ! vaapijpegdec ! vaapipostproc ! glimagesink

FurongZhang, 2021-06-07:
We are checking this. Will update it when we root cause it.

FurongZhang, 2021-06-07:
I have checked in the code.

SiewHoon, 2021-06-10:
This issue still can reproduce with latest commit id from media-driver: 
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/302
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/301

MicroYY, 2021-07-02:
Hi @SiewHoon 
Pls try this PR https://github.com/intel/media-driver/pull/1218. It'll be merged after internal tests.

SiewHoon, 2021-07-02:
Hi @MicroYY ,

Ok, sure. I will test this PR on my side.

SiewHoon, 2021-07-05:
Hi @MicroYY  

I applied PR #1218 on top of Q2'21 media driver and running this test on my TGL board.   

https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/301 -> This issue still can reproduce it. 
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/302 -> This issue can not reproduce anymore.

MicroYY, 2021-07-05:
Hi @SiewHoon 
Could you pls share the content (MJPEG 4K 8bit 444 video)? I think the root-causes of the two issues aren't the same, but maybe all related to BT2020.

SiewHoon, 2021-07-06:
Hi @MicroYY 

The file size is too big upload at here. I share MPEG4 4k 8bit 444 video test clip thru internal share drive. Let me if you know able to access it.  Thanks.

closed 2021-07-15 by FurongZhang

MicroYY, 2021-07-15:
Hi @SiewHoon  Could you pls create a new issue to track?

PR #1218 (2021-07-15): [VP] Fix 2pass CSC PROCAMP not work issue
Enable PROCAMP when 2pass CSC is also enabled.
Fixes: #1138
Signed off by: jay.yang@intel.com
commit 06905e60 (2021-07-15): [VP] Fix 2pass CSC PROCAMP not work issue

Enable PROCAMP when 2pass CSC is also enabled.
Fixes: #1138
Signed off by: jay.yang@intel.com
PR #1230 (2021-08-10): [VP] Fix 2pass CSC PROCAMP not work issue
Enable PROCAMP when 2pass CSC is also enabled.
Fixes: #1138
Signed off by: jay.yang@intel.com
commit 86868f1e (2021-08-10): [VP] Fix 2pass CSC PROCAMP not work issue

Enable PROCAMP when 2pass CSC is also enabled.
Fixes: #1138
Signed off by: jay.yang@intel.com