intel/media-driver #1143: Tone Mapping Bug using Tiger Lake CPU

softworkz, 2021-02-11:
I'm experiencing a bug when trying to use VAAPI H2S tone mapping with a Tiger Lake CPU.

I can reproduce that bug with both, ffmpeg and gstreamer.

## ffmpeg

- Latest master version of ffmpeg
- Applied patch: https://github.com/intel-media-ci/cartwheel-ffmpeg/blob/master/patches/0031-avfilter-tonemap_vaapi-pass-filter-parameters-to-VA-.patch
- OS: **Ubuntu 20.04 LTS**

### Command line:

`sudo ./ffmpeg -loglevel verbose -y -copyts -start_at_zero -c:v:0 hevc -hwaccel:v:0 vaapi -hwaccel_device:v:0 /dev/dri/renderD128 -hwaccel_output_format:v:0 vaapi -i "Sony 4K HDR Demo - Camping in Nature.mp4" -filter_complex "[0:0]tonemap_vaapi@f1=p=bt709:m=bt709:t=bt709:format=nv12[f1_out0]" -map [f1_out0] -an -sn -c:v:0 h264_vaapi -b:v:0 119616000 -g:v:0 72 -maxrate:v:0 119616000 -bufsize:v:0 23923200 out.mp4
`

### Output

#### Driver versions

```
[AVHWDeviceContext @ 0x5648a8f3d300] libva: VA-API version 1.11.0
[AVHWDeviceContext @ 0x5648a8f3d300] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x5648a8f3d300] libva: Found init function __vaDriverInit_1_10
[AVHWDeviceContext @ 0x5648a8f3d300] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x5648a8f3d300] Initialised VAAPI connection: version 1.11
[AVHWDeviceContext @ 0x5648a8f3d300] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.0 (13c0e14).
[AVHWDeviceContext @ 0x5648a8f3d300] Driver not found in known nonstandard list, using standard behaviour.
```

#### Color values are properly set

```
[tonemap_vaapi@f1 @ 0x5648aadffd00] Mapped colour properties tv bt2020nc/bt2020/smpte2084 topleft to VA standard 13 chroma siting 0x5 range 0x1.
[tonemap_vaapi@f1 @ 0x5648aadffd00] Mapped colour properties tv bt709/bt709/smpte2084 topleft to VA standard 13 chroma siting 0x5 range 0x1.
```

### Visual Result

[image]

## GStreamer

- Version 1.18
- no patches applied
- OS: **Ubuntu 20.04 LTS**
- For some reason (gsteamer issue in that version), HDR data needs to be specified manually in the command line to activate tone mapping

### Command line:

`sudo gst-launch-1.0 -vf filesrc location="Sony Food Fizzle UHD 4K Demo.mp4" ! qtdemux ! h265parse ! vaapih265dec !  "video/x-raw(memory:VASurface),mastering-display-info=35400:14599:8500:39850:6550:2300:15634:16450:10000000:0" ! vaapipostproc hdr-tone-map=auto ! "video/x-raw(memory:VASurface),format=NV12,colorimetry=bt709,width=1920,height=1080"  ! vaapih264enc ! h264parse ! mp4mux ! filesink location=out.mp4
`

### Output

Can provide logs on request

### Visual Result

[image]

softworkz, 2021-02-11:
Possibly related: https://github.com/intel/media-driver/issues/861

FurongZhang, 2021-02-19:
I am debugging this issue.

softworkz, 2021-02-19:
Let me know when you need help reproducing or testing any potential fix or workaround.

FurongZhang, 2021-04-16:
I have fixed that and already provided the patch to the bug submitter. Will provide the commit ID in this link.

softworkz, 2021-04-16:
> I have fixed that and already provided the patch to the bug submitter.

Yup, that was me (different account). I've seen that you've been frequently pushing changes since then, that's why we haven't rolled out the patch yet. 

Please let me know once it is committed or PRed. 
Did you already submit the dependency changes to libva?

FurongZhang, 2021-04-20:
It is committed.

Libva commit ID: 89f9f7673737ed2c425d6adad016f7cd35cd319a

FurongZhang, 2021-04-20:
Media Driver Commit ID: b7bc4bd99d4b5cac721721e1cbb6bf82e1a37b48

FurongZhang, 2021-04-20:
Basically, please use the latest libva and driver.  

I will close this issue, Please let me know if you have other issues.

closed 2021-04-20 by FurongZhang

softworkz, 2021-04-20:
Thank you very much for taking care of this!

softworkz, 2021-04-20:
BTW: Does this also fix https://github.com/intel/media-driver/issues/861?

softworkz, 2021-04-20:
We haven't had a user report about ICL yet. There was one about Comet Lake recently, but in that case, tonemap_vaapi didn't work at all (rather than just showing wrong colors).

I'm not sure whether it should work on Comet Lake or not, but VAAPI indicated that tone mapping would be supported (via `VAProcFilterHighDynamicRangeToneMapping`), so that might be the same as #861, but I'm not sure.

FurongZhang, 2021-04-26:
I got it, will continue to check #861 as you mentioned. 

CML should be working per the design. #861 should be a different issue. Anyway, will check that.