intel/media-driver #1148: Failed to output to non (0,0) position on render target by VEBOX + SFC scaling from NV12 to YUY2

joojler, 2021-02-23:
With Media SDK sample_vpp application on TGL-U, it failed to output to non (0,0) position on render target by VEBOX + SFC scaling from NV12 to YUY2. The command is as below:

$ ./sample_vpp -sw 1920 -sh 1080 -scc nv12 -dw 3840 -dh 2160 -dcc yuy2 -dcrX 1920 -dcrW 1920 -dcrH 1080 -i /home/james/video/1080p.nv12 -o test_sfc.yuy2 -n 1 -scaling_mode 1

The expected position in above command should be (1920, 0), but the result will be in (0,0).  Looks like the media-driver ignore the dcrX parameter. At the same time, NV12->NV12, NV12->RGB4 with VEBOX + SFC  don't have similar issues. Also if we choose -scaling_mode as 0 which means scaling with EU, there is also no issue. 

It's tested based on Media SDK stack 2020.3 including the corresponding version of media-driver.

XinfengZhang, 2021-02-25:
thanks for reporting, @FurongZhang  could you help to take a look and assign it?

FurongZhang, 2021-03-01:
Sure.

ljia5, 2021-04-26:
The issue should be caused by Vebox only for this NV12 to YUY2 case. The left and top of the rcDst are (0, 0) for Vebox only. The latest build already switch to Vebox+SFC and fixed this issue.

FurongZhang, 2021-04-26:
Thanks @ljia5 for the fix.

FurongZhang, 2021-04-26:
@ljia5 , is it possible for you to post the commit ID?

closed 2021-04-26 by FurongZhang

ljia5, 2021-04-29:
Only checked the output path wrong with the produce version. The latest build use Vebox+SFC(fixed the issue). Maybe need to locate the commit ID.

FurongZhang, 2021-06-16:
In our test, this issue has fixed in this commit: https://github.com/intel/media-driver/commit/205f76cca083c8feeaa20ee7052832de460c3cb6  (Apr.15)

Test cmd: $ ./sample_vpp -sw 1920 -sh 1080 -scc nv12 -dw 3840 -dh 2160 -dcc yuy2 -dcrX 1920 -dcrW 1920 -dcrH 1080 -i /home/james/video/1080p.nv12 -o test_sfc.yuy2 -n 1 -scaling_mode 1