intel/media-driver #1151: [TGL] hevc 10bit 444 decode found GPU hang by using  sample_decode on iHD driver

fanli-hub, 2021-02-26:
Platform: TGL
OS: Ubuntu 20.04.2 LTS
Kernel: 5.8.0-44-generic
MediaSDK: b3fbeabb29574dbf955e68b18a3dc402efa16522
libva: 7a92a2b7cc2ec89b19d50e990ca1a3580bbf297a
media-driver: 81531bd58064f019d513a21025d15d6f967907d1

Defect Description:
GPU Hang occured when using sample_decode H265 10bit 444 on iHD driver. 

Command and error:
sample_decode h265 -i Fireworks_3840x2160_12mbps_30fps_Main_at_L5.0_10bit_444.265 -o test.yuv -vaapi

dmesg:
[ 7902.805594] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[ 7902.805612] i915 0000:00:02.0: [drm] sample_decode[23374] context reset due to GPU hang
[ 7902.806677] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:10:28fffffd, in sample_decode [23374]

XinfengZhang, 2021-03-01:
@fanli-hub , could you help to share the i915_error_state?

fanli-hub, 2021-03-01:
@XinfengZhang 
dmesg |grep i915
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-45-generic root=UUID=81a7a377-c87c-4b03-90df-b1a177eca604 ro i915.force_probe=9a49 i915.enable_guc=2 quiet splash vt.handoff=7
[    0.044512] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.8.0-45-generic root=UUID=81a7a377-c87c-4b03-90df-b1a177eca604 ro i915.force_probe=9a49 i915.enable_guc=2 quiet splash vt.handoff=7
[    2.702657] i915 0000:00:02.0: [drm] VT-d active for gfx access
[    2.702968] i915 0000:00:02.0: vgaarb: deactivate vga console
[    2.748783] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    2.754935] mei_hdcp 0000:00:16.0-b638ab7e-94e2-4ea2-a552-d1c54b627f04: bound 0000:00:02.0 (ops i915_hdcp_component_ops [i915])
[    2.755162] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/tgl_dmc_ver2_06.bin (v2.6)
[    2.922112] i915 0000:00:02.0: [drm] GuC firmware i915/tgl_guc_35.2.0.bin version 35.2 submission:disabled
[    2.922114] i915 0000:00:02.0: [drm] HuC firmware i915/tgl_huc_7.0.12.bin version 7.0 authenticated:yes
[    2.923458] [drm] Initialized i915 1.6.0 20200515 for 0000:00:02.0 on minor 0
[    2.988880] fbcon: i915drmfb (fb0) is primary device
[    2.988969] i915 0000:00:02.0: fb0: i915drmfb frame buffer device
[ 1989.684582] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[ 1989.684596] i915 0000:00:02.0: [drm] sample_decode[2380] context reset due to GPU hang
[ 1989.686270] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:10:28fffffd, in sample_decode [2380]
[ 1998.580704] i915 0000:00:02.0: [drm] Resetting vcs1 for preemption time out
[ 1998.580732] i915 0000:00:02.0: [drm] sample_decode[2381] context reset due to GPU hang
[ 1998.581813] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:10:28fffffd, in sample_decode [2381]

Jexu, 2021-03-26:
it maybe kernel scheduling issue and have reported the issue to i915

Jexu, 2021-11-23:
Some fix in i915 driver, can you try with latest kernel + media driver to check if the issue still exists. Thanks a lot.

fanli-hub, 2022-03-01:
> Some fix in i915 driver, can you try with latest kernel + media driver to check if the issue still exists. Thanks a lot.

I tried to use kernel 5.15 and the latest media driver, this issue still can be reproduced.

Jexu, 2022-03-02:
It is about i915 driver issue for the bond extention submission. Please refer the link https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs to filse the issue for i915 driver.

Jexu, 2022-04-07:
I915 driver has plan to fix the issue by droping the bond extention in the future.

Jexu, 2023-07-12:
Since it is known issue in i915 driver and they have no plan to fix in bond submission. I will mark this issue as Reserved and let me know if you have any concerns.

By the way, this issue could be solved on platforms after ADL-P since i915 driver has parallel submission support that solve it perfectlly(vs bond submission on tgl).