intel/media-driver #1152: Very bad artifacts at specific quality levels with jpegenc

linkmauve, 2021-03-03:
Here is an encoding at quality 90:
[image]
Quality 94:
[image]
Quality 95:
[image]
Quality 100:
[image]

As we can see, at quality 94 and 95, iHD generates terrible blocking artifacts everywhere, and neither luma nor chroma are anywhere near correct. This also happens at lower quality (not pictured here) but is less and less present.

This issue doesn’t happen with the older i965 driver.

I’ve reproduced with every YUV image I’ve tried, using:
- libva 2.10.0
- libva-utils 2.10.0
- intel-media-driver 20.4.5
- (libva-intel-driver 2.4.1 for comparison)

All from ArchLinux’s repositories.

uartie, 2021-03-03:
Seems related to #281

linkmauve, 2021-03-03:
Indeed, but that coarse testing in #281 doesn’t seem to exhibit the periodicity (per quality) of the issue.

Here is my data for this one sample from quality 1 to quality 100: [psnr.csv](https://github.com/intel/media-driver/files/6077966/psnr.csv.txt)

And as a chart:
[image]

XinfengZhang, 2021-03-09:
@aleonenk @Alexandr-Konovalov could you help to check the issue

ivan-losev, 2021-03-12:
Hi @linkmauve,
Could you, please, share what environment (platform, driver version, etc.) you are using and steps to reproduce this quality drop issue.
I tried to reproduce issue on KBL on the latest driver using sample_encode (as in #281 ) but didn't observe quality drop

ivan-losev, 2021-03-17:
Hi @linkmauve,
Could you, please, share steps to reproduce quality drop issue?

ivan-losev, 2021-03-17:
Thanks, I'll take a look, what is wrong.

Could you, also try to reproduce issue using MSDK samples?
sample_encode jpeg -i input.yuv -n 1 -hw -vaapi -w 1280 -h 720 -device /dev/dri/renderD128 -q 96 -o se_1280x720_q96.jpeg &> log.txt; echo Quality 96; ffmpeg -pix_fmt yuv420p -s:v 1280x720 -i input.yuv -i se_1280x720_q96.jpeg      -frames:v 1 -lavfi psnr=stats_file=psnr_logfile.txt -f null - 2>&1 | grep PSNR; echo

linkmauve, 2021-03-17:
I just did, and here are my observations:
- this `sample_encode` tool seems to prefer y4m files to yuv, here is its log output when I give it a raw yuv file:

ivan-losev, 2021-03-25:
Hmm, this is strange,
For sample encode, by default it expected I420/NV12 surface as input (type 0 and 1 in jpegenc)

Related to jpegenc, it's media-driver issue related to different quantization matrix scaling
jpegenc formula:
`(quantLuma.Qk[i] * quality)/100;`

media-driver formula (with round up):
`quantValue = ( ((uint32_t)(quantMatrix->m_quantTable[qMatrixCount].m_qm[i])) * quality + 50) / 100;`

As a result, jpeg header use one qmatrix, while driver - another and quality drop occurs

ivan-losev, 2021-04-08:
Hi @linkmauve,
Sorry for delay, we expect that commit will be merged till the end of the next week (16 Apr 2021)

XinfengZhang, 2021-04-08:
thanks @ivan-losev , please add "fixes" keyword in your commit , or link the issue with your commit , thanks

linkmauve, 2021-04-08:
Thanks! I’ll test that once it gets merged!

ivan-losev, 2021-06-30:
Hi @linkmauve,
Could you, please, try to reproduce quality issue on the latest master?
Fix commit: https://github.com/intel/media-driver/commit/1ec93f2888860ef992595e33c49e7b7d4f94fc86

Xiaogangli-intel, 2022-04-20:
Hi @linkmauve, did you try the fix [1ec93f2](https://github.com/intel/media-driver/commit/1ec93f2888860ef992595e33c49e7b7d4f94fc86)? Can it fix your issue?

Xiaogangli-intel, 2022-04-25:
@linkmauve I will close this issue. If you have other concern, let us know. Thank you.

closed 2022-04-25 by Xiaogangli-intel