intel/media-driver #1153: [ffmpeg qsv] hevc_qsv encoder does not use preset option properly.

caiusseverus, 2021-03-05:
When using ffmpeg to encode to hevc, using the ``-preset `` option should set the TargetUsage parameter in ffmpeg accordingly. This works correctly with the h264_qsv encoder, with ``-preset 5`` selecting TargetUsage 5 for example.

With hevc_qsv however, only three different levels appear to be available as follows: [1, 2] -> TU 1; [3, 4, 5] -> TU 5; and [6, 7] -> TU 7.

For example, the command line:
``ffmpeg -hwaccel qsv -c:v h264_qsv -i test_in.mkv -c:v hevc_qsv -preset 5 -global_quality 22 test_out.mkv -loglevel verbose``

gives this result:

```
[hevc_qsv @ 0x55870b329480] Using input frames context (format qsv) with hevc_qsv encoder.
[hevc_qsv @ 0x55870b329480] Using the intelligent constant quality (ICQ) ratecontrol method
[hevc_qsv @ 0x55870b329480] profile: simple; level: 296
[hevc_qsv @ 0x55870b329480] GopPicSize: 248; GopRefDist: 9; GopOptFlag: closed ; IdrInterval: 1
[hevc_qsv @ 0x55870b329480] TargetUsage: 4; RateControlMethod: ICQ
[hevc_qsv @ 0x55870b329480] ICQQuality: 22
```

I'm not sure if this behaviour is a result of a driver bug, a bug with ffmpeg, or is working as expect as a result of hardware limitations etc.

The system is a Celeron J4105 Gemini Lake board running Ubuntu 20.04. iHD driver details as follows:
```
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.1.1 (2c80a69)
vainfo: Supported profile and entrypoints
```

This behaviour also occurs on an i9-9900k running windows.

XinfengZhang, 2021-03-08:
@onabiull could you help to take look , about target usage <===> preset  mapping.

OttoYang, 2023-03-28:
@caiusseverus  Only TU 1/4/7 are supported in hardware so far.  
So could we close this issue?

caiusseverus, 2023-03-28:
Yes, as it's working as intended that's fine. I couldn't find any reference to it in documentation so wasn't sure if that was the case.

closed 2023-03-28 by caiusseverus