intel/media-driver #1155: AV1 hwdec vaapi visual glitches on Intel Tiger Lake Xe Graphics

rmkimathi, 2021-03-06:
I am not sure if the problem is libva or intel-media-va-driver-non-free? I encoded a video file using ffmpeg libsvtav1 which mpv hwdec=vaapi hwdec-codecs=av1 was able to play flawless. ... All test videos downloaded online will only software decode using libdav1d. (AV1 Hardware decode through mpv produces glitches as per screen shot)

**mpv version:** mpv 0.33.0-93-gaac1844179/FFmpeg version: N-101390-g088f35f036
**Linux Distribution and Version:** Ubuntu 21.04 / 5.10.0-14-generic
**Window Manager and version:** Wayland / GNOME 3.38.3
**CPU:** intel i7-1165G7
**GPU driver and version:** VA-API version: 1.11 (libva 2.10.0) / iHD driver 21.1.2 (7a858c83)

### Sample screenshot
[image]

```
vainfo
libva info: VA-API version 1.11.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_11
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.11 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.1.2 (7a858c83)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain               :	VAEntrypointEncSliceLP
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileHEVCMain10             :	VAEntrypointEncSliceLP
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile1            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileVP9Profile3            :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointEncSlice
      VAProfileHEVCMain422_10         :	VAEntrypointVLD
      VAProfileHEVCMain422_10         :	VAEntrypointEncSlice
      VAProfileHEVCMain422_12         :	VAEntrypointVLD
      VAProfileHEVCMain422_12         :	VAEntrypointEncSlice
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointVLD
      VAProfileHEVCSccMain            :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          :	VAEntrypointVLD
      VAProfileHEVCSccMain10          :	VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         :	VAEntrypointVLD
      VAProfileHEVCSccMain444         :	VAEntrypointEncSliceLP
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointVLD
      VAProfileHEVCSccMain444_10      :	VAEntrypointEncSliceLP
```

XinfengZhang, 2021-03-08:
@rmkimathi  have you tried other method to enable HW decode, such as FFmpeg? to check it is player issue or a driver issue?

rmkimathi, 2021-03-08:
> @rmkimathi have you tried other method to enable HW decode, such as FFmpeg? to check it is player issue or a driver issue?

Okay will continue testing with ffmpeg and see. Thanks

mark-herbert42, 2021-03-09:
Have exactly same issue with mpv. I have tried ffplay and vlc - they work well, but they both fallback to dav1d software decoding ( no GPU activity shown in intel_gpu_top). Only mpv forces hardware AV1 ( intel_gpu_top shows hardware decoding activity) but the video is full of garbage like on screenshot attached.

XinfengZhang, 2021-03-09:
I have not tried ffplayer , could you try the command line like this:
ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -i input.ivf -pix_fmt yuv420p -f rawvideo -vsync passthrough -autoscale 0 -vframes 100 -y output.yuv, could you help to try it?

mark-herbert42, 2021-03-09:
[AVHWDeviceContext @ 0x563c72c8c6c0] libva: VA-API version 1.10.0
[AVHWDeviceContext @ 0x563c72c8c6c0] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x563c72c8c6c0] libva: Trying to open /usr/lib64/va/drivers/iHD_drv_video.so
[AVHWDeviceContext @ 0x563c72c8c6c0] libva: Found init function __vaDriverInit_1_10
[AVHWDeviceContext @ 0x563c72c8c6c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x563c72c8c6c0] Initialised VAAPI connection: version 1.10
[AVHWDeviceContext @ 0x563c72c8c6c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.2 ().
[AVHWDeviceContext @ 0x563c72c8c6c0] Driver not found in known nonstandard list, using standard behaviour.
[libdav1d @ 0x563c72d5c000] libdav1d 0.8.2
Input #0, matroska,webm, from 'Stream3_AV1_720p_3.9mbps.webm':
  Metadata:
    encoder         : libwebm-0.2.1.0
  Duration: 00:03:01.52, start: 0.000000, bitrate: 3924 kb/s
  Stream #0:0(eng): Video: av1 (Main), 1 reference frame, yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1k tbn, 1k tbc (default)
[libdav1d @ 0x563c72d9c040] libdav1d 0.8.2
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (libdav1d) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x563c72dd4140] w:1280 h:720 pixfmt:yuv420p tb:1/1000 fr:25/1 sar:1/1

mark-herbert42, 2021-03-09:
I am using AV1 examples from here 
https://www.elecard.com/videos

They all works well with dav1d software decoding, but with hardware it either fallback to dav1d (ffmpeg, vlc) and work perfectly, or continue using vaapi (mpv) with garbage on screen

rmkimathi, 2021-03-09:
Full log ...
```
ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -i "Stream1_AV1_720p_1.5mbps.webm" -pix_fmt yuv420p -f rawvideo -vsync passthrough -autoscale 0 -vframes 100 -y output.yuv
ffmpeg version N-101387-gf784091 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --prefix=/home/kimathi/ffmpeg-static/build/snapshot --pkg-config-flags=--static --extra-cflags=-I/home/kimathi/ffmpeg-static/build/snapshot/include --extra-ldflags=-L/home/kimathi/ffmpeg-static/build/snapshot/lib --extra-libs='-lpthread -lm' --bindir=/home/kimathi/ffmpeg-static/build/snapshot/bin --enable-gpl --enable-gnutls --enable-libass --enable-libfdk-aac --enable-libfreetype --enable-libmp3lame --enable-libopus --enable-libvpx --enable-libx264 --enable-libx265 --enable-libdav1d --enable-libsvtav1 --enable-nonfree
  libavutil      56. 66.100 / 56. 66.100
  libavcodec     58.128.100 / 58.128.100
  libavformat    58. 70.100 / 58. 70.100
  libavdevice    58. 12.100 / 58. 12.100
  libavfilter     7.107.100 /  7.107.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
  libpostproc    55.  8.100 / 55.  8.100
[AVHWDeviceContext @ 0x5591dde74080] libva: VA-API version 1.11.0
[AVHWDeviceContext @ 0x5591dde74080] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x5591dde74080] libva: Trying to open /usr/local/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x5591dde74080] libva: Found init function __vaDriverInit_1_11
[AVHWDeviceContext @ 0x5591dde74080] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x5591dde74080] Initialised VAAPI connection: version 1.11
[AVHWDeviceContext @ 0x5591dde74080] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.2 (7a858c83).
[AVHWDeviceContext @ 0x5591dde74080] Driver not found in known nonstandard list, using standard behaviour.
[libdav1d @ 0x5591ddf43400] libdav1d 0.8.2
Input #0, matroska,webm, from 'Stream1_AV1_720p_1.5mbps.webm':
  Metadata:
    encoder         : libwebm-0.2.1.0
  Duration: 00:02:54.20, start: 0.000000, bitrate: 1458 kb/s
  Stream #0:0(eng): Video: av1 (Main), 1 reference frame, yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1k tbn, 1k tbc (default)
[libdav1d @ 0x5591ddf447c0] libdav1d 0.8.2
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (libdav1d) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x5591ddf90780] w:1280 h:720 pixfmt:yuv420p tb:1/1000 fr:25/1 sar:1/1
Output #0, rawvideo, to 'output.yuv':
  Metadata:
    encoder         : Lavf58.70.100
  Stream #0:0(eng): Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p(tv, progressive), 1280x720 (0x0) [SAR 1:1 DAR 16:9], q=2-31, 276480 kb/s, 25 fps, 25 tbn (default)
    Metadata:
      encoder         : Lavc58.128.100 rawvideo
No more output streams to write to, finishing.e=00:00:00.04 bitrate=262144.0kbits/s speed= 348x    
frame=  100 fps=0.0 q=-0.0 Lsize=  135000kB time=00:00:04.00 bitrate=276480.0kbits/s speed=32.5x    
video:135000kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (Stream1_AV1_720p_1.5mbps.webm):
  Input stream #0:0 (video): 103 packets read (669042 bytes); 101 frames decoded; 
  Total: 103 packets (669042 bytes) demuxed
Output file #0 (output.yuv):
  Output stream #0:0 (video): 100 frames encoded; 100 packets muxed (138240000 bytes); 
  Total: 100 packets (138240000 bytes) muxed
[AVIOContext @ 0x5591ddf46880] Statistics: 0 seeks, 528 writeouts
[AVIOContext @ 0x5591ddf49a40] Statistics: 674279 bytes read, 0 seeks
```

XinfengZhang, 2021-03-11:
@rmkimathi , is there output.yuv? is there also glitches on the output.yuv?

rmkimathi, 2021-03-11:
> @rmkimathi , is there output.yuv? is there also glitches on the output.yuv?

yes the output.yuv also has glitches

[image]

Jexu, 2021-03-30:
@rmkimathi 
can you get the av1 bitstream from the .wbem file with ffmpeg: 
    ffmpeg -i Stream3_AV1_720p_3.9mbps.webm -vcodec copy -an  av1.ivf
then try ffmpeg to decode the av1.ivf:
    ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -i "av1.ivf" -pix_fmt yuv420p -f rawvideo -vsync passthrough  -vframes 100 -y output.yuv

the result looks correct from my side.

[image]

mark-herbert42, 2021-03-30:
I have tried this 2 step operation. av1.ivf in mpv is still showing garbage. output.yuv I could not open - both vlc and mpv shows nothing

mark-herbert42, 2021-03-31:
yes, tried it this way it works - but it is a must send video_size paramater to ffplay otherwise it aborts immediately same way as vlc or mpv.

rmkimathi, 2021-03-31:
Yes, using your parameters produces a clean picture. I am wondering why it is not possible to get clean picture with mpv hwdec or with ffplay playing the sample files directly? (videos exported with libsvtav1 however do work) .... for example:

ffmpeg -v verbose -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -c:v av1 -i Stream3_AV1_720p_3.9mbps.webm -c:v rawvideo -c:a copy -f matroska - | ffplay -x 1280 -y 720 -i -

[image]

Jexu, 2021-04-06:
> yes, tried it this way it works - but it is a must send video_size paramater to ffplay otherwise it aborts immediately same way as vlc or mpv.

it is app behavior of how to play the yuv file which is not related to media driver.  regarding ffplay has video_size paramater, i think picture size is required

rmkimathi, 2021-04-06:
@Jexu 

I have sent same bug report to @mpv-player As for @ffmpeg don't know how to file their bugs. Kindly assist if possible to keep @ffmpeg and @mpv-player in the loop for a possible diagnosis and fix. Thanks

mark-herbert42, 2021-04-11:
New ffmpeg 4.4 released enabling hardware av1 by default. Now mpv works as it worked - terrible artifacts all over. Vlc also started to use the hardware - and with vlc it much worse. it starts with normal picture sometimes - but it lasts couple of seconds. Then it hangs - absolutely deadly , so even  kill -9 can not remove it, only hard reset.  In that test of YUV conversion that was thr reason to blame mpv for all the issues not the buggy driver only small part of video was converted - about the same that VLC tries to show before it dies.

Jexu, 2021-04-14:
since the av1 bistream is wrapped by .webm and app needs to do demux firstly. the av1 bitstream itself can be decoded by ffmpeg hw. so can we conclude it is app issue?

rmkimathi, 2021-04-14:
> since the av1 bistream is wrapped by .webm and app needs to do demux firstly. the av1 bitstream itself can be decoded by ffmpeg hw. so can we conclude it is app issue?

@Jexu 

According to:

`sudo intel_gpu_top`

You must specify `-c:v av1` before -i "av1.ivf" otherwise the decoding is done in software libdav1d in my case.

So the parameters that seem to go through hw are (the resultant output.yuv picture will be corrupted):

`ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -c:v av1 -i "av1.ivf" -pix_fmt yuv420p -f rawvideo -vsync passthrough -vframes 100 -y output.yuv`

rmkimathi, 2021-04-14:
## Update

### With "-c:v av1"
(av1 (native) -> rawvideo (native))

```
[AVHWDeviceContext @ 0x55e9724598c0] libva: VA-API version 1.12.0
[AVHWDeviceContext @ 0x55e9724598c0] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x55e9724598c0] libva: Trying to open /usr/local/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55e9724598c0] libva: Found init function __vaDriverInit_1_12
[AVHWDeviceContext @ 0x55e9724598c0] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x55e9724598c0] Initialised VAAPI connection: version 1.12
[AVHWDeviceContext @ 0x55e9724598c0] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.3 (bc101f33).
[AVHWDeviceContext @ 0x55e9724598c0] Driver not found in known nonstandard list, using standard behaviour.
[libdav1d @ 0x55e972528140] libdav1d 0.8.2
Input #0, ivf, from 'av1.ivf':
  Duration: 00:03:01.56, start: 0.000000, bitrate: 3924 kb/s
  Stream #0:0: Video: av1 (Main), 1 reference frame (AV01 / 0x31305641), yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1k tbn, 1k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (native) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x55e972b2e0c0] w:1280 h:720 pixfmt:nv12 tb:1/1000 fr:25/1 sar:1/1
[auto_scaler_0 @ 0x55e972b30d80] w:iw h:ih flags:'bicubic' interl:0
[format @ 0x55e972b2ef80] auto-inserting filter 'auto_scaler_0' between the filter 'Parsed_null_0' and the filter 'format'
[auto_scaler_0 @ 0x55e972b30d80] w:1280 h:720 fmt:nv12 sar:1/1 -> w:1280 h:720 fmt:yuv420p sar:1/1 flags:0x4
Output #0, rawvideo, to 'output.yuv':
  Metadata:
    encoder         : Lavf58.78.100
  Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p(tv, progressive), 1280x720 (0x0) [SAR 1:1 DAR 16:9], q=2-31, 276480 kb/s, 25 fps, 25 tbn
    Metadata:
      encoder         : Lavc58.136.101 rawvideo
No more output streams to write to, finishing.e=00:00:00.04 bitrate=262144.0kbits/s speed=4e+04x    
frame=  100 fps=0.0 q=-0.0 Lsize=  135000kB time=00:00:04.00 bitrate=276480.0kbits/s speed=13.2x    
video:135000kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (av1.ivf):
  Input stream #0:0 (video): 100 packets read (2813063 bytes); 100 frames decoded; 
  Total: 100 packets (2813063 bytes) demuxed
Output file #0 (output.yuv):
  Output stream #0:0 (video): 100 frames encoded; 100 packets muxed (138240000 bytes); 
  Total: 100 packets (138240000 bytes) muxed
[AVIOContext @ 0x55e972529c80] Statistics: 0 seeks, 528 writeouts
[AVIOContext @ 0x55e97252f680] Statistics: 2841097 bytes read, 0 seeks
```

### Without "-c:v av1"
(av1 (libdav1d) -> rawvideo (native))

```
[AVHWDeviceContext @ 0x558cc0521880] libva: VA-API version 1.12.0
[AVHWDeviceContext @ 0x558cc0521880] libva: User environment variable requested driver 'iHD'
[AVHWDeviceContext @ 0x558cc0521880] libva: Trying to open /usr/local/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x558cc0521880] libva: Found init function __vaDriverInit_1_12
[AVHWDeviceContext @ 0x558cc0521880] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x558cc0521880] Initialised VAAPI connection: version 1.12
[AVHWDeviceContext @ 0x558cc0521880] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.3 (bc101f33).
[AVHWDeviceContext @ 0x558cc0521880] Driver not found in known nonstandard list, using standard behaviour.
[libdav1d @ 0x558cc05f0100] libdav1d 0.8.2
Input #0, ivf, from 'av1.ivf':
  Duration: 00:03:01.56, start: 0.000000, bitrate: 3924 kb/s
  Stream #0:0: Video: av1 (Main), 1 reference frame (AV01 / 0x31305641), yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1k tbn, 1k tbc
[libdav1d @ 0x558cc05f1340] libdav1d 0.8.2
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (libdav1d) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x558cc0668780] w:1280 h:720 pixfmt:yuv420p tb:1/1000 fr:25/1 sar:1/1
Output #0, rawvideo, to 'output.yuv':
  Metadata:
    encoder         : Lavf58.78.100
  Stream #0:0: Video: rawvideo, 1 reference frame (I420 / 0x30323449), yuv420p(tv, progressive), 1280x720 (0x0) [SAR 1:1 DAR 16:9], q=2-31, 276480 kb/s, 25 fps, 25 tbn
    Metadata:
      encoder         : Lavc58.136.101 rawvideo
No more output streams to write to, finishing.e=00:00:00.04 bitrate=262144.0kbits/s speed=1e+03x    
frame=  100 fps=0.0 q=-0.0 Lsize=  135000kB time=00:00:04.00 bitrate=276480.0kbits/s dup=0 drop=1 speed=10.4x    
video:135000kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000000%
Input file #0 (av1.ivf):
  Input stream #0:0 (video): 103 packets read (3142759 bytes); 102 frames decoded; 
  Total: 103 packets (3142759 bytes) demuxed
Output file #0 (output.yuv):
  Output stream #0:0 (video): 100 frames encoded; 100 packets muxed (138240000 bytes); 
  Total: 100 packets (138240000 bytes) muxed
[AVIOContext @ 0x558cc05f6040] Statistics: 0 seeks, 528 writeouts
[AVIOContext @ 0x558cc05f7640] Statistics: 3176778 bytes read, 0 seeks
```

rmkimathi, 2021-05-06:
I reported this AV1 bug way back in March 2021 hoping for a fix in time for AV1 VAAPI integration into KODI XBMC ..... is there someone who is able to reach out to @ffmpeg or @intel directly? @a1rwulf you may have to raise a new issue as this one seems to have been ignored.

a1rwulf, 2021-05-06:
@Jexu mentioned ffplay can render correctly. I'll give it a shot and see if I can find the difference.

rmkimathi, 2021-05-06:
AV1 hardware only available on (av1 (native) -> rawvideo (native)). .... You must specify -c:v av1 before ffplay input

Jexu, 2021-05-13:
@rmkimathi 
jsut double confirm again. the decode really uses hw decode even without param '-c:v av1', if you are interested in that, you can just debug into media driver to see it. or you can also enable libva trace to confirm it.

so, without param '-c:v av1', the output yuv does be correct, and i am not sure of this input param in ffmpeg, but it maybe related ffmpeg itself.

rmkimathi, 2021-05-13:
I think this can be closed. @1rwulf will follow up

Jexu, 2021-05-13:
jsut confirmed again, ffmpeg really cannot decode correct this content, but i also tried with sample_decode, it can decode correct output yuv:
sample_decode av1 -i  av1.ivf -n 100  -hw -o out.yuv

Jexu, 2021-05-13:
compared with sample_decode, ffmpeg seems send incorrect bitstream to media_driver, so can you help report issue to ffmpeg?

rmkimathi, 2021-05-13:
I will let @1rwulf follow up with @ffmpeg. He has better connections. Thanks

a1rwulf, 2021-05-13:
@Jexu can you be a bit more specific about the difference you found regarding the bitstream?

Jexu, 2021-05-14:
@a1rwulf  i add the libva trace dump of 10 frames here for both ffmpeg and sample_decode. you can search the buffer type of 'VASliceDataBufferType' to check the bitstream data.

[ffmpeg_trace.txt](https://github.com/intel/media-driver/files/6476152/ffmpeg_trace.txt)
[sample_decode_trace.txt](https://github.com/intel/media-driver/files/6476153/sample_decode_trace.txt)

XinfengZhang, 2021-05-14:
hi @Jexu @rmkimathi  for multiple tile cases, seems ffmpeg need a patch https://github.com/intel-media-ci/ffmpeg/pull/403

Jexu, 2021-05-14:
@XinfengZhang @a1rwulf  By applying above patch, the ffmepg can also decode the clip correctly now.  with above patch, ffmpeg can send correct size of bitstream to driver.

a1rwulf, 2021-05-14:
@XinfengZhang @Jexu  I've compiled https://github.com/intel-media-ci/ffmpeg/pull/403 and there's a major improvement to notice, however I still see artifacts.
They disappear every few seconds and reappear afterwards.
Looks a bit like IDR-Frames reset the image and it starts to artifact on reference frames then next IDR-Frame will correct the picture again.

rmkimathi, 2021-05-14:
Well Done!

Patched ffmpeg-git with your fix and success! mpv hwdec vaapi av1 is now working.
```
diff --git a/libavcodec/vaapi_av1.c b/libavcodec/vaapi_av1.c
index 1809b485aa..16b7e35747 100644
--- a/libavcodec/vaapi_av1.c
+++ b/libavcodec/vaapi_av1.c
@@ -292,7 +292,7 @@ static int vaapi_av1_decode_slice(AVCodecContext *avctx,
         err = ff_vaapi_decode_make_slice_buffer(avctx, pic, &slice_param,
                                                 sizeof(VASliceParameterBufferAV1),
                                                 buffer,
-                                                s->tile_group_info[i].tile_size);
+                                                size);
         if (err) {
             ff_vaapi_decode_cancel(avctx, pic);
             return err;
```

[image]

a1rwulf, 2021-05-15:
@mark-herbert42 @rmkimathi I'm trying to verify using mpv as well as I don't see the issue solved when using kodi.
However my mpv doesn't work at all and always falls back to sw decoding.
Can you share your exact configuration and/or a full log of mpv's output?

rmkimathi, 2021-05-15:
> @mark-herbert42 @rmkimathi I'm trying to verify using mpv as well as I don't see the issue solved when using kodi.
> However my mpv doesn't work at all and always falls back to sw decoding.
> Can you share your exact configuration and/or a full log of mpv's output?

Ubuntu 21.04 Wayland, Linux 5.11.0-18-generic
mpv 0.33.0-153-gded36a4470 with FFmpeg version: N-102544-gc9a79532e5 (compiled with mpv-build script)

### I compiled and installed everything git for intel-media-va-driver-non-free
gmmlib
libva
libva-utils
media-driver

```
vainfo
libva info: VA-API version 1.12.0
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.12 (libva 2.12.0.pre1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.2.1 (0368d583)
```
### mpv short log
```
kimathi@acer:~/Videos/AV1$ mpv Stream3_AV1_720p_3.9mbps.webm
 (+) Video --vid=1 (*) (av1 1280x720)
[vo/gpu/wayland] GNOME's wayland compositor lacks support for the idle inhibit protocol. This means the screen can blank during playback.
[ffmpeg] AVHWDeviceContext: Failed to query surface attributes: 20 (the requested function is not implemented).
[vo/gpu/vaapi-egl] failed to retrieve libavutil frame constraints
Using hardware decoding (vaapi).
VO: [gpu] 1280x720 vaapi[nv12]
V: 00:00:03 / 00:03:01 (2%)

Exiting... (Quit)
```

### mpv verbose log
```
kimathi@acer:~/Videos/AV1$ mpv -v Stream3_AV1_720p_3.9mbps.webm
[cplayer] Command line options: '-v' 'Stream3_AV1_720p_3.9mbps.webm'
[cplayer] mpv 0.33.0-153-gded36a4470 Copyright © 2000-2020 mpv/MPlayer/mplayer2 projects
[cplayer]  built on Thu May 13 20:11:56 EAT 2021
[cplayer] FFmpeg library versions:
[cplayer]    libavutil       57.0.100
[cplayer]    libavcodec      59.1.100
[cplayer]    libavformat     59.2.100
[cplayer]    libswscale      6.0.100
[cplayer]    libavfilter     8.0.101
[cplayer]    libswresample   4.0.100
[cplayer] FFmpeg version: N-102544-gc9a79532e5
[cplayer] 
[cplayer] Configuration: ./waf configure --enable-libmpv-shared --enable-html-build --enable-pdf-build
[cplayer] List of enabled features: 52arch alsa asm build-date caca cplayer cplugins debug-build drm egl egl-drm egl-helpers egl-x11 ffmpeg gbm gbm.h gl gl-wayland glibc-thread-name glob glob-posix gpl iconv jack jpeg lcms2 libarchive libass libavdevice libbluray libdl libm libmpv-shared librt linux-fstatfs linux-input-event-codes lua memfd_create optimize plain-gl posix posix-or-mingw pthreads pulse rubberband stdatomic uchardet vaapi vaapi-drm vaapi-egl vaapi-wayland vaapi-x-egl vaapi-x11 vdpau vt.h wayland wayland-protocols x11 xv zlib
[cplayer] Reading config file /home/kimathi/.config/mpv/mpv.conf
[cplayer] Applying profile 'default'...
[cplayer] Setting option 'hwdec' = 'vaapi' (flags = 4)
[cplayer] Setting option 'hwdec-codecs' = 'h264,vc1,hevc,vp8,vp9,av1' (flags = 4)
[cplayer] Setting option 'geometry' = '1280x720' (flags = 4)
[cplayer] Setting option 'screenshot-directory' = '~/Pictures' (flags = 4)
[cplayer] Setting option 'sub-font' = 'DejaVu Sans Bold' (flags = 4)
[cplayer] Setting option 'sub-font-size' = '40' (flags = 4)
[cplayer] Setting option 'v' = '' (flags = 8)
[cplayer] Waiting for scripts...
[osd/libass] libass API version: 0x1501000
[osd/libass] libass source: commit: 0.15.1-5-gb9f34688f6ef1988136ba5e90a91ae8ac9b8a55f
[osd/libass] Shaper: FriBidi 1.0.8 (SIMPLE) HarfBuzz-ng 2.7.4 (COMPLEX)
[osd/libass] Setting up fonts...
[osd/libass] Using font provider fontconfig
[osd/libass] Done.
[cplayer] Set property: shared-script-properties -> 1
[... 105 lines cut ...]
[vo/gpu/wayland] Handling resize on the egl side
[vo/gpu] Resize: 1280x720
[vo/gpu] Window size: 1280x720 (Borders: l=0 t=0 r=0 b=0)
[vo/gpu] Video source: 1280x720 (1:1)
[vo/gpu] Video display: (0, 0) 1280x720 -> (0, 0) 1280x720
[vo/gpu] Video scale: 1.000000/1.000000
[vo/gpu] OSD borders: l=0 t=0 r=0 b=0
[vo/gpu] Video borders: l=0 t=0 r=0 b=0
[vo/gpu] Reported display depth: 10
[cplayer] first video frame after restart shown
[cplayer] playback restart complete @ 0.000000, audio=eof, video=playing
[statusline] V: 00:00:00 / 00:03:01 (0%)
[vo/gpu/wayland] Given DND offer with mime type text/plain;charset=utf-8
[vo/gpu/wayland] Surface entered output LGD 0x065a (0x4), scale = 1
[vo/gpu] Assuming 59.977000 FPS for display sync.
[statusline] V: 00:00:00 / 00:03:01 (0%)
[cplayer] Set property: shared-script-properties -> 1
[statusline] V: 00:00:02 / 00:03:01 (1%)
[cplayer] EOF code: 5  
[cplayer] finished playback, success (reason 3)
[cplayer] 
[cplayer] Exiting... (Quit)
[cplayer] Set property: shared-script-properties -> 1
[vo/gpu/wayland] Deregistering output LGD 0x065a (0x4)
```

rmkimathi, 2021-05-15:
### ffplay pipe

`
ffmpeg -v verbose -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -c:v av1 -i Stream3_AV1_720p_3.9mbps.webm -c:v rawvideo -c:a copy -f matroska - | ffplay -x 1280 -y 720 -i -
`

### ffmpeg log

```
[libdav1d @ 0x55be442728c0] libdav1d 0.8.2
Input #0, matroska,webm, from 'Stream3_AV1_720p_3.9mbps.webm':
  Metadata:
    encoder         : libwebm-0.2.1.0
  Duration: 00:03:01.52, start: 0.000000, bitrate: 3924 kb/s
  Stream #0:0(eng): Video: av1 (Main), 1 reference frame, yuv420p(tv), 1280x720 [SAR 1:1 DAR 16:9], 25 fps, 25 tbr, 1k tbn (default)
[AVHWDeviceContext @ 0x55be442a7a00] libva: VA-API version 1.12.0
[AVHWDeviceContext @ 0x55be442a7a00] libva: Trying to open /usr/local/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x55be442a7a00] libva: Found init function __vaDriverInit_1_12
[AVHWDeviceContext @ 0x55be442a7a00] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x55be442a7a00] Initialised VAAPI connection: version 1.12
[AVHWDeviceContext @ 0x55be442a7a00] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.2.1 (0368d583).
[AVHWDeviceContext @ 0x55be442a7a00] Driver not found in known nonstandard list, using standard behaviour.
Stream mapping:
  Stream #0:0 -> #0:0 (av1 (native) -> rawvideo (native))
Press [q] to stop, [?] for help
[graph 0 input from stream 0:0 @ 0x55be44b72340] w:1280 h:720 pixfmt:nv12 tb:1/1000 fr:25/1 sar:1/1
Output #0, matroska, to 'pipe:':
  Metadata:
    encoder         : Lavf59.2.100
  Stream #0:0(eng): Video: rawvideo, 1 reference frame (NV12 / 0x3231564E), nv12(tv, progressive), 1280x720 (0x0) [SAR 1:1 DAR 16:9], q=2-31, 276480 kb/s, 25 fps, 1k tbn (default)
    Metadata:
      encoder         : Lavc59.1.100 rawvideo
Input #0, matroska,webm, from 'pipe:': vq=    0KB sq=    0B f=0/0   4104.0kbits/s speed=N/A    
  Metadata:
    ENCODER         : Lavf59.2.100
  Duration: N/A, start: 0.000000, bitrate: N/A
  Stream #0:0(eng): Video: rawvideo (NV12 / 0x3231564E), nv12(tv, progressive), 1280x720, SAR 1:1 DAR 16:9, 25 fps, 25 tbr, 1k tbn (default)
    Metadata:
      ENCODER         : Lavc59.1.100 rawvideo
   3.46 M-V:  0.000 fd=   0 aq=    0KB vq=16200KB sq=    0B f=0/0   276411.8kbits/s speed= 1.2x
```

a1rwulf, 2021-05-15:
Alright, I can also confirm that the patch works fine in mpv/kodi.

Jexu, 2021-06-15:
Hi @XinfengZhang 
Please help close this issue while the fix has been merged.

a1rwulf, 2021-06-15:
@Jexu that's right for 8bit non film grain content the fix has been merged and merged into upstream.
For 10bit film grain there is this open PR: https://github.com/intel-media-ci/ffmpeg/pull/412

babam86, 2021-10-19:
@rmkimathi

Can Mpv using HW decoding properly?

Distro: Slackware64 Current
Cpu: Intel i3 1115G4
Graphic: Intel UHD Graphics Xe G4
Ffmpeg-git (20211017)
Mpv 0.33.0

mpv --hwdec=vaapi Stream1_AV1_HD_2.7mbps.webm  ---> Broken

ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -c:v av1 -i Stream1_AV1_HD_2.7mbps.webm -c:v rawvideo -c:a copy -f nut - | mpv -  ---> Works fine

Thanks.

rmkimathi, 2021-10-19:
> @rmkimathi
>
> Can Mpv using HW decoding properly?
>
> Distro: Slackware64 Current Cpu: Intel i3 1115G4 Graphic: Intel UHD Graphics Xe G4 Ffmpeg-git (20211017) Mpv 0.33.0
>
> mpv --hwdec=vaapi [...]

The fix was applied in ffmpeg git. So you have to mpv-build with latest ffmpeg to enjoy proper AV1 hw decode