intel/media-driver #1157: do_drawable_hash crashes on centos7.6

canyuedao, 2021-03-09:
the stack trace:

```libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva info: va_openDriver() returns 0
[New Thread 0x7fff6405f700 (LWP 7436)]
[New Thread 0x7fff54125700 (LWP 7437)]
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffe1c8a9700 (LWP 7434)]
0x0000000000000000 in ?? ()
Missing separate debuginfos, use: debuginfo-install alsa-lib-1.1.6-2.el7.x86_64 bzip2-libs-1.0.6-13.el7.x86_64 cairo-1.15.12-4.el7.x86_64 dbus-libs-1.10.24-12.el7.x86_64 elfutils-libelf-0.172-2.el7.x86_64 elfutils-libs-0.172-2.el7.x86_64 expat-2.1.0-12.el7.x86_64 fontconfig-2.13.0-4.3.el7.x86_64 freetype-2.8-14.el7_9.1.x86_64 glib2-2.56.1-8.el7.x86_64 glibc-2.17-260.el7.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.15.1-34.el7.x86_64 libX11-1.6.5-2.el7.x86_64 libXau-1.0.8-2.1.el7.x86_64 libXcursor-1.1.15-1.el7.x86_64 libXdamage-1.1.4-4.1.el7.x86_64 libXext-1.3.3-3.el7.x86_64 libXfixes-5.0.3-1.el7.x86_64 libXrandr-1.5.1-2.el7.x86_64 libXrender-0.9.10-1.el7.x86_64 libXv-1.0.11-1.el7.x86_64 libXxf86vm-1.1.4-1.el7.x86_64 libattr-2.4.46-13.el7.x86_64 libcap-2.22-9.el7.x86_64 libcom_err-1.42.9-13.el7.x86_64 libdrm-2.4.97-2.el7.x86_64 libgcc-4.8.5-36.el7.x86_64 libgcrypt-1.5.3-14.el7.x86_64 libglvnd-1.0.1-0.8.git5baa1e5.el7.x86_64 libglvnd-egl-1.0.1-0.8.git5baa1e5.el7.x86_64 libglvnd-glx-1.0.1-0.8.git5baa1e5.el7.x86_64 libgomp-4.8.5-36.el7.x86_64 libgpg-error-1.12-3.el7.x86_64 libpng-1.5.13-8.el7.x86_64 libpng12-1.2.50-10.el7.x86_64 libselinux-2.5-14.1.el7.x86_64 libstdc++-4.8.5-36.el7.x86_64 libuuid-2.23.2-65.el7_9.1.x86_64 libxcb-1.13-1.el7.x86_64 libxkbcommon-0.7.1-1.el7.x86_64 libxkbcommon-x11-0.7.1-1.el7.x86_64 libxshmfence-1.2-1.el7.x86_64 llvm-private-6.0.1-2.el7.x86_64 lz4-1.7.5-2.el7.x86_64 mesa-dri-drivers-18.0.5-3.el7.x86_64 mesa-libGL-18.3.4-12.el7_9.x86_64 mesa-libglapi-18.3.4-12.el7_9.x86_64 openssl-libs-1.0.2k-16.el7.x86_64 pcre-8.32-17.el7.x86_64 pixman-0.34.0-1.el7.x86_64 sqlite-3.7.17-8.el7.x86_64 systemd-libs-219-62.el7.x86_64 xz-libs-5.2.2-1.el7.x86_64 zlib-1.2.7-19.el7_9.x86_64
(gdb) bt
#0  0x0000000000000000 in ?? ()
#1  0x00007fffdece50bc in do_drawable_hash (drawable=48234497, ctx=0x7ffde4011290) at va_dricommon.c:74
#2  va_dri_get_drawable (ctx=0x7ffde4011290, drawable=48234497) at va_dricommon.c:122
#3  0x00007ffdedfa2ceb in DdiCodec_PutSurfaceLinuxHW (ctx=ctx@entry=0x7ffde4011290, surface=surface@entry=1, draw=draw@entry=0x2e00001, srcx=srcx@entry=0, srcy=srcy@entry=0, srcw=srcw@entry=352, srch=288, destx=0, desty=0, destw=791, 
    desth=388, cliprects=0x0, number_cliprects=0, flags=0) at /home/media-driver-intel-media-20.4.5/media_driver/linux/common/ddi/media_libva_putsurface_linux.cpp:392
#4  0x00007ffdedf85c29 in DdiMedia_PutSurface (ctx=0x7ffde4011290, surface=1, draw=0x2e00001, srcx=<optimized out>, srcy=<optimized out>, srcw=<optimized out>, srch=288, destx=0, desty=0, destw=791, desth=388, cliprects=0x0, 
    number_cliprects=0, flags=0) at /home/media-driver-intel-media-20.4.5/media_driver/linux/common/ddi/media_libva.cpp:4374
#5  0x00007fffeaae1ff2 in vlc_glx_map () from /usr/jstnvr/libnvvapi.so
#6  0x00007ffff77d9181 in gpuRender::ReceiveFrame(AVFrame*, int, int, long, char*) () from /usr/jstnvr/libNVPlayM4.so.1
#7  0x00007ffff77dea47 in CDisplayMgr::Render_InputV12Ex(long, int*, void**, int, int, unsigned long long, char*, void*, int) () from /usr/jstnvr/libNVPlayM4.so.1
#8  0x00007ffff77e8423 in NVPlayM4_Render_InputV12Ex () from /usr/jstnvr/libNVPlayM4.so.1
#9  0x00007ffff77ed43e in stTunnelItem::ReceiveFrame(AVFrame*, int, int, long, char*) () from /usr/jstnvr/libNVPlayM4.so.1
#10 0x00007ffff77f1372 in CTunnelStream::ReceiveFrame(AVFrame*, int, int) () from /usr/jstnvr/libNVPlayM4.so.1
#11 0x00007ffff77e3def in CFFVideo::decOneFrame(unsigned char*, int) () from /usr/jstnvr/libNVPlayM4.so.1
#12 0x00007ffff77ee80d in CTunnelStream::DecodeVideo(unsigned char*, int) () from /usr/jstnvr/libNVPlayM4.so.1
#13 0x00007ffff77f0c73 in CTunnelStream::playRealStream() () from /usr/jstnvr/libNVPlayM4.so.1
#14 0x00007ffff77f0f88 in CTunnelStream::singleThreadRun() () from /usr/jstnvr/libNVPlayM4.so.1
#15 0x00007ffff6f109ea in nmnvrlib::NVRSingleThread::singleThreadStreamFunc(void*) () from /usr/jstnvr/libnvrlib.so.1
#16 0x00007ffff6f122c7 in tp_work_thread () from /usr/jstnvr/libnvrlib.so.1
#17 0x00007ffff0065dd5 in start_thread () from /lib64/libpthread.so.0
#18 0x00007fffef56fead in clone () from /lib64/libc.so.6
```

the vainfo:
```
[root@nvss jstnvr]# vainfo 
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 20.4.5 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```

Could you tell me how to solve this problem?

XinfengZhang, 2021-03-15:
suppose you are not using  X11, right?

canyuedao, 2021-03-16:
> suppose you are not using X11, right?

I'm using X11, GNOME.
This is my X version and cpu info:

```
[root@nvss jstnvr]# X -version

X.Org X Server 1.20.1
X Protocol Version 11, Revision 0
Build Operating System:  3.10.0-693.17.1.el7.x86_64 
Current Operating System: Linux nvss 5.2.10-1.el7.elrepo.x86_64 #1 SMP Sun Aug 25 11:20:04 EDT 2019 x86_64
Kernel command line: BOOT_IMAGE=/vmlinuz-5.2.10-1.el7.elrepo.x86_64 root=/dev/mapper/centos-root ro crashkernel=auto rd.lvm.lv=centos/root rd.lvm.lv=centos/swap rhgb quiet
Build Date: 02 November 2018  05:36:09PM
Build ID: xorg-x11-server 1.20.1-3.el7 
Current version of pixman: 0.34.0
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[root@nvss jstnvr]# lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 165
Model name:            Intel(R) Core(TM) i3-10100 CPU @ 3.60GHz
Stepping:              3
CPU MHz:               4100.285
CPU max MHz:           4300.0000
CPU min MHz:           800.0000
BogoMIPS:              7200.00
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              6144K
NUMA node0 CPU(s):     0-7
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb invpcid_single ssbd ibrs ibpb stibp ibrs_enhanced tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid mpx rdseed adx smap clflushopt intel_pt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp md_clear flush_l1d arch_capabilities
```

canyuedao, 2021-03-18:
> suppose you are not using X11, right?

The other libva interface seem to be ok,  just when I call the vaPutSurface interface, my program crashes.
How is the problem occurs?
Is my environment configuration incorrect?

canyuedao, 2021-03-18:
> suppose you are not using X11, right?

Hi, @XinfengZhang 
I have enable LIBVA_TRACE, and generate two files:

1. 1.142753.thd-0x00006fb5
```
[25449.028275][ctx       none]==========va_TraceInitialize
[25449.028282][ctx       none]=========vaInitialize ret = VA_STATUS_SUCCESS, success (no error) 
[25449.028285][ctx       none]=========vaQueryConfigProfiles ret = VA_STATUS_SUCCESS, success (no error) 
[25449.028286][ctx       none]=========vaGetConfigAttributes ret = VA_STATUS_SUCCESS, success (no error) 
[25449.028287][ctx       none]==========va_TraceCreateConfig
[25449.028287][ctx       none]	profile = 7
[25449.028288][ctx       none]	entrypoint = 1
[25449.028288][ctx       none]	num_attribs = 1
[25449.028288][ctx       none]		attrib_list[0].type = 0x00000000
[25449.028289][ctx       none]		attrib_list[0].value = 0x00020003
[25449.028294][ctx       none]=========vaCreateConfig ret = VA_STATUS_SUCCESS, success (no error) 
[25449.030792][ctx       none]==========va_TraceCreateSurfaces
[25449.030794][ctx       none]	width = 352
[25449.030794][ctx       none]	height = 288
[25449.030795][ctx       none]	format = 1
[25449.030795][ctx       none]	num_surfaces = 18
[25449.030796][ctx       none]		surfaces[0] = 0x00000000
[25449.030796][ctx       none]		surfaces[1] = 0x00000001
[25449.030797][ctx       none]		surfaces[2] = 0x00000002
[25449.030797][ctx       none]		surfaces[3] = 0x00000003
[25449.030797][ctx       none]		surfaces[4] = 0x00000004
[25449.030798][ctx       none]		surfaces[5] = 0x00000005
[25449.030798][ctx       none]		surfaces[6] = 0x00000006
[25449.030799][ctx       none]		surfaces[7] = 0x00000007
[25449.030799][ctx       none]		surfaces[8] = 0x00000008
[25449.030800][ctx       none]		surfaces[9] = 0x00000009
[25449.030800][ctx       none]		surfaces[10] = 0x0000000a
[25449.030801][ctx       none]		surfaces[11] = 0x0000000b
[25449.030801][ctx       none]		surfaces[12] = 0x0000000c
[25449.030802][ctx       none]		surfaces[13] = 0x0000000d
[25449.030802][ctx       none]		surfaces[14] = 0x0000000e
[25449.030802][ctx       none]		surfaces[15] = 0x0000000f
[25449.030803][ctx       none]		surfaces[16] = 0x00000010
[25449.030803][ctx       none]		surfaces[17] = 0x00000011
[25449.030804][ctx       none]	attrib_list[0] =
[25449.030804][ctx       none]		type = 1
[25449.030804][ctx       none]		flags = 2
[25449.030805][ctx       none]		value.type = 1
[25449.030805][ctx       none]		value.value.i = 0x3231564e
[25449.030811][ctx       none]=========vaCreateSurfaces ret = VA_STATUS_SUCCESS, success (no error) 
[25449.031223][ctx 0x10000000]==========va_TraceCreateContext
[25449.031225][ctx 0x10000000]	context = 0x10000000 va_trace_flag 0x1
[25449.031225][ctx 0x10000000]	profile = 7 entrypoint = 1
[25449.031226][ctx 0x10000000]	config = 0x00000004
[25449.031226][ctx 0x10000000]	width = 352
[25449.031227][ctx 0x10000000]	height = 288
[25449.031227][ctx 0x10000000]	flag = 0x00000001
[25449.031228][ctx 0x10000000]	num_render_targets = 18
[25449.031228][ctx 0x10000000]		render_targets[0] = 0x00000000
[25449.031229][ctx 0x10000000]		render_targets[1] = 0x00000001
[25449.031229][ctx 0x10000000]		render_targets[2] = 0x00000002
[25449.031230][ctx 0x10000000]		render_targets[3] = 0x00000003
[25449.031230][ctx 0x10000000]		render_targets[4] = 0x00000004
[25449.031230][ctx 0x10000000]		render_targets[5] = 0x00000005
[25449.031231][ctx 0x10000000]		render_targets[6] = 0x00000006
[25449.031231][ctx 0x10000000]		render_targets[7] = 0x00000007
[25449.031231][ctx 0x10000000]		render_targets[8] = 0x00000008
[25449.031232][ctx 0x10000000]		render_targets[9] = 0x00000009
[25449.031232][ctx 0x10000000]		render_targets[10] = 0x0000000a
[25449.031232][ctx 0x10000000]		render_targets[11] = 0x0000000b
[25449.031232][ctx 0x10000000]		render_targets[12] = 0x0000000c
[25449.031233][ctx 0x10000000]		render_targets[13] = 0x0000000d
[25449.031233][ctx 0x10000000]		render_targets[14] = 0x0000000e
[... 377 lines cut ...]
[25449.030482][ctx 0x10000000]	context = 0x10000000
[25449.030483][ctx 0x10000000]	render_targets = 0x00000001
[25449.030976][ctx       none]=========vaEndPicture ret = VA_STATUS_SUCCESS, success (no error) 
[25449.030986][ctx       none]=========vaDestroyBuffer ret = VA_STATUS_SUCCESS, success (no error) 
[25449.030987][ctx       none]=========vaDestroyBuffer ret = VA_STATUS_SUCCESS, success (no error) 
[25449.030988][ctx       none]=========vaDestroyBuffer ret = VA_STATUS_SUCCESS, success (no error) 
[25449.030988][ctx       none]=========vaDestroyBuffer ret = VA_STATUS_SUCCESS, success (no error) 
[25449.032257][ctx       none]==========va_TracePutSurface
[25449.032259][ctx       none]	surface = 0x00000001
[25449.032260][ctx       none]	draw = 0x0x2a00001
[25449.032260][ctx       none]	srcx = 0
[25449.032261][ctx       none]	srcy = 0
[25449.032261][ctx       none]	srcw = 640
[25449.032261][ctx       none]	srch = 360
[25449.032262][ctx       none]	destx = 0
[25449.032262][ctx       none]	desty = 0
[25449.032263][ctx       none]	destw = 472
[25449.032263][ctx       none]	desth = 359
[25449.032264][ctx       none]	cliprects = 0x(nil)
[25449.032264][ctx       none]	number_cliprects = 0
[25449.032264][ctx       none]	flags = 0x00000000

```

seems no error occurs?

canyuedao, 2021-03-18:
> suppose you are not using X11, right?

I kown why it crashes:

```
libva info: Open new log file 1.163132.thd-0x00007801 for the thread 0x00007801
libva info: LIBVA_TRACE is on, save log into 1.163132.thd-0x00007801
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva info: va_openDriver() returns 0
libva info: Save context 0x10000000 into log file 1.163130.thd-0x000077e5
[New Thread 0x7fff54086700 (LWP 30723)]
libva info: Save context 0x10000000 into log file 1.163132.thd-0x00007801
[New Thread 0x7fff54045700 (LWP 30724)]
[Switching to Thread 0x7ffe1d321700 (LWP 30693)]

Breakpoint 1, va_dri_get_drawable (ctx=0x7ffde8011300, drawable=44040193) at va_dricommon.c:122
122	    return do_drawable_hash(ctx, drawable);
Missing separate debuginfos, use: debuginfo-install alsa-lib-1.1.8-1.el7.x86_64 bzip2-libs-1.0.6-13.el7.x86_64 cairo-1.15.12-4.el7.x86_64 dbus-libs-1.10.24-15.el7.x86_64 elfutils-libelf-0.176-5.el7.x86_64 elfutils-libs-0.176-5.el7.x86_64 expat-2.1.0-12.el7.x86_64 fontconfig-2.13.0-4.3.el7.x86_64 freetype-2.8-14.el7_9.1.x86_64 glib2-2.56.1-8.el7.x86_64 glibc-2.17-323.el7_9.x86_64 keyutils-libs-1.5.8-3.el7.x86_64 krb5-libs-1.15.1-50.el7.x86_64 libX11-1.6.7-3.el7_9.x86_64 libXau-1.0.8-2.1.el7.x86_64 libXcursor-1.1.15-1.el7.x86_64 libXdamage-1.1.4-4.1.el7.x86_64 libXext-1.3.3-3.el7.x86_64 libXfixes-5.0.3-1.el7.x86_64 libXrandr-1.5.1-2.el7.x86_64 libXrender-0.9.10-1.el7.x86_64 libXv-1.0.11-1.el7.x86_64 libXxf86vm-1.1.4-1.el7.x86_64 libattr-2.4.46-13.el7.x86_64 libcap-2.22-11.el7.x86_64 libcom_err-1.42.9-19.el7.x86_64 libdrm-2.4.97-2.el7.x86_64 libgcc-4.8.5-44.el7.x86_64 libgcrypt-1.5.3-14.el7.x86_64 libglvnd-1.0.1-0.8.git5baa1e5.el7.x86_64 libglvnd-egl-1.0.1-0.8.git5baa1e5.el7.x86_64 libglvnd-glx-1.0.1-0.8.git5baa1e5.el7.x86_64 libgomp-4.8.5-44.el7.x86_64 libgpg-error-1.12-3.el7.x86_64 libpng-1.5.13-8.el7.x86_64 libpng12-1.2.50-10.el7.x86_64 libselinux-2.5-15.el7.x86_64 libstdc++-4.8.5-44.el7.x86_64 libuuid-2.23.2-65.el7_9.1.x86_64 libxcb-1.13-1.el7.x86_64 libxkbcommon-0.7.1-3.el7.x86_64 libxkbcommon-x11-0.7.1-3.el7.x86_64 libxshmfence-1.2-1.el7.x86_64 llvm-private-7.0.1-1.el7.x86_64 lz4-1.8.3-1.el7.x86_64 mesa-dri-drivers-18.3.4-12.el7_9.x86_64 mesa-libGL-18.3.4-12.el7_9.x86_64 mesa-libglapi-18.3.4-12.el7_9.x86_64 openssl-libs-1.0.2k-21.el7_9.x86_64 pcre-8.32-17.el7.x86_64 pixman-0.34.0-1.el7.x86_64 sqlite-3.7.17-8.el7_7.1.x86_64 systemd-libs-219-78.el7_9.3.x86_64 xz-libs-5.2.2-1.el7.x86_64 zlib-1.2.7-19.el7_9.x86_64
(gdb) s
do_drawable_hash (drawable=44040193, ctx=0x7ffde8011300) at va_dricommon.c:66
66	    struct dri_drawable *dri_drawable = dri_state->drawable_hash[index];
(gdb) n
68	    while (dri_drawable) {
(gdb) p dri_drawable
$1 = (struct dri_drawable *) 0x0
(gdb) n
122	    return do_drawable_hash(ctx, drawable);
(gdb) s
do_drawable_hash (drawable=44040193, ctx=0x7ffde8011300) at va_dricommon.c:74
74	    dri_drawable = dri_state->createDrawable(ctx, drawable);
(gdb) p dri_drawable
$2 = (struct dri_drawable *) 0x0
(gdb) p dri_state 
$3 = (struct dri_state *) 0x7ffde80114f0
(gdb) p *dri_state 
$4 = {base = {fd = 97, auth_type = 0, va_reserved = {0, 0, 0, 0, 0, 0, 0, 0}}, drawable_hash = {0x0 <repeats 32 times>}, createDrawable = 0x0, destroyDrawable = 0x0, swapBuffer = 0x0, getRenderingBuffer = 0x0, close = 0x0, 
  va_reserved = {0 <repeats 16 times>}}
```
the `createDrawable` func is NULL. 
So how to solve this problem?  I am using `libva-2.10.0` and `intel-media-20.4.5`.

canyuedao, 2021-03-19:
> suppose you are not using X11, right?
 
@XinfengZhang  I tracked the prolem, here is the stack trace:
```
#0  VA_DRI2Connect () at va_dri2.c:170
#1  0x00007fffdece2d2b in va_isDRI2Connected (ctx=0x7ffdfc011300, driver_name=driver_name@entry=0x7ffe21c3a8e8) at dri2_util.c:193
#2  0x00007fffdece454e in va_DRI2_GetNumCandidates (pDisplayContext=<optimized out>, num_candidates=0x7ffe21c3a944) at va_x11.c:101
#3  va_DisplayContextGetNumCandidates (pDisplayContext=<optimized out>, num_candidates=0x7ffe21c3a944) at va_x11.c:241
#4  0x00007fffdeabd65f in va_getDriverNumCandidates (num_candidates=0x7ffe21c3a944, dpy=0x7ffdfc011190) at va.c:359
#5  vaInitialize (dpy=0x7ffdfc011190, major_version=0x7ffdfc000ca8, minor_version=0x7ffdfc000cac) at va.c:724
#6  0x00007fffeaadeae4 in Open(vlc_va_t*, int) () from /usr/jstnvr/libnvvapi.so
#7  0x00007fffeaadfbe9 in vlc_va_New () from /usr/jstnvr/libnvvapi.so
#8  0x00007ffff77e4262 in CFFVideo::OpenDecode(enum_streamType, unsigned char*, int, int, int) () from /usr/jstnvr/libNVPlayM4.so.1
#9  0x00007ffff77ee44a in CTunnelStream::CreateDecoder(int, enum_streamType, unsigned char*, int, int, int) () from /usr/jstnvr/libNVPlayM4.so.1
#10 0x00007ffff77f0b98 in CTunnelStream::playRealStream() () from /usr/jstnvr/libNVPlayM4.so.1
#11 0x00007ffff77f0f88 in CTunnelStream::singleThreadRun() () from /usr/jstnvr/libNVPlayM4.so.1
#12 0x00007ffff6f109ea in nmnvrlib::NVRSingleThread::singleThreadStreamFunc(void*) () from /usr/jstnvr/libnvrlib.so.1
#13 0x00007ffff6f122c7 in tp_work_thread () from /usr/jstnvr/libnvrlib.so.1
#14 0x00007ffff0065ea5 in start_thread () from /lib64/libpthread.so.0
#15 0x00007fffef56e9fd in clone () from /lib64/libc.so.6
```

I find in `va_dri2.c` file, `VA_DRI2Connect` function:
```
Bool VA_DRI2Connect(Display *dpy, XID window,
		 char **driverName, char **deviceName)
{
    XExtDisplayInfo *info = DRI2FindDisplay(dpy);
    xDRI2ConnectReply rep;
    xDRI2ConnectReq *req;

    XextCheckExtension (dpy, info, va_dri2ExtensionName, False);

    LockDisplay(dpy);
    GetReq(DRI2Connect, req);
    req->reqType = info->codes->major_opcode;
    req->dri2ReqType = X_DRI2Connect;
    req->window = window;
    req->driverType = DRI2DriverDRI;
    if (!_XReply(dpy, (xReply *)&rep, 0, xFalse)) {
	UnlockDisplay(dpy);
	SyncHandle();
	return False;
    }

    if (rep.driverNameLength == 0 && rep.deviceNameLength == 0) {
	UnlockDisplay(dpy);
	SyncHandle();
	return False;
    }

    *driverName = Xmalloc(rep.driverNameLength + 1);
    if (*driverName == NULL) {
	_XEatData(dpy, 
		  ((rep.driverNameLength + 3) & ~3) +
		  ((rep.deviceNameLength + 3) & ~3));
	UnlockDisplay(dpy);
	SyncHandle();
	return False;
    }
    _XReadPad(dpy, *driverName, rep.driverNameLength);
    (*driverName)[rep.driverNameLength] = '\0';

    *deviceName = Xmalloc(rep.deviceNameLength + 1);
    if (*deviceName == NULL) {
	Xfree(*driverName);
	_XEatData(dpy, ((rep.deviceNameLength + 3) & ~3));
	UnlockDisplay(dpy);
	SyncHandle();
	return False;
    }
    _XReadPad(dpy, *deviceName, rep.deviceNameLength);
    (*deviceName)[rep.deviceNameLength] = '\0';

    UnlockDisplay(dpy);
    SyncHandle();

    return True;
}
```

It return `False` from here:
```
    if (rep.driverNameLength == 0 && rep.deviceNameLength == 0) {
	UnlockDisplay(dpy);
	SyncHandle();
	return False;
    }
```

So I guess there is something wrong with my X setting.
How can I check where the problem is?

XinfengZhang, 2023-05-08:
@canyuedao , suppose it was fixed by https://github.com/intel/libva/pull/537, I close it, any concern , feel free to reopen it.