intel/media-driver #1159: HDR Tonemapping doesn't seem to be enabled for Apollo Lake

StableNarwhal, 2021-03-09:
It appears to have been added to [Kabylake](https://github.com/intel/media-driver/commit/15038615da2e5714ef86ebbbe752692b53acac84) and [Gemini Lake](https://github.com/intel/media-driver/commit/15038615da2e5714ef86ebbbe752692b53acac84).

Is there a specific reason it hasn't been added for Apollo Lake?

softworkz, 2021-03-09:
Apollo Lake has the graphics from SkyLake and AFAIK, KabyLake is the minimum platform supporting HDHR tone mapping.

StableNarwhal, 2021-03-09:
Oh, I was assuming that APL would support HDR tone mapping. Thank you for the quick response!

softworkz, 2021-03-09:
It was accidental as I had thought this would be one of our repositories. But the answer should be accurate 😆

XinfengZhang, 2021-03-15:
yes, the answer is accurate. thanks

closed 2021-03-15 by XinfengZhang