intel/media-driver #1165: Coffee Lake: iHD driver fails on Ubuntu 20.04

haiyang1992, 2021-03-18:
I have a i3-8100T system running Proxmox and Ubuntu 20.04 as a headless VM. I have configured GPU passthrough.
Inside the Ubuntu VM:
`$ sudo lshw -C video
  *-display
       description: VGA compatible controller
       product: 8th Gen Core Processor Gaussian Mixture Model
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:01:00.0
       version: 00
       width: 64 bits
       clock: 33MHz
       capabilities: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       resources: irq:45 memory:fb000000-fbffffff memory:e0000000-efffffff ioport:5000(size=64) memory:c0000-dffff`

`$ lspci -v | grep VGA
01:00.0 VGA compatible controller: Intel Corporation 8th Gen Core Processor Gaussian Mixture Model (prog-if 00 [VGA controller])`

`$ ls -al /dev/dri
total 0
drwxr-xr-x  3 root root        100 Mar 18 10:28 .
drwxr-xr-x 20 root root       4200 Mar 18 11:40 ..
drwxr-xr-x  2 root root         80 Mar 18 10:28 by-path
crw-rw----  1 root video  226,   0 Mar 18 10:28 card0
crw-rw----  1 root render 226, 128 Mar 18 10:28 renderD128`

I also have the latest intel-media-va-driver-non-free installed from the intel repo:
`$ apt-cache policy intel-media-va-driver-non-free
intel-media-va-driver-non-free:
  Installed: 21.1.1+i509~u20.04
  Candidate: 21.1.1+i509~u20.04
  Version table:
 *** 21.1.1+i509~u20.04 500
        500 https://repositories.intel.com/graphics/ubuntu focal/main amd64 Packages
        100 /var/lib/dpkg/status`

Running `vainfo` though, gives me the following error:
`$ vainfo
error: can't connect to X server!
libva info: VA-API version 1.10.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit`

If I fall back on the i965 driver it seems to be working:
`$ vainfo
error: can't connect to X server!
libva info: VA-API version 1.10.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.10 (libva 2.10.0)
vainfo: Driver version: Intel i965 driver for Intel(R) Coffee Lake - 2.4.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileH264MultiviewHigh      : VAEntrypointVLD
      VAProfileH264MultiviewHigh      : VAEntrypointEncSlice
      VAProfileH264StereoHigh         : VAEntrypointVLD
      VAProfileH264StereoHigh         : VAEntrypointEncSlice
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointEncSlice
      VAProfileVP9Profile2            : VAEntrypointVLD`

Attached is [strace.log](https://github.com/intel/media-driver/files/6165378/strace.log) running `strace vainfo`.

I'm not sure if this is a bug or a configuration issue on my end. Help would be appreciated!

XinfengZhang, 2021-03-25:
mos_gem_context_create call failed, @wyc55069407 could you help to take a look?

XinfengZhang, 2021-03-26:
and seems i965 has no DRM_IOCTL_I915_GEM_CONTEXT_CREATE call

haiyang1992, 2021-04-04:
I wonder if the following log from dmesg helps

```
[  +0.000001] fb0: switching to inteldrmfb from EFI VGA
[  +0.155975] Console: switching to colour dummy device 80x25
[  +0.000028] i915 0000:01:00.0: vgaarb: deactivate vga console
[  +0.013406] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[  +0.000003] [drm] Driver supports precise vblank timestamp query.
[  +0.000012] i915 0000:01:00.0: No more image in the PCI ROM
[  +0.000184] [drm] Failed to find VBIOS tables (VBT)
[  +0.000240] i915 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[  +0.000422] [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[  +0.106079] ata2: SATA link down (SStatus 0 SControl 300)
[  +0.005218] ata1: SATA link down (SStatus 0 SControl 300)
[  +0.000132] ata3: SATA link down (SStatus 0 SControl 300)
[  +0.007243] ata6: SATA link down (SStatus 0 SControl 300)
[  +0.003086] ata5: SATA link down (SStatus 0 SControl 300)
[  +0.000164] ata4: SATA link down (SStatus 0 SControl 300)
[  +0.484774] [drm] failed to retrieve link info, disabling eDP
[  +0.002709] i915 0000:01:00.0: Failed to program MOCS registers; expect performance issues.
[  +0.202442] i915 0000:01:00.0: Failed to idle engines, declaring wedged!
[  +0.036067] i915 0000:01:00.0: Failed to initialize GPU, declaring it wedged!
[  +0.000003] i915 0000:01:00.0: Please file a bug on drm/i915; see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
[  +0.033810] [drm] Initialized i915 1.6.0 20190822 for 0000:01:00.0 on minor 0
[  +0.058294] [drm] Cannot find any crtc or sizes
[  +0.067748] [drm] Cannot find any crtc or sizes
[  +0.049304] [drm] Cannot find any crtc or sizes
```

haiyang1992, 2021-04-04:
It's working. I turned on the `video=efifb:off` option in GRUB on the host OS. Closing this now.

closed 2021-04-04 by haiyang1992