intel/media-driver #1170: Compile failed with 21.1.3

heitbaum, 2021-04-01:
See logs: http://paste.ubuntu.com/p/BpxqPsHDkv/

build with 21.1.2 was fine.

../media_driver/linux/gen12/ddi/media_libva_caps_g12.cpp:2542:47: error: 'IGFX_ALDERLAKE_P' was not declared in this scope; did you mean 'IGFX_ALDERLAKE_S'?
 2542 |     RegisterCaps<MediaLibvaCapsG12>((uint32_t)IGFX_ALDERLAKE_P);
      |                                               ^~~~~~~~~~~~~~~~
      |                                               IGFX_ALDERLAKE_S

XinfengZhang, 2021-04-01:
have you updated gmmlib? the definition is from latest gmmlib

heitbaum, 2021-04-01:
Closing my own ticket. 
Gmmlib 21.1.1 required for media-driver 21.1.3 (was not required for 21.1.2j

closed 2021-04-01 by heitbaum