intel/media-driver #1171: HEVCe FEI gets 0 qp in output

egorovdanil, 2021-04-01:
Reproduce Steps:

1. sample_hevc_fei -i test_stream_176x96.yuv -o 001.hevc -n 100 -w 176 -h 96 -qp 24 -g 1 -encode -EncodedOrder
2. hevc_fei_extractor 001.hevc -multi_pak_str 001.multipak
3. asg-hevc -generate -gen_repack_ctrl -n 100 -w 176 -h 96 -g 1 -repack_ctrl_file 001.repakctrl -repack_str_file 001.multipak -InitialQP 24 -DeltaQP 1 1 2 2 3 3 4 4
4. sample_hevc_fei -i test_stream_176x96.yuv -o 0007.repack -n 100 -w 176 -h 96 -qp 24 -g 1 -encode -EncodedOrder -repackctrl 001.repakctrl -repackstat 0007.repakstat

At step 4, we get 0 qp, while the incoming was 24, in [hevcehw_base_fei_lin.cpp#L378](https://github.com/Intel-Media-SDK/MediaSDK/blob/c4fbaedd8a827ec36ee312e978e993d3f938201c/_studio/mfx_lib/encode_hw/hevc/linux/base/hevcehw_base_fei_lin.cpp#L378)
`mfxU8 QP = fb.status & VA_CODED_BUF_STATUS_PICTURE_AVE_QP_MASK;`

XinfengZhang, 2021-04-03:
suppose the issue is caused by https://github.com/intel/media-driver/commit/b5e5aedd27ecdd71e9f4b03b784ee0bf85126a73?
could you help to have a check?

egorovdanil, 2021-04-05:
@XinfengZhang, you are right, issue is reproducible since `b5e5aed`.

XinfengZhang, 2021-06-13:
suppose it already been fixed by 1f25be0a, could you help to check?

egorovdanil, 2021-07-16:
@XinfengZhang, yes, the problem is gone. Thanks!

closed 2021-07-16 by egorovdanil