intel/media-driver #1172: VP9 10-bit videos unplayable on GLK

smp79, 2021-04-01:
[Sample video 1](https://mega.nz/file/hJdFEIBI#uSOjZtkkjIVYSfqD9aSfONf1yq__uQvlsf47pCtFvdQ)
[Sample video 2](https://mega.nz/#!pQEGgRwY!pD9whIlM-U9tJIA-LojxSt582BAZGfdSA5wAQLT06I4)

Both of those samples play fine with intel-vaapi-driver but not with media-driver.

Tested on J4105, Kodi 19, LibreELEC.

wangyan-intel, 2021-04-06:
@Jexu Could you please take a look? Thanks.

Jexu, 2021-04-06:
@smp79 
sorry i cannot access to above video, can you share it to me.
by the way, what app do you use or how do you play the video?

smp79, 2021-04-06:
https://www.dropbox.com/s/8xwgxdyo60um9xf/The%20World%20in%20HDR.mkv?dl=1

I use Kodi to play it.

Jexu, 2021-04-07:
can you try ffmpeg with bellow cmdline:
    ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -i "**glk-vp9-not-work.mkv**" -pix_fmt yuv420p -f rawvideo -vsync passthrough -vframes 10 -y output.yuv

ffmpeg can decode above video on my GLK machine with media-driver. and can you also help confirm if Kodi really supports iHD driver?

smp79, 2021-04-07:
```
LibreELEC:~ # ffmpeg -hwaccel vaapi -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_flags allow_profile_mismatch -filter_hw_device hw -v verbose -i "glk-vp9-not-work.mkv" -pix_fmt yuv420p -f rawvideo -vsync passthrough -vframes 10
-y output.yuv
ffmpeg version 4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 10.2.0 (GCC)
  configuration: --enable-ffmpeg --disable-ffplay --enable-ffprobe --enable-static --disable-shared --enable-gpl --disable-doc --enable-encoder=h264_vaapi --enable-encoder=hevc_vaapi --enable-encoder=mjpeg_vaapi --enable-encoder=mpeg2_vaapi --enable-encoder=vp8_vaapi --enable-encoder=vp9_vaapi --disable-encoder=h264_nvenc --disable-encoder=hevc_nvenc --enable-hwaccel=h263_vaapi --enable-hwaccel=h264_vaapi --enable-hwaccel=hevc_vaapi --enable-hwaccel=mjpeg_vaapi --enable-hwaccel=mpeg2_vaapi --enable-hwaccel=mpeg4_vaapi --enable-hwaccel=vc1_vaapi --enable-hwaccel=vp8_vaapi --enable-hwaccel=vp9_vaapi --enable-hwaccel=wmv3_vaapi --enable-avresample --disable-lzma --disable-alsa --arch=x86_64 --cpu=goldmont-plus --cross-prefix=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu- --enable-cross-compile --sysroot=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot --sysinclude=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/include --target-os=linux --nm=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-nm --ar=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-ar --as=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc --cc=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc --ld=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc --pkg-config=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/pkg-config --host-cc=/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/bin/host-gcc --host-cflags='-march=native -O2 -Wall -pipe -I/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/include -Wno-format-security' --host-ldflags='-Wl,-rpath,/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/lib -L/home/a/LE/LibreELEC.tv/build.LibreELEC-Generic.x86_64-10.0-devel/toolchain/lib' --host-extralibs=-lm --extra-cflags='-march=goldmont-plus -Wall -pipe -O2 -fomit-frame-pointer' --extra-ldflags='-march=goldmont-plus -Wl,--as-needed' --extra-libs= --enable-pic --enable-gnutls --disable-openssl --disable-hardcoded-tables
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[AVHWDeviceContext @ 0x3884600] libva: VA-API version 1.11.0
[AVHWDeviceContext @ 0x3884600] libva: Trying to open /usr/lib/dri/iHD_drv_video.so
[AVHWDeviceContext @ 0x3884600] libva: Found init function __vaDriverInit_1_11
[AVHWDeviceContext @ 0x3884600] libva: va_openDriver() returns 0
[AVHWDeviceContext @ 0x3884600] Initialised VAAPI connection: version 1.11
[AVHWDeviceContext @ 0x3884600] VAAPI driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.3 (06b63fd9bf).
[AVHWDeviceContext @ 0x3884600] Driver not found in known nonstandard list, using standard behaviour.
Input #0, matroska,webm, from 'glk-vp9-not-work.mkv':
  Metadata:
    encoder         : libebml v1.3.4 + libmatroska v1.4.5
    creation_time   : 2016-11-08T02:38:04.000000Z
  Duration: 00:02:34.57, start: 0.000000, bitrate: 18316 kb/s
    Stream #0:0(eng): Video: vp9 (Profile 2), 1 reference frame, yuv420p10le(tv, bt2020nc/bt2020/smpte2084), 3840x2160, SAR 1:1 DAR 16:9, 59.94 fps, 59.94 tbr, 1k tbn, 1k tbc (default)
    Metadata:
      BPS             : 18203235
      BPS-eng         : 18203235
      DURATION        : 00:02:34.555000000
      DURATION-eng    : 00:02:34.555000000
      NUMBER_OF_FRAMES: 9264
      NUMBER_OF_FRAMES-eng: 9264
      NUMBER_OF_BYTES : 351675124
      NUMBER_OF_BYTES-eng: 351675124
      _STATISTICS_WRITING_APP: mkvmerge v9.5.0 ('Quiet Fire') 64bit
      _STATISTICS_WRITING_APP-eng: mkvmerge v9.5.0 ('Quiet Fire') 64bit
      _STATISTICS_WRITING_DATE_UTC: 2016-11-08 02:38:04
      _STATISTICS_WRITING_DATE_UTC-eng: 2016-11-08 02:38:04
      _STATISTICS_TAGS: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES
      _STATISTICS_TAGS-eng: BPS DURATION NUMBER_OF_FRAMES NUMBER_OF_BYTES
    Side data:
      Mastering Display Metadata, has_primaries:1 has_luminance:1 r(0.6800,0.3200) g(0.2649,0.6900) b(0.1500 0.0600) wp(0.3127, 0.3127) min_luminance=0.001000, max_luminance=1000.000000
    Stream #0:1(eng): Audio: vorbis, 44100 Hz, stereo, fltp (default)
    Metadata:
      BPS             : 109558
      BPS-eng         : 109558
      DURATION        : 00:02:34.569000000
      DURATION-eng    : 00:02:34.569000000
      NUMBER_OF_FRAMES: 7450
      NUMBER_OF_FRAMES-eng: 7450
      NUMBER_OF_BYTES : 2116797
      NUMBER_OF_BYTES-eng: 2116797
[... cut ...]
  Total: 15 packets (1176030 bytes) demuxed
Output file #0 (output.yuv):
  Output stream #0:0 (video): 10 frames encoded; 10 packets muxed (124416000 bytes);
  Total: 10 packets (124416000 bytes) muxed
[AVIOContext @ 0x38f24c0] Statistics: 0 seeks, 475 writeouts
[AVIOContext @ 0x38df8c0] Statistics: 1209933 bytes read, 2 seeks
```

> can you also help confirm if Kodi really supports iHD driver?

Yes, Kodi really supports iHD driver. All supported formats are HW-decoded. The issue is only with VP9 10-bit (VP9 8-bit works fine).

Jexu, 2021-04-08:
media-driver supports vp9 10bit decode, and ffmpeg decoding with iHD looks good on your machine as well.
the error seems come from Kodi, can you report issue to Kodi.

you can also enable libva trace to get trace log in media-driver by bellow steps:

enable libva tracing with (substitute with the real user, please):

mkdir /home/<user>/trace
echo "LIBVA_TRACE=/home/<user>/trace/file" | sudo tee /etc/libva.conf
Mind that having /file on the very end of LIBVA_TRACE is not a mistake. Libva will actually create a bunch of files with the template /home/<user>/trace/file*

Once done, please, rerun your application (make it a single run to avoid mess in the logs), get all the trace files from /home/<user>/trace (there likely will be more than one if libva/driver is used by multiple threads) and send them back to us.

Note: remove /etc/libva.conf once you will be done.

smp79, 2021-04-26:
It does seem to be a Kodi issue. Sorry for the noise.

closed 2021-04-26 by smp79