intel/media-driver #1173: Baseline (66) profile is not supported, even though Main and High are

marmistrz, 2021-04-04:
I'm running Arch Linux, Intel Media Driver 21.1.2-1. My hardware is Intel i5-8265U. The supported VA profiles are:

```
$  vainfo  | grep H264
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
```
In particular, Baseline support is missing. 

Some mobile devices are using the Baseline AVC profile to encode the videos, for instance the default LineageOS camera. Consequently, decoding of such videos cannot be hardware accelerated. Since Baseline should be a subset of Main and High, one would expect it to be supported on modern hardware.

Was Baseline support omitted on purpose or by accident?

XinfengZhang, 2021-06-13:
VAProfileH264Baseline is deprecated in libva by commit https://github.com/intel/libva/commit/6f69256f8ccc9a73c0b196ab77ac69ab1f4f33c2

VAProfileH264Baseline is special, it not a subset, I suppose what you are mentioning is constrained baseline profile. constrained baseline profile is subset of main, but baseline has additional feature than constrained baseline.

marmistrz, 2021-06-13:
Hi, I'm precisely mentioning the baseline profile, which is the one being used by the default LineageOS camera.

XinfengZhang, 2021-06-13:
Constrained Baseline Profile bitstreams are also considered to be Baseline Profile bitstreams, the profileID is both 66, but "constraint_set1_flag " is set to be 1.

 I dont think the baseline profile is used now, because as I know, most of HW or SW does not support FMO... 
at least, media-driver does not support it.

marmistrz, 2021-06-13:
I raised an issue in the LineageOS bugtracker: https://gitlab.com/LineageOS/issues/android/-/issues/3616