intel/media-driver #1174: Build failure

dbermond, 2021-04-07:
I'm getting the following build error on the git master branch (currently at commit 95d413e519a980f6a7178880ccbfedace092316c):

```
[100%] Built target iHD_drv_video_OBJ
make[2]: Entering directory '/build/intel-media-driver-git/src/build'
make[2]: Entering directory '/build/intel-media-driver-git/src/build'
make[2]: Leaving directory '/build/intel-media-driver-git/src/build'
make[2]: Leaving directory '/build/intel-media-driver-git/src/build'
make[2]: Entering directory '/build/intel-media-driver-git/src/build'
make[2]: Entering directory '/build/intel-media-driver-git/src/build'
[100%] Linking CXX shared library iHD_drv_video.so
[100%] Linking CXX static library iHD_drv_video.a
make[2]: Leaving directory '/build/intel-media-driver-git/src/build'
[100%] Built target iHD_drv_video_STATIC
make[2]: Leaving directory '/build/intel-media-driver-git/src/build'
[100%] Built target iHD_drv_video
make[2]: Entering directory '/build/intel-media-driver-git/src/build'
make[2]: Leaving directory '/build/intel-media-driver-git/src/build'
make[2]: Entering directory '/build/intel-media-driver-git/src/build'
Running devult...
./devult: symbol lookup error: ../libdrm_mock/libdrm_mock.so: undefined symbol: mos_vma_heap_init
make[2]: *** [media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/build.make:75: RunULT] Error 127
make[2]: Leaving directory '/build/intel-media-driver-git/src/build'
make[1]: *** [CMakeFiles/Makefile2:395: media_driver/linux/ult/ult_app/CMakeFiles/RunULT.dir/all] Error 2
make[1]: Leaving directory '/build/intel-media-driver-git/src/build'
make: *** [Makefile:156: all] Error 2
make: Leaving directory '/build/intel-media-driver-git/src/build'
```

A git bisect shows that this is caused by commit f7927753c5999861d42def7839ea5ed819d55199.

System information:

- OS: Arch Linux x86_64
- Compiler: gcc 10.2.0
- gmmlib: git master
- libva: git master

XinfengZhang, 2021-04-08:
assigned to @Jister

XinfengZhang, 2021-04-20:
the change was reverted by 1f91096ea0fb4971f25998ad96d969be31fc50ac, but we still need to check why there are build issue. 
from the message, it should related with the mock function.  if it could not be reproduced on ubuntu. it suppose because  some option prevent the call from libdrm_mock.so to the function in iHD. maybe we should add mos_vma.h & mos_vma.c to be a part of libdrm_mock.so

Jister, 2021-04-20:
@dbermond Could you help try #1185? Thanks

dbermond, 2021-04-20:
Thank you for dedicating time to this issue.

@Jister Applying #1185 causes CMake to fail for me:

```
CMake Error at media_driver/linux/ult/libdrm_mock/CMakeLists.txt:42 (add_library):
  Cannot find source file:

    /build/intel-media-driver-git/src/media-driver/media_driver/linux/common/os/mos_vma.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .h .hh .h++
  .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .ispc

CMake Error at media_driver/linux/ult/libdrm_mock/CMakeLists.txt:42 (add_library):
  No SOURCES given to target: drm_mock

CMake Generate step failed.  Build files cannot be regenerated correctly.
```

Now the clear master branch is building fine for me. Probably because the commit that caused the issue was reverted as mentioned by @XinfengZhang .

Jister, 2021-04-20:
@dbermond Could you help rebase to previous commit? Or just using the latest PR of #1185 to have a try to see whether the build issue fixed or not. Thanks.

XinfengZhang, 2021-04-21:
@dbermond , yes, we reverted it , but I suppose we will re-checkin the change in future, this commit is used to enable softpin , it is important because i915 will deprecate relocation support in future. it's why @Jister  ask your help to check whether the issue could be fixed basing on previous commit. you also could use "git fetch origin pull/1185/head:localbranch" to fetch the code of #1185 directly.

dbermond, 2021-04-21:
#1185 now works fine for me after the latest changes made on it by @Jister . Builds fine, thank you.

Jister, 2021-04-21:
@dbermond Thank you very much for your verification!

XinfengZhang, 2021-04-25:
thanks @dbermond , could we close the issue now, suppose there will be no such issue when we merge the code in future.

dbermond, 2021-04-25:
@XinfengZhang Sure. If something happens after the merge I can reopen.

Thanks to everyone that dedicated time to this issue.

Closing.

closed 2021-04-25 by dbermond