intel/media-driver #1175: vaPutSurface: Suffer GPU hung in VEBOX when AVC decoding and moving window of vaapisink

zcwang, 2021-04-08:
Because of internal issue **VIZ-16502** then crease new one for sighting it easily in media-driver.

For SKL/CFL/CML platforms, it suffers GUP hung in VEBOX when moving and resizing a window which is showing a video to some specified positions and sizes, kernel's log show GPU hang. It only happens with iHD driver. There is no issue in i965. Please kindly help to check it. 

```
...
i915 0000:00:02.0: [drm] Resetting vecs0 for preemption time out
i915 0000:00:02.0: [drm] qtdemux0:sink[2295] context reset due to GPU hang
i915 0000:00:02.0: [drm] GPU HANG: ecode 9:8:2cffbffd, in qtdemux0:sink [2295]
GPU hangs can indicate a bug anywhere in the entire gfx stack, including userspace.
Please file a _new_ bug report at https://gitlab.freedesktop.org/drm/intel/issues/new.
Please see https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs for details.
drm/i915 developers can t
[CML-kernel-03252021.log](https://github.com/intel/media-driver/files/6275257/CML-kernel-03252021.log)
hen reassign to the right component if it's not a kernel issue.
The GPU crash dump is required to analyze GPU hangs, so please always attach it.
GPU crash dump saved to /sys/class/drm/card0/error
...
```

1. Launch 1 channels of decoding with following command to test AVC decode (sample video in here) with normal window size but not maximum size (manually to click button in upper right corner of video window)
`“gst-launch -v videotestsrc ! vaapisink”`
2. Issue following commands by another terminal for resizing & moving vaapisink window repeatly many times (refer following as example; find the window ID of the gstreamer)
```
$ wmctrl -lG
0x00800001 0 2 40 1918 1055 N/A N/A
(The last field shows "N/A", and 0x00800001 in Hex is 8388609 in Dec)

#Change the window position and size by repeating following commands
export remote_camera_wid=8388609
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
sleep 2
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
sleep 2

```
Or via following scripts,
```
#!/bin/bash
export remote_camera_wid=$1
echo $remote_camera_wid
 
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
sleep 2
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
sleep 2
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
[CML-kernel-03252021.log](https://github.com/intel/media-driver/files/6275261/CML-kernel-03252021.log)

xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
sleep 2
xdotool windowmove $remote_camera_wid 960 0
xdotool windowsize $remote_camera_wid 960 1080
xdotool windowmove $remote_camera_wid 816 858
xdotool windowsize $remote_camera_wid 288 162
```

In DdiCodec_PutSurfaceLinuxHW, it might be root cause in line-425,
```
...
423     //Init source rectangle 
424     Rect_init(&srcRect, srcx, srcy, srcw, srch);
425     Rect_init(&dstRect, destx, desty, destw, desth); <-----------------------window sizing
426     
427     // Source Surface Information
428     Surf.Format                 = VpGetFormatFromMediaFormat(bufferObject->format);           // Surface format
429     Surf.SurfType               = SURF_IN_PRIMARY;       // Surface type (context)
430     Surf.SampleType             = SAMPLE_PROGRESSIVE;
431     Surf.ScalingMode            = VPHAL_SCALING_AVS;
432     
433     Surf.OsResource.Format      = VpGetFormatFromMediaFormat(bufferObject->format);
434     Surf.OsResource.iWidth      = bufferObject->iWidth;
435     Surf.OsResource.iHeight     = bufferObject->iHeight;
436     Surf.OsResource.iPitch      = bufferObject->iPitch;
437     Surf.OsResource.iCount      = 0;
438     Surf.OsResource.TileType    = VpGetTileTypeFromMediaTileType(bufferObject->TileType);
439     Surf.OsResource.bMapped     = bufferObject->bMapped;
440     Surf.OsResource.bo          = bufferObject->bo;
441     Surf.OsResource.pGmmResInfo = bufferObject->pGmmResourceInfo;
442     
443     Surf.dwWidth                = bufferObject->iWidth;
444     Surf.dwHeight               = bufferObject->iHeight;
445     Surf.dwPitch                = bufferObject->iPitch;
446     Surf.TileType               = VpGetTileTypeFromMediaTileType(bufferObject->TileType);
447     Surf.ColorSpace             = DdiVp_GetColorSpaceFromMediaFormat(bufferObject->format);
448     Surf.ExtendedGamut          = false;
449     Surf.rcSrc                  = srcRect;
450     Surf.rcDst                  = dstRect;
...
```

zcwang, 2021-04-08:
[CML-kernel-03252021.log](https://github.com/intel/media-driver/files/6275274/CML-kernel-03252021.log)

FurongZhang, 2021-04-20:
WIP.

FurongZhang, 2021-05-26:
I talked with @XinfengZhang  just now. He will double confirm since it is in DdiCodec_PutSurfaceLinuxHW.

PR #1199 (2021-06-03): Add area check in DdiMedia_PutSurface to avoid dst rectangle out of display region
fixes #1175
commit 0133e252 (2021-06-03): Add area check in DdiMedia_PutSurface to avoid dst rectangle out of display region

fixes #1175
closed 2021-06-03 by intel-mediadev