intel/media-driver #1177: vainfo only shows profiles when using sudo

satmandu, 2021-04-09:
```
% vainfo
libva info: VA-API version 1.11.0
libva info: User environment variable requested driver 'i965'
libva info: Trying to open /usr/local/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_10
libva error: /usr/local/lib64/dri/i965_drv_video.so init failed
libva info: va_openDriver() returns -1
vaInitialize failed with error code -1 (unknown libva error),exit
```
```
% sudo vainfo
error: XDG_RUNTIME_DIR not set in the environment.
Invalid MIT-MAGIC-COOKIE-1 keyerror: can't connect to X server!
libva info: VA-API version 1.11.0
libva info: Trying to open /usr/local/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_11
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.11 (libva 2.11.1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.1.3 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```
The user appears to be in the appropriate groups:
```
% ls -aFl /dev/dri
/
total 0
drwxr-xr-x  2 root root       160 Apr  8 14:09 ./
drwxr-xr-x 17 root root      3100 Apr  8 14:09 ../
crw-rw----  1 root video 226,   0 Apr  8 14:09 card0
crw-rw----  1 root video 226,   1 Apr  8 14:09 card1
crw-rw----  1 root video 226,  64 Apr  8 14:09 controlD64
crw-rw----  1 root video 226, 128 Apr  8 14:09 renderD128
crw-rw----  1 root video 226, 129 Apr  8 14:09 renderD129
lrwxrwxrwx  1 root root        10 Apr  8 14:09 vgem -> renderD129
 % id
uid=1000(chronos) gid=1000(chronos) groups=1000(chronos),7(lp),18(audio),27(video),208(pkcs11),222(input),240(brltty),303(policy-readers),304(fuse-drivefs),415(suzy-q),420(crash-user-access),600(cras),601(wayland),602(arc-bridge),603(arc-camera),608(midis),1001(chronos-access),20162(traced-producer),20164(traced-consumer),665357(android-everybody)
```

satmandu, 2021-04-09:
Curiously running this as a regular user works:
```
# vainfo --display drm --device /dev/dri/renderD128
libva info: VA-API version 1.11.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_11
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.11 (libva 2.11.1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.1.3 ()
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

satmandu, 2021-04-09:
Seems like this could be related to https://github.com/intel/libva/issues/221

XinfengZhang, 2021-06-13:
the original issue should be caused by env variables, from the log you showed here:
libva info: **User environment variable requested driver 'i965'**
libva info: Trying to open /usr/local/lib64/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_10

but other command when it success, it is 
**Trying to open /usr/local/lib64/dri/iHD_drv_video.so**

XinfengZhang, 2021-07-13:
@satmandu could we close it, it caused by different driver loaded. 
libva info: User environment variable requested driver 'i965'
libva info: User environment variable requested driver 'iHD'

closed 2021-07-13 by satmandu