intel/media-driver #1178: [ffmpeg-vaapi][regression]one commit of media-driver will cause ffmpeg-vaapi jpeg encode fail

zhangyuankun-star, 2021-04-12:
the commit c25a2915b66faf3481af3888f413ab2891e9a021 in media-driver will cause ffmpeg-vaapi jpeg encode fail.
**cmd-line**
ffmpeg -hwaccel vaapi -vaapi_device /dev/dri/renderD128 -v verbose -f rawvideo -pix_fmt yuv420p -s:v ${width}x${height} -i input.yuv -vf 'format=nv12,hwupload' -c:v mjpeg_vaapi -global_quality 10 -vframes 10 -f null -
**error log**:
[LIBVA]:CRITICAL - RenderPicture:207: not supported buffer type.
[LIBVA]:CRITICAL - QmatrixFromHeader:756: Invalid QTable size in header
[LIBVA]:CRITICAL - EncodeInCodecHal:520: Failed to parse Quant Matrix from header [18].
[LIBVA]:CRITICAL - EndPicture:81: DDI:DdiEncode_EncodeInCodecHal return failure.
[mjpeg_vaapi @ 0x5604c488e080] Failed to end picture encode issue: 24 (internal encoding error).
[mjpeg_vaapi @ 0x5604c488e080] Encode failed: -5.
Video encoding failed

FocusLuo, 2021-04-12:
Comment: this media-driver commit info is c25a2915b66f

uartie, 2021-04-12:
This also breaks ffmpeg-qsv JPEGe.

XinfengZhang, 2021-04-13:
@ivan-losev is working on this issue.

uartie, 2021-04-13:
> This also breaks ffmpeg-qsv JPEGe.

Nevermind.  The ffmpeg-qsv JPEGe regression was caused by ffmpeg commit that landed after this driver regression ( https://trac.ffmpeg.org/ticket/9186).

With ffmpeg fixed again, only ffmpeg-vaapi JPEGe appears to be regressed by this driver commit.

uartie, 2021-04-14:
> @ivan-losev is working on this issue.

It doesn't hurt to just revert the commit, immediately, until a proper solution can be provided.  You never know how long a rework can take.  So reverting is the best course of immediate action for regressions, IMO, so your not forcing everyone/everything to workaround it in their daily workflow.

XinfengZhang, 2021-04-18:
@hanlong1 lets revert the commit firstly, after @ivan-losev's change, we could merge it again. thanks

uartie, 2021-04-19:
Reverted in eab77ad484cb.  Reworked in e0c41c11711e.  The rework does not regress ffmpeg-vaapi.  However, now gstreamer-vaapi is broken as reported in #1183

XinfengZhang, 2021-06-13:
#1183 also was closed by reverting the regression commit.

closed 2021-06-13 by XinfengZhang