intel/media-driver #1179: sample encode h265 with intra refresh have GPU hang

yskuridi, 2021-04-14:
platform GEN 9
MediaSDK 77f673853c23a91241
Driver: Intel iHD driver for Intel(R) Gen Graphics - 21.1.3-6097 (e5b8ffd)

command line:
`sample_encode h265 -i 1920x1080.yuv -o sample_output.265 -ir_type 1 -ir_cycle_size 4 -ir_qp_delta -20 -ir_cycle_dist 20 -w 1920 -h 1080 -r 1`

`./mfx_transcoder h265 -i 1920x1072_250.yuv -o output.265 -w 1920 -h 1072 -f 25 -b 600000 -u 4 -hw -d3d -s 0 -async 1 -async_depth 1 -IntRefType 1 -IntRefCycleSize 20 -IntRefCycleDist 20 -n 40 -IntRefQPDelta -20 -r 1`

error:
```
ERROR: Hardware device was lost or returned an unexpected error. Recovering...

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), SynchronizeFirstTask, SyncOperation failed at /home/wenbin/software/media/src/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:168

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), GetFreeTask, m_TaskPool.SynchronizeFirstTask failed at /home/wenbin/software/media/src/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:2193

[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), Run, m_pmfxENC->EncodeFrameAsync failed at /home/wenbin/software/media/src/MediaSDK/samples/sample_encode/src/pipeline_encode.cpp:2497
```

wenbinc-Bin, 2021-07-19:
@Xiaogangli-intel Any progress on this issue?
Thanks
Wenbin

Xiaogangli-intel, 2021-11-09:
@wenbinc-Bin Won't this issue be fixed by [https://github.com/Intel-Media-SDK/MediaSDK/pull/2667](url)?

wenbinc-Bin, 2021-11-09:
@Xiaogangli-intel Yes, this issue is fixed. Now this issue can be closed.

Sherry-Lin, 2021-12-08:
Close this issue which has been fixed.

closed 2021-12-08 by Sherry-Lin