intel/media-driver #1181: List for supported format for encoding requires update

djie1, 2021-04-16:
1. In the documentation, for example, https://github.com/intel/media-driver/blob/master/docs/media_features.md#hardware-encoding-low-power-encodingvdenchuc, the supported format list does not comply with that in the code: https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.h#L52 . For example, RGB32 is not listed in the documentation.
2. on Gen12, for DualPipe HEVC encoding path, seems RGB32 is also supported, but it's not listed in the code: https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_libva_caps_g12.h#L50

XinfengZhang, 2021-05-17:
will update the document , thanks @djie1

Xiaogangli-intel, 2022-04-25:
#1393 will fix this issue. And for  DualPipe HEVC, RGB is not supported.

closed 2022-04-27 by intel-mediadev

djie1, 2022-04-28:
>  for DualPipe HEVC, RGB is not supported.
 I tried RGB32 input, looks like it's supported for DualPipe HEVC encoding.

Xiaogangli-intel, 2022-04-28:
> > for DualPipe HEVC, RGB is not supported.
> 
> I tried RGB32 input, looks like it's supported for DualPipe HEVC encoding.

:) Yes, but we cannot ensure it works properly.

commit 798cafe2 (2022-04-27): Update media feature table

1. Correct supporting formats for Gen12
2. Add note for max resolution for DG2/ATSM
3. fixes #1181