intel/media-driver #1190: Corrupted HEVC encoded files on Tiger Lake

angryziber, 2021-04-28:
Ubuntu 21.04
intel-media-va-driver-non-free 21.1.1+ds1-1
ffmpeg         7:4.3.2-0+deb11u1ubuntu1

When encoding a video to HEVC with ffmpeg & vaapi, no errors are give, but the output is corrupted:

```
ffmpeg -hide_banner -hwaccel vaapi -hwaccel_output_format vaapi -vaapi_device /dev/dri/renderD128 -i $FILE -c:v hevc_vaapi -qp 27 -rc_mode ICQ -c:a copy $OUT
```

output:

```
[graph 0 input from stream 0:0 @ 0x556208026540] w:2720 h:1530 pixfmt:vaapi_vld tb:1/30000 fr:30000/1001 sar:1/1
[hevc_vaapi @ 0x5562070bb0c0] Using input frames context (format vaapi_vld) with hevc_vaapi encoder.
[hevc_vaapi @ 0x5562070bb0c0] Input surface format is nv12.
[hevc_vaapi @ 0x5562070bb0c0] Using VAAPI profile VAProfileHEVCMain (17).
[hevc_vaapi @ 0x5562070bb0c0] Using VAAPI entrypoint VAEntrypointEncSlice (6).
[hevc_vaapi @ 0x5562070bb0c0] Using VAAPI render target format YUV420 (0x1).
[hevc_vaapi @ 0x5562070bb0c0] RC mode: ICQ.
[hevc_vaapi @ 0x5562070bb0c0] RC quality: 27.
[hevc_vaapi @ 0x5562070bb0c0] RC framerate: 30000/1001 (29.97 fps).
[hevc_vaapi @ 0x5562070bb0c0] Using intra, P- and B-frames (supported references: 4 / 4).
[hevc_vaapi @ 0x5562070bb0c0] All wanted packed headers available (wanted 0xd, found 0x1f).
[hevc_vaapi @ 0x5562070bb0c0] Using level 5.
Output #0, mp4, to '20210428_233121.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2mp41
    encoder         : Lavf58.45.100
    Stream #0:0(und): Video: hevc (hevc_vaapi) (Main), 1 reference frame (hev1 / 0x31766568), vaapi_vld, 2720x1530 [SAR 1:1 DAR 16:9], q=-1--1, 29.97 fps, 30k tbn, 29.97 tbc (default)
    Metadata:
      handler_name    : VideoHandler
      encoder         : Lavc58.91.100 hevc_vaapi
```

angryziber, 2021-04-28:
mpv screenshot when playing back output file:
[image]

angryziber, 2021-04-28:
Just in case lspci -x:
```
0000:00:02.0 VGA compatible controller: Intel Corporation TigerLake GT2 [Iris Xe Graphics] (rev 01)
00: 86 80 49 9a 07 04 10 00 01 00 00 03 10 00 00 00
10: 04 00 00 3e 60 00 00 00 0c 00 00 00 40 00 00 00
20: 01 30 00 00 00 00 00 00 00 00 00 00 43 10 e2 13
30: 00 00 00 00 40 00 00 00 00 00 00 00 ff 01 00 00
```

rmkimathi, 2021-07-06:
@fulinjie built latest ffmpeg master, got same corruption with hevc_vaapi encode on intel TGL

Xiaogangli-intel, 2021-07-06:
Thank you @rmkimathi, working on this issue, will update it once we have some progress.

georgnistor, 2021-11-03:
Hi,
I have the very same problem on i9-11900T
with UHD 750.

The setup I have is:
Open SUSE Tumbleweed 20211101 with ffmpeg 4.4.1
Kernel 5.14.14-1 64 bit
Mesa Intel Graphics

I am not able to encode with ffmpeg using va-api
ffmpeg -y -vaapi_device /dev/dri/renderD128  -i  input.ts  -vf 'format=nv12,hwupload' -c:v hevc_vaapi  -b:v 6M  -c:a copy out.mkv

@lobomfz 
Did you find a solution?

feiwan1, 2021-11-09:
Hi @Xiaogangli-intel @XinfengZhang this may be caused by leak of VAConfigAttribEncHEVCFeatures&VAConfigAttribEncHEVCBlockSizes attrib report on ICL+ in media-driver. Previously Mark submitted a patch https://github.com/intel/media-driver/pull/866 which based on legacy interface. It need port to lastest libva interface. And I am not sure if Mark's patch is correct, in my understanding, ICL/TGL attrib should be sperated with common platform. So could you help to submit a new one ASAP? Since it blocks hevc encode on ICL+ by using upstream ffmpeg-vaapi. Only after the attrib report patch merged in media-driver, we can verify and submit ffmpeg-vaapi patch.
Also CC @lwan108 who reported some incorrect setting on Mark's patch.

@angryziber @lobomfz @georgnistor, here is a workarround patch for ffmpeg-vaapi, which may fix your issue temporarily:
https://github.com/intel-media-ci/cartwheel-ffmpeg/blob/master/patches/0025-lavc-vaapi_encode_h265-fix-max_transform_hierarchy_d.patch

rmkimathi, 2021-11-09:
Thank you @feiwan1

Workaround patch for ffmpeg fixes hevc_vaapi encode on TGL Xe Graphics.

xhaihao, 2021-11-11:
@feiwan1 My understanding is that the added interfaces should be applied to TGL+ only, the legacy code works on ICL.

feiwan1, 2021-11-11:
Thanks @xhaihao for reminder. Current code works on ICL indeed. Just need to update TGL+'s code only. And corrected my previous comment.

XinfengZhang, 2021-11-11:
https://github.com/intel/media-driver/pull/1121

closed 2021-12-18 by intel-mediadev

nyanmisaka, 2023-06-10:
@flyingfalling The ffmpeg fix exists in n5.1+ versions. Also make sure the ffmpeg is build against libva 2.13+.

https://github.com/FFmpeg/FFmpeg/commit/9f02e033875185409c861846f209b04a3be339d2

PR #1312 (2021-12-18): [Encode] Report sub features for HEVC
Fill in VAConfigAttribValEncHEVCFeatures and VAConfigAttribValEncHEVCBlockSizes
with the approriate values for HEVC VDENC.
Fixes #1190.
commit b7946150 (2021-12-18): [Encode] Report sub features for HEVC

Fill in VAConfigAttribValEncHEVCFeatures and VAConfigAttribValEncHEVCBlockSizes
with the approriate values for HEVC.
Fixes #1190.