intel/media-driver #1191: [BRX/APL] Very Slow frame fetch from the GPU to CPU

andreyv1978, 2021-05-04:
Hi

I hope this is a right place to ask about the issue I have with Apollo Lake customized platform
We are using Gstreamer 1.18.4 with VAAPI to decoder 1080p stream. The decoder is working in rate 1080p@30fps with no ptoblem. However, we need to copy the decoded NV12 frame from the GPU into the CPU (we have our dedicated DMA to make all required further transfers). The problem is the VAAPI provides us with untiled NV12 frame, which we are trying to copy, and the copy process is very slow (sometimes 90ms-300ms per a NV12 frame). 
We had a similar problem with our old platform 5 years ago, but we sorted it out to access tiled NV12 instead of untiled (it was Intel Baytrail not Apollo Lake)
We found now we have to make maybe changes in intel-media drivers not in intel-vaapi ? Is it right ?
Can you help us to find the right place in the driver, where we can get access to the tiled NV12 instead of untiled ?

Andrey

XinfengZhang, 2021-05-17:
hi @andreyv1978 ,sorry for late response, , it should because of the map vs map_gtt.  for apollo Lake we are using map_gtt which will enable HW tile-detile conversion. but it will be slow for read, because map_gtt is for write operation. so "write" is faster than "read".  if you want to tile unmap the surface directly. 

if you want to modify the code by yourself, you could refer #646 , it is used for linear surface for Apollo lake, you need both tile and linear

andreyv1978, 2021-05-17:
Hi

Thank you for the support
I can confirm rolling back all the changes where the mapping uses gtt to standard map makes the read from the memory fast
It requires to write the code to untile the frame, but I had this code already 
So closing this issue

closed 2021-05-17 by andreyv1978