intel/media-driver #1193: JPEG decoding changes the image contrast and/or lacks minor details

artlav, 2021-05-08:
Tried on:

    i5-6260U (Iris Graphics 540)
    m5-6Y54 (HD Graphics 515)
    i3-10110U (UHD Graphics)

Not sure if it's an issue or a limitation on the decoding.

Take loadjpeg form libva-utils, run it on any file with high contrast and/or fine details, compare to software decoding.

What is expected: https://orbides.org/img/ref.png

What is produced: https://orbides.org/img/res.png

JPG file parameters do not appear to matter, every single file i tried exhibit this effect to some degree.

I was not able to google if the hardware decode is supposed to be this imprecise or if there are some color space correction settings that need to be set or something, so i file this as an issue for now.

wangyan-intel, 2021-05-10:
@Jexu Could you please take a look? Thanks.

XinfengZhang, 2021-05-10:
hi @artlav  I suppose it is not related with JPEG decoding itself. as you know, there are different format jpeg. the format of jpeg and output surface maybe is different. so maybe sometime there are a VPP followed 
for example:  the jpeg is 444 or 422Y/422H, but application create a NV12 surface for the output. then a internal color format conversion was called. some details left.
so, could you help to check the color format of jpeg and output surface . or share the reproducer?

XinfengZhang, 2021-05-10:
of course, it also maybe related with Qmatrix (value and layout) , so, still need reproducer :)

artlav, 2021-05-10:
Problem is, EVERY jpg file is off like that, which is what makes me wonder if there is something wrong with loadjpeg form libva-utils. It does create the output surface to match the input file.

Samples:
420: https://orbides.org/img/420.jpg

422V: https://orbides.org/img/422.jpg

444: https://orbides.org/img/444.jpg

All decode to: https://orbides.org/img/bad.png

Should look like: https://orbides.org/img/good.png (i.e. in browser)

Another observation is that HW decode and encode are mutually consistent in their wrongness - jpegenc output would appear brighter for an SW decoder, but correct for loadjpeg.

artlav, 2022-07-02:
This is still an issue as of 22.4.3, and is still perfectly reproducible every time for every image.

Can also add i7-1195G7 to the list of CPUs i tried this on.

Jexu, 2022-07-04:
This maybe related with qmatrix, and it can get right output in bellow picture by raising the value of qmatrix. But hard to say it is hw limitation or something others since the qmatrix is provided by uplayer app/lib.
[image]

artlav, 2022-07-05:
The iq_matrix.quantiser_table is copied over from the JPG file, verbatim. Not sure what else the lib can do.

I tried to tweak it as well, and it's too dim with +1 and too bright with -1. Almost feels like it have to be +0.5, but there is no way to make it so.

Jexu, 2022-07-07:
So, may i know any impacts for you with the hw result?

artlav, 2022-07-07:
I'm working on an image viewer. HW decoding of JPG files is several times faster than software, which matters at DSLR camera image sizes. However i can not keep it on by default, since all images come out with the gamma visibly off.