intel/media-driver #1194: AV1 hw decoding artifacts when using VAAPI surface

a1rwulf, 2021-05-10:
media-driver: bec8e1382637589bc3f302edf6c866bec3d754dd
libva: fb31fe67f303dbbca15a8aff060ff609895406c2
ffmpeg: 4.4

I tried to implement AV1 hw decoding in Kodi, but I suffer from similar problems reported here: https://github.com/intel/media-driver/issues/1155

I read the other ticket and ffplay seems to work when used in conjunction with the sw pix format `-pix_fmt yuv420p` however we want to use the VAAPI surface to do zero-copy rendering and all applications (ffplay, mpv, kodi) seem to have
the same decoding artifacts when using the vaapi surface, which indicates to me that this is not an app issue.

Any ideas on how to improve this situation?

XinfengZhang, 2021-05-14:
hi @a1rwulf , #1155 already been rootcaused. there is a patch with ffmpeg

a1rwulf, 2021-05-14:
@XinfengZhang let's close this ticket and continue in #1155 as it's obviously the same issue and seems we are close to have a working patch.

closed 2021-05-14 by a1rwulf