intel/media-driver #1195: There is Mosaic in the background if the rendered picture is smaller than dst surface size when using EU for scaling

Nicki-fu, 2021-05-13:
Hi, 
I meet an issue that there is Mosaic in the background if the rendered picture is smaller than the output VASurface size when using EU for scaling.
I verified on the latest Media-driver, libva, libva-utils on master branch, it also exists this issue. 

Step 1. Change code in videoprocess/vavpp.cpp to test this case:
Change the pipeline buffer output_region parameter to a small size when do video frame process.
Change from:
    output_region.width = g_out_pic_width;
    output_region.height = g_out_pic_height;
to:
    output_region.width = 128;
    output_region.height = 127;

Step 2. Rebuild and run the vavpp using the process.cfg.template with right setting
./vavpp process.cfg.template

Excepted: the image is place in the output_region defined place, and other place in the surface should present transparent background, so that other surfaces below it can be seen.

Actually Results: There are Mosaics in the background unless I set the output background color to a non-transparent. 
This issue happened when output_region's width or the height is smaller than 128, or is smaller than the 1/8 of the original size.
Actually I verified that if I set the filter_flags to VA_FILTER_SCALING_HQ to enable the EU for scaling engine, this issue always happen when scaling to any size.

[image]

FurongZhang, 2021-06-07:
I rooted cause this issue. In DdiVp_SetProcPipelineParams, if Alpha !=0, driver will color Fill; otherwise, will not fill color. That is why you see corruption

FurongZhang, 2021-06-07:
@Nicki-fu is communicating with customers. @Nicki-fu , any update?

FurongZhang, 2021-06-07:
// Background Colorfill
    // According to libva  definition, if alpha in output background color is zero, then colorfill is not needed
    if ((pPipelineParam->output_background_color >> 24) != 0)
    {
        if (pVpHalRenderParams->pColorFillParams == nullptr)
        {
            pVpHalRenderParams->pColorFillParams = (PVPHAL_COLORFILL_PARAMS)MOS_AllocAndZeroMemory(sizeof(VPHAL_COLORFILL_PARAMS));
        }

        DDI_CHK_NULL(pVpHalRenderParams->pColorFillParams, "Null pColorFillParams.", VA_STATUS_ERROR_UNKNOWN);

        // set background colorfill option
        pVpHalRenderParams->pColorFillParams->Color     = pPipelineParam->output_background_color;
        pVpHalRenderParams->pColorFillParams->bYCbCr    = false;
        pVpHalRenderParams->pColorFillParams->CSpace    = CSpace_sRGB;
    }else
    {
        MOS_FreeMemAndSetNull(pVpHalRenderParams->pColorFillParams);
    }

Nicki-fu, 2021-06-07:
The customer has checked that if set the alpha of background to 01 value(0x01000000) , the background is black but not transparent as expected, and the image under it can’t be seen at all. 
Then we provided the customer another solution that to render all the videos' frame to a same picture at one VPP API time(Can see below example code to get more detail). This solution can resolve the customer’s issue.

Example code of the VPP calling:
//Create buffer for video0
    va_status = vaCreateBuffer(va_dpy,
                               context_id,
                               VAProcPipelineParameterBufferType,
                               sizeof(pipeline_param),
                               1,
                               &pipeline_param,
                               &pipeline_param_buf_id);
//Create buffer for video1  
  va_status = vaCreateBuffer(va_dpy,
                               context_id,
                               VAProcPipelineParameterBufferType,
                               sizeof(pipeline_param_1),
                               1,
                              &pipeline_param_1,
                               &pipeline_param_buf_id_1);

//Begin picture
    va_status = vaBeginPicture(va_dpy,
                               context_id,
                               out_surface_id);
    CHECK_VASTATUS(va_status, "vaBeginPicture");
//Render Picture for video0
    va_status = vaRenderPicture(va_dpy,
                                context_id,
                                &pipeline_param_buf_id,
                                1);

//Render Picture for video1
    va_status = vaRenderPicture(va_dpy,
                               context_id,
                              &pipeline_param_buf_id_1,
                             1);

    va_status = vaEndPicture(va_dpy, context_id);
//Distory Buffer for video0
    if (filter_param_buf_id != VA_INVALID_ID)
        vaDestroyBuffer(va_dpy,filter_param_buf_id);
    if (pipeline_param_buf_id != VA_INVALID_ID)
        vaDestroyBuffer(va_dpy,pipeline_param_buf_id);

//Distory Buffer for video1
    if (filter_param_buf_id_1 != VA_INVALID_ID)
        vaDestroyBuffer(va_dpy,filter_param_buf_id_1);

closed 2021-06-26 by FurongZhang