intel/media-driver #1198: video screen become black after I stop the video

widon1104, 2021-05-22:
I use chrome 90.0.4430.212 for linux. My cpu is i5 8350U hd620, after I use this driver I can use hardware accelerate now.
But when I play video in bilibili.com or ixigua.com, after I stop the video the video screen will become all black.
Sometimes I stop the video because I want to see the frame clearly I do not want the screen become black.

Thanks.

XinfengZhang, 2021-05-22:
hi thanks for reporting, but I suppose it is not a decoder issue, it looks like a webrtc or browser  issue.  because media_driver just decode the pictures.

XinfengZhang, 2021-06-03:
hi @widon1104 , could we close it, if no other concern?

widon1104, 2021-06-03:
@XinfengZhang You can close this issue.

closed 2021-06-03 by XinfengZhang