intel/media-driver #1200: VP9 Encode performance(fps) drop while comparing with H264 & HEVC

sreerenjb, 2021-06-02:
**Software stack:**
Kernel: 5.11.0-17
iHD: git master
GStreamer: git master

**Input:** 
smpte stream
res: 720p
bitrate:2500kpps

**Encoder thoughput:**
vp9: ~150fps
h264: 500+
hevc: 600+

Test command for h264 enc: 
gst-launch-1.0 -v videotestsrc num-buffers=10000 ! video/x-raw, format=NV12, width=720, height=1280, framerate=30/1 ! vaapihvp9enc rate-control=cbr bitrate=2500 tune=low-power ! fpsdisplaysink "video-sink=fakesink" text-overlay=false sync=false

h264/avc: change vaapivp9enc with vaapih264enc in the above command
h265/hevc:  change vaapivp9enc with vaapih265enc in the above command

XinfengZhang, 2021-06-03:
it is TGL, right?

sreerenjb, 2021-06-07:
It is TGL

XinfengZhang, 2021-06-13:
@vvbuzove , please help to check this issue.

sreerenjb, 2021-06-16:
BTW, here is the MSDK results:
gst-launch-1.0 -v videotestsrc num-buffers=10000 ! video/x-raw, format=NV12, width=720, height=1280, framerate=30/1 ! msdkvp9enc rate-control=cbr bitrate=2500   ! fpsdisplaysink "video-sink=fakesink" text-overlay=false sync=false

fps ~185

So it doesn't seem to be a software-level issue.

XinfengZhang, 2021-06-17:
@vvbuzove could you help to provide a sample app and command to sree, to check whether it is indeed a driver issue?

vvbuzove, 2021-06-17:
@XinfengZhang the issue was reproduced using Media SDK tools. The team is looking into this further.

pdemenko, 2021-06-25:
I tried to run hevc and vp9 with vbr and cqp with different streams, here are results
_____________________________________________________
|  resolution  | 720x480 |  1920x1080  | 3840x2160 |
-----------------------------------------------------
| num frames|    1000   |     1000        |     900         |
-----------------------------------------------------
| CQP  VP9   |    291     |     140          |       42         | <- fps
-----------------------------------------------------
| CQP HEVC |    1006    |     226         |       52         | <- fps
-----------------------------------------------------

____________________________________________________
|  resolution  | 720x480 |  1920x1080  | 3840x2160 |
-----------------------------------------------------
| num frames|    600     |     1000        |     900         |
-----------------------------------------------------
| VBR VP9    |   146       |      77          |        29        | <- fps
----------------------------------------------------
| VBR HEVC |    839      |     250         |        61        | <- fps 
----------------------------------------------------

sushma08, 2021-09-14:
@vvbuzove any updates on this?

vvbuzove, 2021-11-16:
@sushma08 we have narrowed down the issue to a component that is specific to VP9 VDEnc and is not used in AVC and HEVC. The perf drop is more noticeable on small resolutions. Detailed performance analysis is in progress.

sushma08, 2021-11-16:
Thank you @vvbuzove. Will look forward to the fix.

ansokolo, 2022-02-15:
Some changes to improve VP9 performance were made. Now collecting new perf data is work in progress

ansokolo, 2022-03-14:
Not all previous performance changes were made for TGL. But we see significant improvements on other platform. Moving changes to TGL - work in progress