intel/media-driver #1201: Fail to recognize the Mpeg2 HighP level IDC

HeJunyan, 2021-06-03:
gst-launch-1.0 videotestsrc ! capsfilter caps=video/x-raw,format=NV12,width=1920,height=1080,framerate=60/1 ! vaapimpeg2enc ! fakesink

fails even with the patch
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/merge_requests/432

The driver fails to recognize the HighP level's IDC 0x02 and return error.

https://www.itu.int/rec/T-REC-H.262/recommendation.asp?lang=en&parent=T-REC-H.262-200903-S!Amd3

is a Mpeg2 spec amendment to add this level.

XinfengZhang, 2021-06-09:
from driver code https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/codec/hal/codechal_encode_mpeg2.cpp#L1699-L1720
it already checked 3 profiles and 4 levels, but new extension is added , 1080p@60, the level is highP, the value is 2. so need to add a new check. thanks for reporting.

HeJunyan, 2021-06-09:
Yes, add the "level" seems only used for saint check, it does not change the encoding result.

PR #1205 (2021-06-18): add new level definition for MPEG2 encode
new extension is added , 1080p@60, the level is highP,
the value is 2. so need to add a new check

Fixes #1201
commit 7d7ef89c (2021-06-18): add new level definition for MPEG2 encode

new extension is added , 1080p@60, the level is highP,
the value is 2. so need to add a new check

Fixes #1201
commit 700341b1 (2021-06-18): add new level definition for MPEG2 encode

new extension is added , 1080p@60, the level is highP,
the value is 2. so need to add a new check

Fixes #1201
closed 2021-06-18 by intel-mediadev