intel/media-driver #1202: [Quality regression][TGL]Hevc vbr encode cases' BDrate was decreased by one commit on TGL.

zhangyuankun-star, 2021-06-04:
the commit 5988b2bd3280fc368e9c693f65032787d4dd309e will impact our quality test and sample_encode will also be affected.
for example:
sample_encode h265 -u $preset -b $bitrate -vbr **-extbrc:off** -x 5 -g 256 -NalHrdConformance:off -VuiNalHrdParameters:off -BufferSizeInKB $bufsize -n $nframes -i $input -w $width -h $height -f $framerate -gpucopy::on -hw -o ${prefix}_${mbps}Mbps_VBR_SENC.h265

xuguangxin, 2021-06-04:
@XinfengZhang , this is an imporant quality regression. could you help check?
thanks

XinfengZhang, 2021-06-13:
@nlu5 is checking it.

nlu5, 2021-06-14:
Comit has been reverted

Xiaogangli-intel, 2021-07-06:
Hi @nlu5, could you share the commit ID of the PR for fixing this issue?

nlu5, 2021-07-22:
The commit was reverted on gfx-driver-ci-master-8291 
commit e31016377d967af5e145945a9679a5507425b42d (tag: build/ci/gfx-driver-ci-master-8291)
https://github.com/intel-innersource/drivers.gpu.unified/pull/26813

xuguangxin, 2021-07-23:
thanks @Xiaogang-Li  and @nlu5 
@zhangyuankun-star could you double confirm this fix caught by our ci and close this
thanks