intel/media-driver #1206: Regression HEVC 10bit Encode have different R2R output stream size

egorovdanil, 2021-06-11:
Platform: DG1
libva: 8803fc5363b635f950dc904e880a35f1c13eae88
media-driver: 66fa14da59ba6a81f1d604d1c71fa93502884599
MediaSDK: 31693415c3aa58dc826748990a59981cbaa9c6fa

Command line:
`sample_encode h265 -i 1080P_1920x1080.P010 -o output1.265 -p010 -w 1920 -h 1080 -b 5000 -lowpower:off -vaapi`

From run to run, the output stream size is different:
```
-rw-r--r-- 1 root root    6284844 Apr  1 09:33 output1.265
-rw-r--r-- 1 root root    6285693 Apr  1 09:33 output2.265
-rw-r--r-- 1 root root    6285727 Apr  1 09:33 output3.265
```
This issue starts from 12b720df5367e4b1ebdc6f42e38f63c739964237 and is reproduced on the current master 66fa14da59ba6a81f1d604d1c71fa93502884599

XinfengZhang, 2021-06-13:
looks r2r with VEbox copy, @Xiaogang-Li , could you help to take a look?

xhaihao, 2021-06-30:
@XinfengZhang any progression on this issue?

wenbinc-Bin, 2021-07-12:
Hi @Xiaogang-Li and @XinfengZhang, Any plan to fix this issue? Could you give me a date and I can check this issue at that day, which helps me to better track this issue.
Thanks
Wenbin

Xiaogangli-intel, 2021-07-19:
Sorry @wenbinc-Bin, will sync with the team about the status, will let you know once I get feedback.

FocusLuo, 2021-08-16:
@Xiaogang-Li @XinfengZhang  This is a Run 2 Run encode issue. May I check with you whether there is any progress on this issue fixing?

Xiaogangli-intel, 2022-04-26:
we tried latest driver, the issue should be fixed.
> Kernel 5.10.54+prerelease3052
Latest libva/gmmlib/media-driver/MediaSDK

> 256c5b0cf16bc29a89fcb0a165a01c60  output1.265
256c5b0cf16bc29a89fcb0a165a01c60  output2.265
256c5b0cf16bc29a89fcb0a165a01c60  output3.265
256c5b0cf16bc29a89fcb0a165a01c60  output4.265
256c5b0cf16bc29a89fcb0a165a01c60  output5.265

Xiaogangli-intel, 2022-05-07:
Hi @egorovdanil , @xhaihao, not sure if you tried the latest driver. If no more concern about this issue, I will close it. Thank you.

closed 2022-05-10 by Xiaogangli-intel