intel/media-driver #1210: VA_EXPORT_SURFACE_SEPARATE_LAYERS broken with auxplane

bkuhls, 2021-06-22:
Kodi uses vaExportSurfaceHandle with VA_EXPORT_SURFACE_SEPARATE_LAYERS 
https://github.com/xbmc/xbmc/blob/master/xbmc/cores/VideoPlayer/VideoRenderers/HwDecRender/VaapiEGL.cpp#L453

and expects only one plane per layer
https://github.com/xbmc/xbmc/blob/master/xbmc/cores/VideoPlayer/VideoRenderers/HwDecRender/VaapiEGL.cpp#L486

Starting with 11th gen i5/i7/i9 cpus Kodi detects two planes however
https://github.com/xbmc/xbmc/blob/master/xbmc/cores/VideoPlayer/VideoRenderers/HwDecRender/VaapiEGL.cpp#L489
https://github.com/xbmc/xbmc/issues/19820
https://forum.kodi.tv/showthread.php?tid=362073&pid=3042144#pid3042144

and the generated image for HD content is green while SD content displays fine.

Afaics VA_EXPORT_SURFACE_SEPARATE_LAYERS is defined as
https://github.com/intel/libva/blob/fb31fe67f303dbbca15a8aff060ff609895406c2/va/va_drmcommon.h#L96
```
* If VA_EXPORT_SURFACE_SEPARATE_LAYERS is specified on export, each
* layer will contain exactly one plane.
```
The code creating the two planes, due to mediaCtx->m_auxTableMgr being true, can be found here :
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva.cpp#L7055

Is there a bug in media-driver not properly supporting VA_EXPORT_SURFACE_SEPARATE_LAYERS or are changes in Kodi needed due to the new features of the gen12 igpu?

XinfengZhang, 2021-06-23:
it is because the MMC enabled. there are some description, https://github.com/torvalds/linux/blob/master/include/uapi/drm/drm_fourcc.h#L528
but form implementation side, it maybe should use 4 layer with 1 plane /layer  instead 2 layer with 2planes/player
@Guangyao-Bai , could you help to check the implementation about the aux data?

Guangyao-Bai, 2021-06-23:
hi  @XinfengZhang as i checked the export logic, it was using 2 layers, and each layer have two planes for pixel buffer and aux data buyffer.

XinfengZhang, 2021-06-23:
@Guangyao-Bai , why we dont use 4 layer, and each layer just has 1 plane?

Guangyao-Bai, 2021-06-23:
@XinfengZhang my Fault, i recall the history, in current libva, for Planar surface like NV12/P010, we support two kinds of report method based on the usages how the APP using.
Composit mode: where SW will treat NV12 as a combined format, in this case, we do export 4 plane and the orider will be Y, YCCS, UV, UVCCS, check defnie here https://github.com/torvalds/linux/blob/master/include/uapi/drm/drm_fourcc.h#L528 

but some SW will use NV12 as seperated mode like R8 for Y Plane, and R8G8 for UV plane, in this case, media will report as seperated mode with two layer and each layer have 2 planes.

XinfengZhang, 2021-06-23:
but from vaapi definition
* If VA_EXPORT_SURFACE_SEPARATE_LAYERS is specified on export, each
* layer will contain exactly one plane.
seems we need four layer and 1 plane in each layer

Guangyao-Bai, 2021-06-23:
from VAAPI define: http://intel.github.io/libva/group__api__core.html#ga1eafc62ad88550c0e9bed7545a07d008
#define VA_EXPORT_SURFACE_SEPARATE_LAYERS   0x0004
Export surface with separate layers
For example, NV12 surfaces should be exported as **two separate** planes for luma and chroma. thus driver inplenment with two seperated layer, but let talk tmr to make it clear

bkuhls, 2021-06-23:
Quoting https://github.com/intel/libva/pull/125#issuecomment-340274373

>  With VA_EXPORT_SURFACE_SEPARATE_LAYERS set, layers should all have exactly one plane. The use for this form is for import to other GPU processing APIs (OpenGL, OpenCL, Vulkan, etc.), which (barring some vendor extensions) always want planes separately.

Guangyao-Bai, 2021-06-23:
then how to understand the VA_EXPORT_SURFACE_COMPOSED_LAYERS, one single composited layers with one plane? then how to define the Luma(Y) and Chroma(UV) layout?  

 /** Export surface with composed layers.
 *
 * For example, NV12 surfaces should be exported as a single NV12
 * composed object.
 */
#define VA_EXPORT_SURFACE_COMPOSED_LAYERS  0x0008

from https://github.com/intel/libva/blob/master/va/va_drmcommon.h , 
"
For export, call vaExportSurfaceHandle() with mem_type set to
 * V**A_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME_2** and pass a pointer to an
 * instance of this structure to fill.
 * If **VA_EXPORT_SURFACE_SEPARATE_LAYERS** is specified on export, each
 * layer will contain exactly one plane.  For example, an NV12
 * surface will be exported as **two layers**, one of DRM_FORMAT_R8 and
 * one of DRM_FORMAT_GR88.
 * If VA_EXPORT_SURFACE_COMPOSED_LAYERS is specified on export,
 * there will be exactly one layer.
 *
 * For import, call vaCreateSurfaces() with the MemoryType attribute
 * set to VA_SURFACE_ATTRIB_MEM_TYPE_DRM_PRIME_2 and the
 * ExternalBufferDescriptor attribute set to point to an array of
 * num_surfaces instances of this structure.
 * The number of planes which need to be provided for a given layer
 * is dependent on both the format and the format modifier used for
 * the objects containing it.  For example, the format DRM_FORMAT_RGBA
 * normally requires one plane, but with the format modifier
 * **I915_FORMAT_MOD_Y_TILED_CCS** it requires **two planes** - **the first
 * being the main data plane and the second containing the color
 * control surface.**
"

Guangyao-Bai, 2021-06-23:
i mean "**VA_EXPORT_SURFACE_SEPARATE_LAYERS**" means libva need to export two spearated layers. and with modifier **I915_FORMAT_MOD_Y_TILED_CCS**, additional CCS plan needed to maintain the color control surfaces

bkuhls, 2021-06-23:
> and with modifier **I915_FORMAT_MOD_Y_TILED_CCS**, additional CCS plan in each layer needed to maintain the color control surfaces

Afaics this was discussed as case 2) and not implemented: https://github.com/intel/libva/pull/125#issuecomment-340274373. This PR was also the source for the, somehow misleading, comment
```
 * For example, NV12 surfaces should be exported as two separate
 * planes for luma and chroma.
```
https://github.com/intel/libva/pull/125/files#diff-e0e7f3004535ebf01d1e34565ff4ba14ccbe5798a7814df2004cd5fd26530812R3318

but reading the [comments](https://github.com/intel/libva/pull/125#issuecomment-341342165) in that PR it is clear to me that VA_EXPORT_SURFACE_SEPARATE_LAYERS "will never return more than one plane per layer", something Kodi depends on.

XinfengZhang, 2021-06-24:
so, this two planes (main plane and color control plane) share same "drm_format" , they are in one layer. ,https://github.com/intel/libva/blob/master/va/va_drmcommon.h#L107-L113 is supplementary to https://github.com/intel/libva/blob/master/va/va_drmcommon.h#L96-L101

bkuhls, 2021-06-27:
> so, this two planes (main plane and color control plane) share same "drm_format" , they are in one layer. ,https://github.com/intel/libva/blob/master/va/va_drmcommon.h#L107-L113 is supplementary to https://github.com/intel/libva/blob/master/va/va_drmcommon.h#L96-L101

But VA_EXPORT_SURFACE_SEPARATE_LAYERS is still defined as "each layer will contain exactly one plane", regardless of other modifiers, like I915_FORMAT_MOD_Y_TILED_CCS.
What do you propose to make Kodi work on newer Intel CPUs?

bkuhls, 2021-06-27:
> it is because the MMC enabled

Building media-driver with -DMMC_Supported=NO does not fix the problem.

XinfengZhang, 2021-06-28:
@bkuhls it should only be used for "surface sharing" between different gpu component , the data is not readable ---- app could not decompress it. if you want  to access the data, you should use vaGetImage or vaDeriveImage + vaMapbuffer, media driver will decompress it.
when using the "surface sharing" application should read the obj number, layer number and planes in each layer and convey these information to other component .

bkuhls, 2021-06-29:
> i mean "**VA_EXPORT_SURFACE_SEPARATE_LAYERS**" means libva need to export two spearated layers. and with modifier **I915_FORMAT_MOD_Y_TILED_CCS**, additional CCS plan in each layer needed to maintain the color control surfaces

How to proceed with this issue then? Do you acknowledge that there is a bug in media-driver not following the specs laid out in libva or is Kodi at fault due to an incomplete understanding of the specs?

starvald, 2021-07-05:
So the intel devs going to just to ghost us?

bkuhls, 2021-07-10:
> So the intel devs going to just to ghost us?

Please test the current [master](https://github.com/intel/media-driver/commits/29d5f8fc118476dee84f3f784443badc819a3241) tree, it seems the problem was fixed:
https://forum.kodi.tv/showthread.php?tid=362073&pid=3048667#pid3048667

lrusak, 2021-07-10:
It would be nice to know what commit fixed it.

bkuhls, 2021-07-11:
> It would be nice to know what commit fixed it.

https://github.com/intel/media-driver/commit/29d5f8fc118476dee84f3f784443badc819a3241 is a likely candidate.

Guangyao-Bai, 2021-07-12:
29d5f8f is a fix only for RKL as some stepping info missed on RKL. but for Aux plan per layer is needed if you want to enable E2E Compression to save power.

bkuhls, 2021-07-12:
> but for Aux plan per layer is needed if you want to enable E2E Compression to save power.

Any news on my [question](https://github.com/intel/media-driver/issues/1210#issuecomment-870842204)?
Currently Kodi works only because the feature "E2E Compression", which creates the aux plane, is disabled for the time being.

Guangyao-Bai, 2021-07-12:
> Any news on my [question](https://github.com/intel/media-driver/issues/1210#issuecomment-870842204)?
> Currently Kodi works only because the feature "E2E Compression", which creates the aux plane, is [...]

E2E Compression was a new feature on Intel platforms like RKL, and it needs a CCS plane to maintain the compression/color control surfaces, thus why a new modify type defined "**I915_FORMAT_MOD_Y_TILED_CCS**". 

but on RKL, there's a HW limitation in display on those steppings, which require media must output a uncompressed surface. thus why this fix(29d5f8f ). 

but if you check same kodi on TGL-R or newer platforms, it still needs this additonal plane for CCS.

bkuhls, 2021-07-12:
> but if you check same kodi on TGL-R or newer platforms, it still needs this additonal plane for CCS.

Would it be possible by using DRM_FORMAT_MOD_LINEAR to turn off the CCS plane so that each layer will contain only one plane?

lrusak, 2021-07-13:
> > but if you check same kodi on TGL-R or newer platforms, it still needs this additonal plane for CCS.
> 
> 
> 
> Would it be possible by using DRM_FORMAT_MOD_LINEAR to turn off the CCS plane so that each layer will contain only one plane?

It's unlikely as tiling will probably be required by the display engine (and depending what the decoder can actually output)

Guangyao-Bai, 2022-10-20:
@bkuhls could you pls help cross check whether issue still there? we already modified code behavious, only enable compressed surface share per requested.  pls double check latest status and close this bug?

bkuhls, 2022-10-27:
> @bkuhls could you pls help cross check whether issue still there?

Closing this bug as resolved, ftr some system infos
```
 # cat /proc/cpuinfo | grep name | uniq
model name      : 11th Gen Intel(R) Core(TM) i9-11900T @ 1.50GHz
# vainfo
Trying display: drm
libva info: VA-API version 1.16.0
libva info: Trying to open /usr/lib/va/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_16
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.16.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.6.0 ()
```

closed 2022-10-27 by bkuhls