intel/media-driver #1213: compile failure with gcc11:  'cscMatrix' may be used uninitialized

satmandu, 2021-06-30:
This with compiling `21.2.3`:

```
../media_driver/agnostic/common/codec/hal/codechal_encode_sfc_base.cpp:371:25: error: 'cscMatrix' may be used uninitialized [-Werror=maybe-uninitialized]
```
```
[107/882] Building CXX object media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_encode_sfc_base.cpp.o
FAILED: media_driver/CMakeFiles/iHD_drv_video_OBJ.dir/agnostic/common/codec/hal/codechal_encode_sfc_base.cpp.o 
/usr/local/bin/c++ -DENABLE_KERNELS -DHEVC_FEI_ENABLE_CMRT -DIGFX_GEN10_CNL_SUPPORTED -DIGFX_GEN10_SUPPORTED -DIGFX_GEN11_ICLLP_SUPPORTED -DIGFX_GEN11_JSL_SUPPORTED -DIGFX_GEN11_SUPPORTED -DIGFX_GEN12_ADLP_SUPPORTED -DIGFX_GEN12_ADLS_SUPPORTED -DIGFX_GEN12_RKL_SUPPORTED -DIGFX_GEN12_SUPPORTED -DIGFX_GEN12_TGLLP_CMFCPATCH_SUPPORTED -DIGFX_GEN12_TGLLP_CMFC_SUPPORTED -DIGFX_GEN12_TGLLP_SUPPORTED -DIGFX_GEN12_TGLLP_SWSB_SUPPORTED -DIGFX_GEN8_BDW_SUPPORTED -DIGFX_GEN8_SUPPORTED -DIGFX_GEN9_BXT_SUPPORTED -DIGFX_GEN9_CFL_SUPPORTED -DIGFX_GEN9_CML_SUPPORTED -DIGFX_GEN9_CMPV_SUPPORTED -DIGFX_GEN9_GLK_SUPPORTED -DIGFX_GEN9_KBL_SUPPORTED -DIGFX_GEN9_SKL_SUPPORTED -DIGFX_GEN9_SUPPORTED -DMEDIA_VERSION=\"21.2.3\" -DMEDIA_VERSION_DETAILS=\"008fb3b\" -DVEBOX_AUTO_DENOISE_SUPPORTED=1 -DX11_FOUND -D_AV1_DECODE_SUPPORTED -D_AVC_DECODE_SUPPORTED -D_AVC_ENCODE_VDENC_SUPPORTED -D_AVC_ENCODE_VME_SUPPORTED -D_COMMON_ENCODE_SUPPORTED -D_DECODE_PROCESSING_SUPPORTED -D_HEVC_DECODE_SUPPORTED -D_HEVC_ENCODE_VDENC_SUPPORTED -D_HEVC_ENCODE_VME_SUPPORTED -D_JPEG_DECODE_SUPPORTED -D_JPEG_ENCODE_SUPPORTED -D_MMC_SUPPORTED -D_MPEG2_DECODE_SUPPORTED -D_MPEG2_ENCODE_VME_SUPPORTED -D_RELEASE -D_VC1_DECODE_SUPPORTED -D_VP8_DECODE_SUPPORTED -D_VP8_ENCODE_SUPPORTED -D_VP9_DECODE_SUPPORTED -D_VP9_ENCODE_VDENC_SUPPORTED -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -D__VPHAL_SFC_SUPPORTED=1 -I../media_driver/linux/common/os/i915/include -I../media_driver/linux/common/os/i915/include/uapi -I/usr/local/include/igdgmm/util -I/usr/local/include/igdgmm/inc/common -I/usr/local/include/igdgmm/inc -I/usr/local/include/igdgmm/GmmLib/inc -I/usr/local/include/igdgmm/GmmLib -I/usr/local/include/igdgmm -I../media_driver/agnostic/common/cm -I../media_driver/agnostic/common/codec/hal -I../media_driver/agnostic/common/codec/kernel -I../media_driver/agnostic/common/codec/shared -I../media_driver/agnostic/common/cp -I../media_driver/agnostic/common/heap_manager -I../media_driver/agnostic/common/hw/vdbox -I../media_driver/agnostic/common/hw -I../media_driver/agnostic/common/media_interfaces -I../media_driver/agnostic/common/os -I../media_driver/agnostic/common/renderhal -I../media_driver/agnostic/common/shared/user_setting -I../media_driver/agnostic/common/shared -I../media_driver/agnostic/common/vp/hal -I../media_driver/agnostic/common/vp/kdll -I../media_driver/agnostic/common/vp/kernel -I../media_driver/agnostic/common/vp/cm_fc_ld -I../media_driver/agnostic/gen8/cm -I../media_driver/agnostic/gen8/codec/hal -I../media_driver/agnostic/gen8/codec/kernel -I../media_driver/agnostic/gen8/hw/vdbox -I../media_driver/agnostic/gen8/hw -I../media_driver/agnostic/gen8/renderhal -I../media_driver/agnostic/gen8/vp/hal -I../media_driver/agnostic/gen8/vp/kernel -I../media_driver/agnostic/gen8_bdw/codec/hal -I../media_driver/agnostic/gen8_bdw/hw/vdbox -I../media_driver/agnostic/gen8_bdw/renderhal -I../media_driver/agnostic/gen9/cm -I../media_driver/agnostic/gen9/codec/hal -I../media_driver/agnostic/gen9/codec/kernel -I../media_driver/agnostic/gen9/hw/vdbox -I../media_driver/agnostic/gen9/hw -I../media_driver/agnostic/gen9/renderhal -I../media_driver/agnostic/gen9/vp/hal -I../media_driver/agnostic/gen9/vp/kernel -I../media_driver/agnostic/gen9_cml/vp/kernel/tgp -I../media_driver/agnostic/gen9_cml/vp/kernel -I../media_driver/agnostic/gen9_cmpv/vp/kernel -I../media_driver/agnostic/gen9_bxt/codec/hal -I../media_driver/agnostic/gen9_bxt/codec/kernel -I../media_driver/agnostic/gen9_bxt/hw/vdbox -I../media_driver/agnostic/gen9_bxt/vp/hal -I../media_driver/agnostic/gen9_skl/codec/hal -I../media_driver/agnostic/gen9_skl/codec/cmrt_kernel -I../media_driver/agnostic/gen9_skl/hw/vdbox -I../media_driver/agnostic/gen9_glk/codec/hal -I../media_driver/agnostic/gen9_glk/hw/vdbox -I../media_driver/agnostic/gen9_glk/vp/hal -I../media_driver/agnostic/gen9_kbl/codec/hal -I../media_driver/agnostic/gen9_kbl/codec/kernel -I../media_driver/agnostic/gen9_kbl/hw/vdbox -I../media_driver/agnostic/gen10/cm -I../media_driver/agnostic/gen10/codec/hal -I../media_driver/agnostic/gen10/codec/kernel -I../media_driver/agnostic/gen10/codec/share -I../media_driver/agnostic/gen10/hw/vdbox -I../media_driver/agnostic/gen10/hw -I../media_driver/agnostic/gen10/renderhal -I../media_driver/agnostic/gen10/vp/hal -I../media_driver/agnostic/gen10/vp/kernel -I../media_driver/agnostic/gen11/cm -I../media_driver/agnostic/gen11/codec/hal -I../media_driver/agnostic/gen11/codec/kernel -I../media_driver/agnostic/gen11/codec/share -I../media_driver/agnostic/gen11/hw/vdbox -I../media_driver/agnostic/gen11/hw -I../media_driver/agnostic/gen11/renderhal -I../media_driver/agnostic/gen11/vp/hal -I../media_driver/agnostic/gen11_icllp/codec/hal -I../media_driver/agnostic/gen11_icllp/codec/kernel -I../media_driver/agnostic/gen11_icllp/vp/hal -I../media_driver/agnostic/gen11_icllp/vp/kernel -I../media_driver/agnostic/gen11_jsl_ehl/vp/hal -I../media_driver/agnostic/gen11_jsl_ehl/renderhal -I../media_driver/agnostic/gen12/cm -I../media_driver/agnostic/gen12/codec/hal -I../media_driver/agnostic/gen12/codec/kernel -I../media_driver/agnostic/gen12/codec/kernelisa -I../media_driver/agnostic/gen12/codec/shared -I../media_driver/agnostic/gen12/hw/vdbox -I../media_driver/agnostic/gen12/hw -I../media_driver/agnostic/gen12/vp/hal -I../media_driver/agnostic/gen12/shared -I../media_driver/agnostic/g12/g12_base/hw/render -I../media_driver/media_driver_next/agnostic/gen12/vp/hal -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/dec/av1/pipeline -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/dec/av1/packet -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/dec/av1/features -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/dec/av1 -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/dec/shared -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/dec -I../media_driver/media_driver_next/agnostic/gen12/codec/hal/shared -I../media_driver/media_driver_next/agnostic/gen12/codec/hal -I../media_driver/agnostic/gen12_tgllp/vp/hal -I../media_driver/agnostic/gen12_tgllp/vp/kernel/cmfc -I../media_driver/agnostic/gen12_tgllp/vp/kernel/cmfcpatch -I../media_driver/agnostic/gen12_tgllp/vp/kernel/cmfccmlpch -I../media_driver/agnostic/gen12_tgllp/vp/kernel -I../media_driver/linux/common/cm/ddi -I../media_driver/linux/common/cm/hal -I../media_driver/linux/common/codec/ddi -I../media_driver/linux/common/cp/ddi -I../media_driver/linux/common/cp/hw -I../media_driver/linux/common/cp/os -I../media_driver/linux/common/cp/shared -I../media_driver/linux/common/ddi -I../media_driver/linux/common/os -I../media_driver/linux/common/vp/ddi -I../media_driver/linux/common/hw -I../media_driver/linux/gen8/ddi -I../media_driver/linux/gen9/ddi -I../media_driver/linux/gen9_bxt/ddi -I../media_driver/linux/gen9_skl/ddi -I../media_driver/linux/gen9_kbl/ddi -I../media_driver/linux/gen9_glk/ddi -I../media_driver/linux/gen9_cfl/ddi -I../media_driver/linux/gen10/ddi -I../media_driver/linux/gen10_cnl/ddi -I../media_driver/linux/gen11/codec/ddi -I../media_driver/linux/gen11/ddi -I../media_driver/linux/gen12/codec/ddi -I../media_driver/linux/gen12/ddi -I../media_driver/media_interface/media_interfaces_m8_bdw -I../media_driver/media_interface/media_interfaces_m9_bxt -I../media_driver/media_interface/media_interfaces_m9_skl -I../media_driver/media_interface/media_interfaces_m9_cfl -I../media_driver/media_interface/media_interfaces_m9_glk -I../media_driver/media_interface/media_interfaces_m9_kbl -I../media_driver/media_interface/media_interfaces_m10_cnl -I../media_driver/media_interface/media_interfaces_m11_icllp -I../media_driver/media_interface/media_interfaces_m11_jsl_ehl -I../media_driver/media_interface/media_interfaces_m12_tgllp -I../media_driver/media_interface/media_interfaces_m12_rkl -I../media_driver/media_interface/media_interfaces_m12_adls -I../media_driver/media_interface/media_interfaces_m12_adlp -I../media_driver/media_driver_next/agnostic/common/shared/pipeline -I../media_driver/media_driver_next/agnostic/common/shared/packet -I../medi [... cut ...]
                 from ../media_driver/agnostic/common/codec/hal/codechal_encode_sfc_base.cpp:28:
../media_driver/agnostic/common/os/mos_utilities.h:1200:12: note: by argument 3 of type 'const void *' to 'MOS_SecureMemcpy' declared here
 1200 | MOS_STATUS MOS_SecureMemcpy(
      |            ^~~~~~~~~~~~~~~~
../media_driver/agnostic/common/codec/hal/codechal_encode_sfc_base.cpp:360:13: note: 'cscMatrix' declared here
  360 |     float   cscMatrix[12];
      |             ^~~~~~~~~
cc1plus: all warnings being treated as errors
```

XinfengZhang, 2021-07-13:
suppose there are some default build option in your env, cscMatrix is initialized in previous call --- GetCSCMatrix.  and it could be build successfully in my local side.  could you share more information, such as g++ versions, linux distribution...

satmandu, 2021-07-13:
```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/local/libexec/gcc/x86_64-cros-linux-gnu/11/lto-wrapper
Target: x86_64-cros-linux-gnu
Configured with: ../configure --prefix=/usr/local --libdir=/usr/local/lib64 --mandir=/usr/local/share/man --build=x86_64-cros-linux-gnu --host=x86_64-cros-linux-gnu --target=x86_64-cros-linux-gnu --program-prefix= --program-suffix= --disable-bootstrap --disable-libmpx --disable-libssp --disable-multilib --disable-werror --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-host-shared --enable-lto --enable-plugin --enable-shared --enable-symvers --enable-static --enable-threads=posix --with-gcc-major-version-only --with-gmp --with-isl --with-mpc --with-mpfr --with-pic --with-system-libunwind --with-system-zlib --with-arch-64=x86-64 --enable-languages=c,c++,jit,objc,fortran,go --program-suffix=-11
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.1.0 (GCC) 
```
Our Distribution is Chromebrew (we're running inside ChromeOS).

dvrogozh, 2021-07-13:
https://github.com/intel/media-driver/blob/29d5f8fc118476dee84f3f784443badc819a3241/media_driver/agnostic/common/codec/hal/codechal_encode_sfc_base.cpp#L360

@XinfengZhang : as you can see, `cscMatrix` is left uninitialized. Then we pass it to `GetCSCMatrix()` function to initialize it, but this function has initialization under conditional branches. So, compiler warns us that under some circumstances initialization code might not be actually execute and this matrix will be left uninitialized. Compiler is absolutely right. We should account on such situation from robustness concerns. This needs to be fixed, default initializer added (at least).

... and gcc-11 is finally out... and not covered by our ci...

XinfengZhang, 2021-07-14:
ok , lets change it to "float cscMatrix[12]{}"

dvrogozh, 2021-07-14:
> ok , lets change it to "float cscMatrix[12]{}"

yep. that's what I would do.

PR #1224 (2021-08-10): initialize the array before use it
fixes #1213
closed 2021-08-10 by intel-mediadev