intel/media-driver #1216: License headers for vphal_render_fast1ton.h,cpp

cdotstout, 2021-07-02:
We're interested in using media-driver as part of a Fuchsia driver for media acceleration on Intel hardware.

The license headers in these files:
https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal_render_fast1ton.cpp
https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal_render_fast1ton.h

don't match the rest of the files in that directory, for example:
https://github.com/intel/media-driver/blob/master/media_driver/agnostic/common/vp/hal/vphal.h

Is there a reason for the discrepancy or is it possible to update the license headers for those two files?

Thanks.

XinfengZhang, 2021-07-02:
sure, will check it and update the license. suppose just because it is missed when update the license.

cdotstout, 2021-07-03:
Thank you.

I raised a similar issue in gmmlib, could you please take a look?

https://github.com/intel/gmmlib/issues/85

PR #1217 (2021-07-08): Update vphal_render_fast1ton file liscense
Fixes: #1216
Signed off by: jay.yang@intel.com
commit d8f9e25c (2021-07-08): Update vphal_render_fast1ton file liscense

Fixes: #1216
Signed off by: jay.yang@intel.com
closed 2021-07-08 by intel-mediadev