intel/media-driver #1219: Video flicker sometimes use this driver

widon1104, 2021-07-04:
I use deepin(linux)，chrome 版本 91.0.4472.114（正式版本） （64 位），I start chrome use: google-chrome-stable --enable-features=VaapiVideoDecoder to start hardware decode.
I find out the video will flicker sometimes, I will give you one example, the url: [https://www.ixigua.com/6948405576535638567?id=6970584012318310926&logTag=51335e14e723ea6aaf01](url), when I play this url on the 00:14 the video will flicker everytime.

XinfengZhang, 2021-07-04:
thanks for reporting.  so , does video flicker with some dedicate/special stream ?

widon1104, 2021-07-04:
many video will flicker，so I think it's a  common phenomenon

wangyan-intel, 2021-07-05:
@widon1104 
We cannot access your URL. Could you please share it by another way?
Thanks.

widon1104, 2021-07-05:
@wangyan-intel 

https://user-images.githubusercontent.com/12093211/124415030-84e6b380-dd86-11eb-99c2-9aff5f24b4f4.mp4

you can upload this video to youtube, and try it use google chrome with --enable-features=VaapiVideoDecoder, by the way I use i5-8350U

wangyan-intel, 2021-07-05:
@Jexu Could you please take a look?

eero-t, 2023-07-04:
@widon1104 Does this this still happen with latest media-driver version?  (and udpdated kernel?)

@Jexu Is there any update on this from your side?

Jexu, 2023-07-06:
I didn't hear similar issue from chrome team and i am not sure if it is related to media driver or chrome browser. So could you try again to update media driver, kernel and browser to the latest one.