intel/media-driver #1225: vaapipostproc fail to convert when playing MJPEG 4K 8bit 444 video with vaEndPicture return VA_STATUS_ERROR_OPERATION_FAILED

SiewHoon, 2021-07-15:
The MJPEG 2k video here go thru vaapipostproc: Input : colorimetry bt709 -> output: colorimetry sRGB. video clip: Firework_2560x1600_160mbps_60fps_444.avi

For MJPEG 4k (444) video decode go thru vaapipostproc: Input: colorimetry bt2020 -> output: colorimetry sRGB. Video clip: Elecard1_3840x2160_8mbps_30fps_444_1.avi

I re-check back for previous for H265 4k (444) no issue video clip go thru vaapipostproc: Input colorimetry bt2020 -> output: colorimetry sRGB. video clip: Bunny_3840x2160_15mbps_60fps_Main_at_L5.1_444.mkv

vaBeginPicture, vaRenderPicture is done with return VA_STATUS_SUCCESS. Only it fail on vaEndPicture with return VA_STATUS_ERROR_OPERATION_FAILED at here. But, after I revert this changed in gstreamer-vaapi https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/commit/a354c14e58f9afc6548668f5d6ed9bcc1a1d35f7 , vaapipostproc will be success. I know this issue may related to media-driver side. But, I will submit at here first.

Command:
gst-launch-1.0 filesrc location=/mnt/media/Elecard1_3840x2160_8mbps_30fps_444_1.avi ! avidemux ! vaapijpegdec ! vaapipostproc ! glimagesink

media driver: intel-media-21.2.2
libva: 2.12.0
gstreamer framework master:
gstreamer-vaapi: master
Platform: TGL-U
OS: Ubuntu 18.04

This issue been report at here, but get closed it because is not gstreamer-vaapi plugins problem.
https://gitlab.freedesktop.org/gstreamer/gstreamer-vaapi/-/issues/301

SiewHoon, 2021-08-04:
Hi @MicroYY 

Any update about this issue?
This also impacted on ADL platform as well.

MicroYY, 2021-09-10:
Hi @SiewHoon 

MJPEG 2k
MJPEG 4k (444)
H265 4k (444)

Could you comment on what's the format of decode output in the above 3 cases?
From the driver log, I see MJPEG 4k (444) decode format is 444 planar, while BT2020 444 planar format  isn't supported by driver.

FurongZhang, 2021-09-14:
@MicroYY , what is BT2020 444 planar format you are seeing? Thanks a lot in advance!

ValZapod, 2021-11-22:
It should use bt601 with full range since that is the only matrix mjpeg supports.

SiewHoon, 2021-11-24:
Hi @MicroYY @jiafengy1 

I tried on this https://github.com/intel/media-driver/pull/1281 and running the test with gstreamer-vaapi in pipeline mention in this issue. It is still failing.

SiewHoon, 2021-11-24:
HI @MicroYY  @jiafengy1 

The PR#1281 can work after git format-patch -1  from jiafeng repo and apply on top of Q3'2021 media driver and able execute on this command pipeline :

Command:
export GST_GL_PLATFORM=egl
gst-launch-1.0 filesrc location=/mnt/media/Elecard1_3840x2160_8mbps_30fps_444_1.avi ! avidemux ! vaapijpegdec ! vaapipostproc ! glimagesink

MicroYY, 2021-12-07:
We discussed internally and got an agreement that PR https://github.com/intel/media-driver/pull/1281 isn't a proper fix as it falls back BT2020 to BT709 silently while APP won't know. Currently Linux caps can't query color format and space simultaneous in one function, so media driver can't handle this case. 
A initial idea is to add new API for caps, but we haven't finalize it. Will give more details when we have any progress.

FurongZhang, 2022-01-17:
Will check the current API if it is good for this case.

MicroYY, 2022-02-12:
Updated docs. Pls refer to https://github.com/intel/media-driver/pull/1351
RGB 444 planar is only supported by composition path, but composition path doesn't support bt2020 color space. So in this case driver returns failure.
Currently Linux caps can't query color format and space simultaneous in one function, so a short term solution is to avoid using this format and BT2020 in apps. Long term solution need new API in Libva.
Close this issue.

closed 2022-02-12 by MicroYY