intel/media-driver #1227: gst va decoder random crash when play some small resolution VP9 stream.

HeJunyan, 2021-07-19:
gst-launch-1.0 -vf filesrc location=./media/Smaller_VP9_209_extra_smaller_1.1.ivf ! ivfparse ! vp9parse ! vavp9dec ! fakesink

generate random crash(about once in two). 
It crash in vaSyncSurface:

0  DdiDecode_StatusReport(DDI_MEDIA_CONTEXT*, CodechalDecode*, _DDI_MEDIA_SURFACE*) (mediaCtx=0x56498beea440, decoder=0x7fa1d403eae0, surface=0x7fa1d5f6c580)
    at /home/robinhe/media/media-driver/media_driver/linux/common/codec/ddi/media_libva_decoder.cpp:315
#1  0x00007fa1dc19620c in DdiMedia_StatusCheck(PDDI_MEDIA_CONTEXT, DDI_MEDIA_SURFACE*, VASurfaceID) (mediaCtx=0x56498beea440, surface=0x7fa1d5f6c580, surface_id=22)
    at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3832
#2  0x00007fa1dc196926 in DdiMedia_SyncSurface(VADriverContext*, unsigned int) (ctx=0x56498c099e20, render_target=22) at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva.cpp:3937
#3  0x00007fa1dc1c499e in MediaLibvaInterface::SyncSurface(VADriverContext*, unsigned int) (ctx=0x56498c099e20, render_target=22)
    at /home/robinhe/media/media-driver/media_driver/linux/common/ddi/media_libva_interface.cpp:335
#4  0x00007fa1de515b4b in vaSyncSurface (dpy=0x56498be64e50, render_target=22) at ../va/va.c:1645
#5  0x00007fa1de585ec9 in _sync_surface (display=0x56498bf999b0 [GstVaDisplayDrm|vadisplaydrm1], surface=22) at ../sys/va/gstvaallocator.c:235
#6  0x00007fa1de5890ec in _ensure_image (display=0x56498bf999b0 [GstVaDisplayDrm|vadisplaydrm1], surface=22, info=0x7fa1d411c630, image=0x7fa1d4129f18, derived=0) at ../sys/va/gstvaallocator.c:1153
#7  0x00007fa1de589878 in _va_map_unlocked (mem=0x7fa1d4129ea0 [GstMemory], flags=GST_MAP_READ) at ../sys/va/gstvaallocator.c:1287
#8  0x00007fa1de589993 in _va_map (mem=0x7fa1d4129ea0 [GstMemory], maxsize=20480, flags=GST_MAP_READ) at ../sys/va/gstvaallocator.c:1321
#9  0x00007fa1dfe53852 in gst_memory_map (mem=0x7fa1d4129ea0 [GstMemory], info=0x7fa1daeff888, flags=GST_MAP_READ) at ../gst/gstmemory.c:308
#10 0x00007fa1dfe535ea in gst_memory_make_mapped (mem=0x7fa1d4129ea0 [GstMemory], info=0x7fa1daeff888, flags=GST_MAP_READ) at ../gst/gstmemory.c:241
#11 0x00007fa1dfe0ae8e in gst_buffer_map_range (buffer=0x56498c07e6c0 [GstBuffer], idx=0, length=1, info=0x7fa1daeff888, flags=GST_MAP_READ) at ../gst/gstbuffer.c:1852
#12 0x00007fa1de70acad in default_map (meta=0x7fa1d4643a90, plane=0, info=0x7fa1daeff888, data=0x7fa1daeff868, stride=0x7fa1daeff818, flags=GST_MAP_READ) at ../gst-libs/gst/video/gstvideometa.c:231
#13 0x00007fa1de70b242 in gst_video_meta_map (meta=0x7fa1d4643a90, plane=0, info=0x7fa1daeff888, data=0x7fa1daeff868, stride=0x7fa1daeff818, flags=GST_MAP_READ) at ../gst-libs/gst/video/gstvideometa.c:373
#14 0x00007fa1de75af68 in gst_video_frame_map_id (frame=0x7fa1daeff7b0, info=0x7fa1d401a698, buffer=0x56498c07e6c0 [GstBuffer], id=-1, flags=GST_MAP_READ) at ../gst-libs/gst/video/video-frame.c:104
#15 0x00007fa1de75b410 in gst_video_frame_map (frame=0x7fa1daeff7b0, info=0x7fa1d401a698, buffer=0x56498c07e6c0 [GstBuffer], flags=GST_MAP_READ) at ../gst-libs/gst/video/video-frame.c:246
#16 0x00007fa1de58e6a4 in gst_va_base_dec_copy_output_buffer (base=0x56498c170ec0, codec_frame=0x7fa1d4129730) at ../sys/va/gstvabasedec.c:915
#17 0x00007fa1de5a500b in gst_va_vp9_dec_output_picture (decoder=0x56498c170ec0 [GstVaVp9Dec|vavp9dec0], frame=0x7fa1d4129730, picture=0x7fa1d5f67e80 [GstVp9Picture]) at ../sys/va/gstvavp9dec.c:499
#18 0x00007fa1de55b5fa in gst_vp9_decoder_handle_frame (decoder=0x56498c170ec0 [GstVaVp9Dec|vavp9dec0], frame=0x7fa1d4129730) at ../gst-libs/gst/codecs/gstvp9decoder.c:394
#19 0x00007fa1de6fabc6 in gst_video_decoder_decode_frame (decoder=0x56498c170ec0 [GstVaVp9Dec|vavp9dec0], frame=0x7fa1d4129730) at ../gst-libs/gst/video/gstvideodecoder.c:3703
#20 0x00007fa1de6f220f in gst_video_decoder_chain_forward (decoder=0x56498c170ec0 [GstVaVp9Dec|vavp9dec0], buf=0x7fa1d4647120 [GstBuffer], at_eos=0) at ../gst-libs/gst/video/gstvideodecoder.c:2332
#21 0x00007fa1de6f4522 in gst_video_decoder_chain (pad=0x56498c074d10 [GstPad|sink], parent=0x56498c170ec0 [GstVaVp9Dec|vavp9dec0], buf=0x7fa1d4647120 [GstBuffer])

If do not call that sync API, it works well.

HeJunyan, 2021-07-25:
It's on TGL.

wangyan-intel, 2021-07-26:
@Jexu Could you please take a look? Thanks.

Jexu, 2021-08-11:
@HeJunyan can you share the clip 'Smaller_VP9_209_extra_smaller_1.1.ivf' to me and then i can have a further debug.

HeJunyan, 2021-08-11:
[Uploading Smaller_VP9_209_extra_smaller_1.1.zip…]()

Jexu, 2021-08-12:
access to heap item in mediaSurfaceHeapElmt is out of boundary in func 'DdiDecode_StatusReport'. seem that the item is destroyed somewhere, but item number 'uiAllocatedHeapElements' is not updated.

Jexu, 2021-08-12:
@HeJunyan can you help verify this fix on your side?

HeJunyan, 2021-08-12:
Verified, OK for me. Thanks.

FocusLuo, 2021-09-29:
@wangyan-intel  @Xiaogangli-intel does this issue fixed and can be used?

PR #1243 (2021-11-09): [Decode] access volation when get status report
Fixes: #1227
commit 2d19c4db (2021-11-09): [Decode] access volation when get status report

Fixes: #1227
closed 2021-11-09 by intel-mediadev