intel/media-driver #1228: Support VAEncMiscParameterTemporalLayerStructure

rosetta-jpn, 2021-07-21:
VAEncMiscParameterTemporalLayerStructure [1] is essential to support temporal scalability encoding.
It is not handled by media-driver. 

[1] https://github.com/intel/libva/blob/6e86b4fb4dafa123b1e31821f61da88f10cfbe91/va/va.h#L2319
[2] https://github.com/intel/media-driver/blob/29d5f8fc118476dee84f3f784443badc819a3241/media_driver/linux/common/codec/ddi/media_ddi_encode_avc.cpp#L2113

rosetta-jpn, 2021-07-21:
It looks supported for VP8 and VP9.

[VP8] https://github.com/intel/media-driver/blob/c2ab2faf51d766840d8d2f3a8b5c1c716d993b3f/media_driver/linux/common/codec/ddi/media_ddi_encode_vp8.cpp#L547
[VP9]https://github.com/intel/media-driver/blob/c2ab2faf51d766840d8d2f3a8b5c1c716d993b3f/media_driver/linux/common/codec/ddi/media_ddi_encode_vp9.cpp#L1033

XinfengZhang, 2021-07-25:
the story is different, actually, media driver support multiple temporal layer encoding for H264, VP8, VP9. but the story is different.  for H264, all of the temporal layer is controlled by application, media driver get the reference relationship from the PPS, SliceParameter of each frame. but it is transparent to media_dirver. media driver does not treat it in a different way.

for VP8 & VP9. it is not transparent to media driver ,  the ddi is not fully implemented, it should support temporal layer BRC. but this feature is not enabled now.

rosetta-jpn, 2021-07-26:
I thought this structure was essential to control bitrate properly within the driver.
Am I wrong? Then, don't we need to pass this structure to a driver in H264 temporal layer encoding?

XinfengZhang, 2021-07-26:
it depends on whether there are multiple temporal layers BRC.  for AVC, it just need VAEncMiscParameterTypeRateControl and VAEncMiscParameterTypeFrameRate, because it just support per-sequence BRC. 

actually, VP8/VP9 should report VAConfigAttribEncRateControlExt and accept VAConfigAttribValEncRateControlExt

rosetta-jpn, 2021-07-26:
Thanks for the clarification.

rosetta-jpn, 2021-07-28:
Sorry, I misunderstood Carl's comment.

> it just support per-sequence BRC.

This means media-driver doesn't support temporal layer BRC.
So H.264 temporal layer encoding is not doable with media-driver.

Can I ask driver folks to support it?

XinfengZhang, 2021-07-28:
which platform?

rosetta-jpn, 2021-07-28:
WhiskeyLake, CometLake, JasperLake and TigerLake.

Specifically Chrome, VME encoder is used on WhiskeyLake and CometLake, and VDEnc is used on JasperLake and TigerLake.

XinfengZhang, 2021-08-04:
@rosetta-jpn as offline discussion , seems this issue could be closed?

closed 2021-08-06 by rosetta-jpn