intel/media-driver #1231: Crash on Intel Iris Xe card.

bbbear2002, 2021-07-22:
Hi, this intel media driver is great. It works fine on my intel 10th gen integrated card. But on intel gen 11 card (Iris Xe), the browser keeps crashing. Any clue of solution for that? Thank you very much.

XinfengZhang, 2021-07-25:
hi @bbbear2002 , no enough information, you mean you are using browser (hw acceleration enabled) to play some video? could you share your HW configuration, OS version, kernel version, also the browser version?

bbbear2002, 2021-08-07:
Hi XinfengZhang,
PC: XiaomiBook Pro 14
OS: Deepin 20.2.2
Kernel: 5.12.14
CPU: 11th-Gen i5-11300H
Browser: Chrome 92.0.4515.131
Exec=/usr/bin/google-chrome-stable --use-gl=desktop --enable-features=VaapiVideoDecoder %U is added in the .desktop file
SwithGraphicCard from Xinghuo Appstore is installed

bbbear2002, 2021-08-07:
Graphics Feature Status
Canvas: Hardware accelerated
Compositing: Hardware accelerated
Multiple Raster Threads: Enabled
Out-of-process Rasterization: Hardware accelerated
OpenGL: Enabled
Rasterization: Hardware accelerated on all pages
Skia Renderer: Enabled
Video Decode: Hardware accelerated
Vulkan: Disabled
WebGL: Hardware accelerated
WebGL2: Hardware accelerated
Driver Bug Workarounds
adjust_src_dst_region_for_blitframebuffer
clear_uniforms_before_first_program_use
count_all_in_varyings_packing
disable_post_sub_buffers_for_onscreen_surfaces
exit_on_context_lost
msaa_is_slow
rely_on_implicit_sync_for_swap_buffers
disabled_extension_GL_KHR_blend_equation_advanced
disabled_extension_GL_KHR_blend_equation_advanced_coherent
disabled_extension_GL_MESA_framebuffer_flip_y
Problems Detected
Clear uniforms before first program use on all platforms: 124764, 349137
Applied Workarounds: clear_uniforms_before_first_program_use
Mesa drivers in Linux handle varyings without static use incorrectly: 333885
Applied Workarounds: count_all_in_varyings_packing
Disable partial swaps on Mesa drivers (detected with GL_RENDERER): 339493
Applied Workarounds: disable_post_sub_buffers_for_onscreen_surfaces
On Intel GPUs MSAA performance is not acceptable for GPU rasterization: 527565
Applied Workarounds: msaa_is_slow
Disable partial swaps on Mesa drivers (detected with GL_VERSION): 339493
Applied Workarounds: disable_post_sub_buffers_for_onscreen_surfaces
adjust src/dst region if blitting pixels outside framebuffer on Linux Intel: 664740
Applied Workarounds: adjust_src_dst_region_for_blitframebuffer
Disable KHR_blend_equation_advanced until cc shaders are updated: 661715
Applied Workarounds: disable(GL_KHR_blend_equation_advanced), disable(GL_KHR_blend_equation_advanced_coherent)
Some drivers can't recover after OUT_OF_MEM and context lost: 893177
Applied Workarounds: exit_on_context_lost
Avoid waiting on a egl fence before swapping buffers and rely on implicit sync on Intel GPUs: 938286
Applied Workarounds: rely_on_implicit_sync_for_swap_buffers
Disable GL_MESA_framebuffer_flip_y for desktop GL: 964010
Applied Workarounds: disable(GL_MESA_framebuffer_flip_y)
Version Information
Data exported	2021-08-07T06:32:08.325Z
Chrome version	Chrome/92.0.4515.131
Operating system	Linux 5.12.14-amd64-desktop
Software rendering list URL	https://chromium.googlesource.com/chromium/src/+/6b8d6c56ce21e38a72f7c4becb5abc1fa5134f29/gpu/config/software_rendering_list.json
Driver bug list URL	https://chromium.googlesource.com/chromium/src/+/6b8d6c56ce21e38a72f7c4becb5abc1fa5134f29/gpu/config/gpu_driver_bug_list.json
ANGLE commit id	d564abe2dada
2D graphics backend	Skia/92 8e69798f33b4a4b777476dcdb7aa2c75177d07ac
Command Line	/usr/bin/google-chrome-stable --use-gl=desktop --enable-features=VaapiVideoDecoder --enable-crashpad --flag-switches-begin --enable-gpu-rasterization --ignore-gpu-blocklist --flag-switches-end --origin-trial-disabled-features=SecurePaymentConfirmation
Driver Information
Initialization time	348
In-process GPU	false
Passthrough Command Decoder	false
Sandboxed	false
GPU0	VENDOR= 0x10de, DEVICE=0x1f97
GPU1	VENDOR= 0x8086, DEVICE=0x9a49 *ACTIVE*
Optimus	true
AMD switchable	false
Driver vendor	Mesa
Driver version	20.3.4
GPU CUDA compute capability major version	0
Pixel shader version	4.60
Vertex shader version	4.60
Max. MSAA samples	16
Machine model name	
Machine model version	
[... 22 lines cut ...]
RGBX_8888	Software only
RGBA_8888	Software only
BGRX_8888	Software only
BGRA_1010102	Software only
RGBA_1010102	Software only
BGRA_8888	Software only
RGBA_F16	Software only
YVU_420	Software only
YUV_420_BIPLANAR	Software only
P010	Software only
Display(s) Information
Info	Display[0] bounds=[0,0 1280x800], workarea=[0,0 1280x766], scale=2, rotation=0, panel_rotation=0 external.
Color space (all)	{primaries:BT709, transfer:IEC61966_2_1, matrix:RGB, range:FULL}
Buffer format (all)	BGRA_8888
SDR white level in nits	100
Bits per color component	8
Bits per pixel	24
Refresh Rate in Hz	120
Video Acceleration Information
Vulkan Information
Device Performance Information
Log Messages
[6414:6414:0807/135933.405160:ERROR:gpu_init.cc(441)] : Passthrough is not supported, GL is desktop
[6414:6414:0807/135933.511410:ERROR:sandbox_linux.cc(374)] : InitializeSandbox() called with multiple threads in process gpu-process.
[6414:6414:0807/140043.845564:ERROR:gl_surface_presentation_helper.cc(260)] : GetVSyncParametersIfAvailable() failed for 1 times!

bbbear2002, 2021-08-07:
On my Redmibook Air 13 it works perfectly well.

bbbear2002, 2021-08-07:
The vainfo is correct.
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri//iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.4.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.1 (3b18136a)
vainfo: Supported profile and entrypoints
      VAProfileNone                   : VAEntrypointVideoProc
      VAProfileNone                   : VAEntrypointStats
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Simple            : VAEntrypointEncSlice
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointFEI
      VAProfileH264Main               : VAEntrypointEncSliceLP
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointFEI
      VAProfileH264High               : VAEntrypointEncSliceLP
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline: VAEntrypointFEI
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointFEI
      VAProfileHEVCMain               : VAEntrypointEncSliceLP
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointEncSliceLP
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileVP9Profile1            : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileVP9Profile3            : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointEncSlice
      VAProfileHEVCMain422_10         : VAEntrypointVLD
      VAProfileHEVCMain422_10         : VAEntrypointEncSlice
      VAProfileHEVCMain422_12         : VAEntrypointVLD
      VAProfileHEVCMain422_12         : VAEntrypointEncSlice
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointEncSliceLP
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointVLD
      VAProfileHEVCSccMain            : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain10          : VAEntrypointVLD
      VAProfileHEVCSccMain10          : VAEntrypointEncSliceLP
      VAProfileHEVCSccMain444         : VAEntrypointVLD
      VAProfileHEVCSccMain444         : VAEntrypointEncSliceLP
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointVLD
      VAProfileHEVCSccMain444_10      : VAEntrypointEncSliceLP

XinfengZhang, 2021-08-18:
sorry for late response. from the information 
1. there are two GPU device,  GPU1 is from intel
GPU0 VENDOR= 0x10de, DEVICE=0x1f97
GPU1 VENDOR= 0x8086, DEVICE=0x9a49 ACTIVE
2.  vainfo works correct, but browser video acceleration failed.
3. the platform is TGL, gpu device is 0x9a49  https://github.com/intel/media-driver/blob/master/media_driver/linux/gen12/ddi/media_sysinfo_g12.cpp#L466

I wonder whether it related with 
https://github.com/intel/libva/issues/536 suspicious one
or
https://github.com/intel/libva/issues/278

could you help to run some command line:
1. ls /dev/dri
2. xdpyinfo | grep DRI

also, could you export LIBVA_TRACE to get the log of libva to check which step caused the crash?

bbbear2002, 2021-08-22:
by-path  card0  card1  renderD128  renderD129

    DRI2
    DRI3

I don't know how to use LIBVA_TRACE

bbbear2002, 2021-08-22:
[62643.472976][ctx       none]==========va_TraceInitialize
[62643.472988][ctx       none]=========vaInitialize ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473024][ctx       none]=========vaQueryConfigProfiles ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473029][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473043][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473053][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473063][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473082][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473101][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473106][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473112][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473117][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473127][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473146][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473152][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473172][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473186][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473192][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473198][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473204][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473210][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473220][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473230][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473240][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473251][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473261][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473267][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473277][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473287][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473298][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473304][ctx       none]=========vaQueryConfigEntrypoints ret = VA_STATUS_SUCCESS, success (no error) 
[62643.473937][ctx       none]==========va_TraceTerminate
[62643.473981][ctx       none]=========vaTerminate ret = VA_STATUS_SUCCESS, success (no error)

XinfengZhang, 2021-08-23:
> [62643.472976][ctx none]==========va_TraceInitialize
> [62643.472988][ctx none]=========vaInitialize ret = VA_STATUS_SUCCESS, success (no error)
> [62643.473024][ctx none]=========vaQueryConfigProfiles [...]

looks like it is libva trace generated by vainfo. not your case

bbbear2002, 2021-08-23:
Then how to use LIBVA_TRACE?

XinfengZhang, 2021-08-24:
export LIBVA_TRACE=logfile

bbbear2002, 2021-08-29:
And then where can i find the logfile?

bbbear2002, 2021-09-22:
It seems that LIBVA_TRACE didn't export anything when i opened the stream media.

catakot, 2021-12-21:
I have same issue. Any ideas how to fix it?

- Windows 11
- 11th Gen Intel(R) Core(TM) i7-11370H @ 3.30GHz   3.30 GHz

Here is crashpad report

`PROCESS_NAME:  chrome.exe

ERROR_CODE: (NTSTATUS) 0xc00000fd - **A new guard page for the stack cannot be created.**

EXCEPTION_CODE_STR:  c00000fd

EXCEPTION_PARAMETER1:  0000000000000001

EXCEPTION_PARAMETER2:  0000009773401ff8

ADDITIONAL_DEBUG_TEXT:  Followup set based on attribute [Is_OriginalExceptionThread] from Frame:[0] on thread:[4c84]`

catakot, 2022-02-05:
@bbbear2002, after months of searching I found the problem. In my case update of Citrix Workspace solved the problem.

XinfengZhang, 2022-02-06:
LIBVA_TRACE could specify the location, also , if you set the LIBVA_TRACE=/tmp/aa , it will generate file with thread id followed.

bbbear2002, 2022-02-08:
Hi, after updating to Debian 11 (Deepin v23) the issues are resolved. Thank you all for your gracious input.

closed 2022-09-06 by Xiaogangli-intel