intel/media-driver #1232: Align a requested surface resolution by 16 for Encoder

rosetta-jpn, 2021-07-28:
Media-driver doesn't align a requested surface resolution by 16 for encoder.
In fact, unless I aligned the surface resolution by 16 in app side, encoding fails.
I think the width and height both should be aligned by 16.
https://github.com/intel/media-driver/blob/master/media_driver/linux/common/ddi/media_libva_util.cpp#L279-L312-

XinfengZhang, 2021-08-04:
the problem is : when you call vaCreateSurfaces, the codec type is not defined yet.  
for example: AVC it should align with 16, HEVC may align with 8.  if we use 16 for all codec. it will waste some memory. is it ok?

rosetta-jpn, 2021-08-04:
I think vaCreateSurfaces() with a current encode profile is called before vaCreateSurfaces(). Can vaCreateSurfaces() know the current profile from some internal variable?

XinfengZhang, 2021-08-04:
vaCreateContext may be called before vaCreateSurfaces or after. it depends on the behavior of application.

XinfengZhang, 2021-08-09:
yes, it is , I am considering another solution, if you set the VA_SURFACE_ATTRIB_USAGE_HINT_ENCODER, we could just record the requirement, but do not allocate the memory, until it is used (vaEndPicture). but the risk is:  how to handle this scenario, someone call vaCreateSurface, then upload data into it before it is used. ...

rosetta-jpn, 2021-08-17:
I would not like to introduced the complexity to handle this alignment problem.
Maybe we should begin with just aligning by 16 regardless of codec, and thereafter look for a better way to avoid the unnecessary alignment.

rosetta-jpn, 2021-08-19:
Hi Carl, any update?

XinfengZhang, 2021-10-21:
merged by 78cddcb692f9e14fa62727f7848200f43e397cd5