intel/media-driver #1240: [ffmpeg][vp8_vaapi encoder] Ignore alpha channel when encoding rawvideo frames to webm.

alittleboys, 2021-08-09:
I use FFMPEG vp8_vaapi to encode a RGBA raw video stream and output a webm stream. But the transparency information of this is lost -- it shows a black background instead of the expected transparent.

My ffmpeg command is as follows:
`ffmpeg -vaapi_device /dev/dri/renderD128 -f rawvideo -pix_fmt rgba -r 30 -s 1280x720 -i - -vf 'hwupload' -c:v vp8_vaapi -b:v 2M -maxrate 2M -metadata:s:v:0 alpha_mode="1" -f webm -`

I use libvpx to decode the stream generated by above. Finally it shows a black background instead of transparent.

I'm sure that the alpha values of raw video frames are effective: I try encode the same raw video stream by libvpx , it shows a transparent background!

Is the alpha value neglected during encode a video supported alpha? 

vainfo:
libva info: VA-API version 1.13.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/local/lib/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.13.0.pre1)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 21.3.1 (36e99728)
vainfo: Supported profile and entrypoints
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileNone                   :	VAEntrypointStats
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointFEI
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointFEI
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointFEI
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointFEI
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointEncSlice
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD

alittleboys, 2021-08-09:
I check the pixel format of webm video encoded by vp8_vaapi.

The output of ffmpeg is as follows:
`Input #0, matroska,webm, from './test_output_hw.webm':`
 ` Metadata:`
    `        ENCODER         : Lavf58.12.100`
 `         Duration: 00:00:10.00, start: 0.000000, bitrate: 62 kb/s`
   `         Stream #0:0: Video: vp8, yuv420p(progressive), 1280x720, SAR 1:1 DAR 16:9, 30 fps, 30 tbr, 1k tbn, 1k tbc (default)`
    `        Metadata:`
      `        ALPHA_MODE      : "1"`
     `         ENCODER         : Lavc58.18.100 vp8_vaapi`
     `        DURATION        : 00:00:10.000000000`

It means the alpha channel is not saved when encoding the video. Could I save it with any special setting or patches?

XinfengZhang, 2021-08-16:
codec does not care about the alpha channel, it means , when it become to an encoded bitstream, it will lost the alpha channel, if you want store the alpha channel, suppose it should be handled in upper layer, such as ffmpeg level

closed 2021-10-17 by XinfengZhang