intel/media-driver #1245: [regression][DG1] catch regression on encode/avc on gst-vaapi

Bin-CI, 2021-08-13:
catch regression at fdd0557866b4

uartie, 2021-08-16:
please provide more details.

uartie, 2021-08-16:
> please provide more details.

Before result...  After result...  Command line... etc.

Bin-CI, 2021-08-16:
Test case command line:
```
gst-launch-1.0 -vf filesrc location=1080p_blue_sky_1920x1080_I420.yuv num-buffers=100 ! rawvideoparse format=i420 width=1920 height=1080 framerate=30 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! vaapih264enc  rate-control=cbr keyframe-period=30 num-slices=1 max-bframes=0 bitrate=8000 tune=none ! video/x-h264,profile=main ! h264parse ! filesink location=_0.test.gst-vaapi.encode.avc/cbr/bluesky-cbr-main-30-30-1-0-8000k-8000k.h264

gst-launch-1.0 -vf filesrc location=1080p_blue_sky_1920x1080_I420.yuv num-buffers=100 ! rawvideoparse format=i420 width=1920 height=1080 framerate=30 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! vaapih264enc  rate-control=cbr keyframe-period=30 num-slices=1 max-bframes=0 bitrate=8000 tune=none ! video/x-h264,profile=high ! h264parse ! filesink location=_0.test.gst-vaapi.encode.avc/cbr/bluesky-cbr-high-30-30-1-0-8000k-8000k.h264

gst-launch-1.0 -vf filesrc location=1080p_blue_sky_1920x1080_I420.yuv num-buffers=100 ! rawvideoparse format=i420 width=1920 height=1080 framerate=30 ! videoconvert chroma-mode=none dither=0 ! video/x-raw,format=NV12 ! vaapih264enc  rate-control=vbr keyframe-period=30 quality-level=4 num-slices=3 bitrate=11428 refs=1 tune=low-power ! video/x-h264,profile=main ! h264parse ! filesink location=_0.test.gst-vaapi.encode.avc/vbr_lp/bluesky-vbr-main-30-30-3-4-8000k-11428k-1-low-power.h264
```

Failed reason::
```
INFO: slash: DETAIL: drv.iHD.psnr:expect = [37.1395, 36.9731, 39.3962, 40.1698, 39.9793, 41.8595]
INFO: slash: DETAIL: drv.iHD.psnr:actual = [37.2682, 36.9115, 39.1778, 39.9905, 39.7476, 41.6404]

INFO: slash: DETAIL: drv.iHD.psnr:expect = [37.1395, 36.9731, 39.3962, 40.1698, 39.9793, 41.8595]
INFO: slash: DETAIL: drv.iHD.psnr:actual = [37.1807, 36.9715, 39.2416, 40.001, 39.7409, 41.6363]

INFO: slash: DETAIL: drv.iHD.psnr:expect = [37.2757, 37.2677, 39.4639, 39.7486, 39.724, 41.5355]
INFO: slash: DETAIL: drv.iHD.psnr:actual = [37.3435, 37.2021, 39.3852, 39.4406, 39.4532, 41.2898]
```

Bin-CI, 2021-09-07:
issue fixed

closed 2021-09-07 by Bin-CI