intel/media-driver #1247: Segfault from MmdDevice::CreateFactory

KenMacD, 2021-08-13:
In testing some hardware acceleration stuff in Firefox I've run to the following segfault:

```
* thread #1, name = 'Web Content', stop reason = signal SIGSEGV
    frame #0: 0x0000000000000000
  * frame #1: 0x00007f7a6c85c338 iHD_drv_video.so`MmdDevice::CreateFactory(_MOS_OS_CONTEXT*) + 392
    frame #2: 0x00007f7a6c938ecb iHD_drv_video.so`MosDecompression::MosDecompression(_MOS_OS_CONTEXT*) + 43
    frame #3: 0x00007f7a6c92ef97 iHD_drv_video.so`OsContextSpecificNext::Init(void*) + 1239
    frame #4: 0x00007f7a6c84803d iHD_drv_video.so`DdiMedia_InitMediaContext(VADriverContext*, int, int*, int*) + 381
    frame #5: 0x00007f7a7f446eac libva.so.2`va_openDriver + 620
    frame #6: 0x00007f7a7f44801b libva.so.2`vaInitialize + 347
    frame #7: 0x00007f7a9fc995d8 libxul.so`___lldb_unnamed_symbol104288$$libxul.so + 680
    frame #8: 0x00007f7a9fc6729a libxul.so`___lldb_unnamed_symbol103771$$libxul.so + 122
    frame #9: 0x00007f7a9fc6e95e libxul.so`___lldb_unnamed_symbol103854$$libxul.so + 46
    frame #10: 0x00007f7a9dfa0f35 libxul.so`___lldb_unnamed_symbol20749$$libxul.so + 421
    frame #11: 0x00007f7a9dfae42c libxul.so`___lldb_unnamed_symbol20976$$libxul.so + 748
    frame #12: 0x00007f7a9dfa8dbd libxul.so`___lldb_unnamed_symbol20905$$libxul.so + 1677
    frame #13: 0x00007f7a9dfacf16 libxul.so`___lldb_unnamed_symbol20959$$libxul.so + 70
    frame #14: 0x00007f7a9e3a1d98 libxul.so`___lldb_unnamed_symbol35159$$libxul.so + 120
    frame #15: 0x00007f7a9e3641c8 libxul.so`___lldb_unnamed_symbol34286$$libxul.so + 88
    frame #16: 0x00007f7a9dfa6c21 libxul.so`___lldb_unnamed_symbol20875$$libxul.so + 321
    frame #17: 0x00007f7aa7137b90 libnspr4.so`_pt_root + 256
    frame #18: 0x00007f7aa759ed3e libpthread.so.0`start_thread + 206
    frame #19: 0x00007f7aa725743f libc.so.6`__clone + 63
```

With mapped in memory:

```
0x00007f7a7f443000 /nix/store/v1x9cggyb408h49n9nhpwlzni4qdfhvx-libva-2.12.0/lib/libva.so.2 
0x00007f7a6c3d2000 /nix/store/91f537nz7rc2w4hns81yqk79611d3blg-intel-media-driver-21.3.1/lib/dri/iHD_drv_video.so
```

I don't have debug symbols but from [the looks of things](https://github.com/intel/media-driver/blob/68f6f2bc8f5c8e69c3a010931f7940d4a7937746/media_driver/linux/common/media_interfaces/media_interfaces.cpp#L336) it looks to me like there's an attempt to call `osInterface->pfnDestroy()` in the cleanup, likely from a failure in the `Mos_InitInterface()` call that didn't set that function.

XinfengZhang, 2021-08-15:
please provide the platform information, thanks

KenMacD, 2021-08-15:
@XinfengZhang Included, please let me know if you need more info:

```
System:
  Kernel: 5.13.9 x86_64
    bits: 64
    compiler: gcc
      v: 10.3.0
  Desktop: sway 1.6.1
  Distro: NixOS 21.11 (Porcupine)

Machine:
  Type: Laptop
  System: Dell
    product: Latitude 7420
    v: N/A
    serial: <filter>
  Mobo: Dell
    model: 0125XH
    v: A00
    serial: <filter>
  UEFI: Dell
    v: 1.8.2
    date: 07/29/2021

CPU:
  Info: Quad Core
    model: 11th Gen Intel Core i7-1185G7
    bits: 64
    type: MT MCP
    arch: Tiger Lake
    rev: 1
    cache: 
      L2: 12 MiB
    flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
    bogomips: 28876
  Speed: 1102 MHz
    min/max: 400/4800 MHz
    Core speeds (MHz): 
      1: 1102
      2: 1137
      3: 930
      4: 600
      5: 524
      6: 595
      7: 1053
      8: 546

Graphics:
  Device-1: Intel TigerLake-LP GT2 [Iris Xe Graphics]
    vendor: Dell
    driver: i915
      v: kernel
    bus-ID: 00:02.0
  Device-2: Sunplus Innovation Integrated_Webcam_FHD
    type: USB
    driver: uvcvideo
    bus-ID: 3-6:3
  Display: wayland
    server: N/A
    compositor: sway
    driver: 
      loaded: modesetting
      unloaded: vesa
    resolution: <missing: xdpyinfo>
  OpenGL: 
    renderer: Mesa Intel Xe Graphics (TGL GT2)
    v: 4.6 Mesa 21.1.5
    direct render: Yes
```

XinfengZhang, 2021-08-16:
suppose this platform should not enable MMC, @Guangyao-Bai , could you help to check it?

Guangyao-Bai, 2021-08-17:
Hi  @XinfengZhang , it's about the MMC Device init fail, not a real compression HW func issue.

@KenMacD do you have more details to repro such issue? we need to check the reason

KenMacD, 2021-08-30:
@Guangyao-Bai I was following the settings in [this reply on the pipewire repo](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/460#note_720627). Specifically I set:

`~/.drirc`:

```
<driconf>
    <device>
        <application name="Firefox" executable="firefox">
            <option name="mesa_glthread" value="false" />
        </application>
    </device>
</driconf>
```

`~/.mozilla/firefox/<PROFILE>/user.js`:
```
// things that completely break FF
user_pref("layers.async-pan-zoom.enabled", false);
user_pref("gfx.use-glx-texture-from-pixmap", false);
user_pref("gfx.xrender.enabled", false);
user_pref("gfx.prefer-mesa-llvmpipe", false);
user_pref("gfx.work-around-driver-bugs", false);
// ffvpx interferes with ffmpeg that is required for GPU offloading
user_pref("media.ffvpx.enabled", false);
user_pref("media.gpu-process-decoder", true);
user_pref("media.hardware-video-decoding.enabled", true);
user_pref("media.ffmpeg.vaapi.enabled", true);
user_pref("media.ffmpeg.vaapi-drm-display.enabled", true);
user_pref("media.ffmpeg.dmabuf-textures.enabled", true);
user_pref("widget.dmabuf-webgl.enabled", true);
user_pref("widget.wayland-dmabuf-vaapi.enabled", true);
user_pref("widget.wayland-dmabuf-video-textures.enabled", true);
user_pref("widget.wayland-dmabuf-webgl.enabled", true);
user_pref("widget.wayland_dmabuf_backend.enabled", true);
// av1 support, vp9 replacement
user_pref("media.av1.enabled", true);
user_pref("image.avif.enabled", true);
// CPU multithreading
user_pref("browser.tabs.remote.autostart", true);
user_pref("browser.tabs.remote.force-enable", false);
user_pref("browser.tabs.remote.force-disable", false);
user_pref("browser.tabs.remote.separateFileUriProcess", true);
user_pref("browser.tabs.remote.separatePrivilegedContentProcess", true);
user_pref("browser.tabs.remote.useCrossOriginOpenerPolicy", true);
user_pref("browser.tabs.remote.useCrossOriginPolicy", true);
user_pref("dom.ipc.processCount", 6);
user_pref("dom.ipc.processCount.file", 2);
user_pref("dom.ipc.processCount.webLargeAllocation", 15);
user_pref("dom.maxHardwareConcurrency", 5);
user_pref("extensions.webextensions.remote", true);
// CPU process for network handling
user_pref("network.process.enabled", true);
// CPU process for GPU handling
user_pref("gfx.canvas.remote", true);
user_pref("layers.gpu-process.allow-software", true);
user_pref("layers.gpu-process.enabled", true);
user_pref("layers.offmainthreadcomposition.enabled", true);
user_pref("webgl.out-of-process", true);
```

Then I've browsed to a site the plays a video. For example [this html5 video tester](https://tools.woolyss.com/html5-audio-video-tester/?u=woolyss.com/f/Big_Buck_Bunny_Trailer.m4v) will always cause the segfault for me.

XinfengZhang, 2021-12-05:
I guess the issue is similar with #854, #1307, #1283, #1289. 
if you are using the older platform before TGL, please refer the PR #1293 , if your platform is TGL, suppose latest media driver master  (with commit 4e70cd6 ) should work

Xiaogangli-intel, 2022-09-06:
Hi @KenMacD, so 4e70cd6 fixed your issue, right? If yes, I will close this issue.

KenMacD, 2022-09-08:
@Xiaogangli-intel it appears to me to be fixed. I re-tested with the exact same settings and I no longer get the crash. Thanks!

closed 2022-09-08 by KenMacD