intel/media-driver #1248: init failed

Amalsarr, 2021-08-19:
vainfo (as a regular user)
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
[MOS]:  CRITICAL - HWInfo_GetGfxInfo:140: Failed to get the device info for Device id: 22b1

[LIBVA]:CRITICAL - DdiMedia_InitMediaContext:1777: Fatal error - unsuccesfull Sku/Wa/GtSystemInfo initialization
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.6.0)
vainfo: Driver version: Intel i965 driver for Intel(R) CherryView - 2.4.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD

as a root : 
vainfo
error: XDG_RUNTIME_DIR not set in the environment.
libva info: VA-API version 1.13.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit

where did I go wrong? any insight is highly appreciated.

uartie, 2021-08-19:
This media driver (iHD) does not support Cherryview (see https://github.com/intel/media-driver#supported-platforms).  Only the i965 media driver supports this platform (see https://github.com/intel/intel-vaapi-driver/blob/master/README).

uartie, 2021-08-19:
Try setting LIBVA_DRIVER_NAME=i965 as both regular and root user.

Amalsarr, 2021-08-19:
What thread and what’s the solution to this?

Amalsarr, 2021-08-19:
> Try setting LIBVA_DRIVER_NAME=i965 as both regular and root user.

What is the LIBVA_DRIVER_NAME? I remember setting it to iHD like this 
Export libva=iHD
But I wasn’t so sure of what i did, could you please assure me if the libva driver neans the libva directory altho I don’t think this makes any sense but I’m really confused because this was the only one to work out for me i tried to set the drive that ends in .os extension but it didn’t work for me.

Amalsarr, 2021-08-19:
i did it and here's the results as a regular user : 
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
[MOS]:  CRITICAL - HWInfo_GetGfxInfo:140: Failed to get the device info for Device id: 22b1

[LIBVA]:CRITICAL - DdiMedia_InitMediaContext:1777: Fatal error - unsuccesfull Sku/Wa/GtSystemInfo initialization
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
libva info: va_openDriver() returns 1
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/i965_drv_video.so
libva info: Found init function __vaDriverInit_1_6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.13 (libva 2.6.0)
vainfo: Driver version: Intel i965 driver for Intel(R) CherryView - 2.4.0
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Simple            :	VAEntrypointEncSlice
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointEncSlice
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSlice
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSlice
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSlice
      VAProfileH264MultiviewHigh      :	VAEntrypointVLD
      VAProfileH264MultiviewHigh      :	VAEntrypointEncSlice
      VAProfileH264StereoHigh         :	VAEntrypointVLD
      VAProfileH264StereoHigh         :	VAEntrypointEncSlice
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointEncSlice
      VAProfileHEVCMain               :	VAEntrypointVLD

XinfengZhang, 2021-10-17:
cherryview is supported by https://github.com/intel/intel-vaapi-driver , not by intel media driver. so suppose the issue was resolved. thanks

closed 2021-10-17 by XinfengZhang