intel/media-driver #1250: TGL:  Multi-sessions low-power HEVC and AVC transcoding failed

dwang26, 2021-08-23:
It can be reproduced  by sample_multi_transcode on Media SDK 21.2.3/libva2.12/Media-driver21.2.3 or Media SDK 21.1.3/libva2.11/Media-driver21.1.3.  
The test par file is like below:
-i::h264 /home/work/video/2.h264 -o::h265 /tmp/1.h265 -qsv-ff -b 4000 
-i::h264 /home/work/video/2.h264 -o::h265 /tmp/2.h265 -qsv-ff -b 4000 
-i::h264 /home/work/video/3.h264 -o::h264 /tmp/3.h264 -qsv-ff -b 4000 

System information:
Ubuntu 18.04
Linux kernel version: Intel LTS  5.10.41 (https://github.com/intel/linux-intel-lts/releases/tag/lts-v5.10.41-yocto-210707T232614Z)
guc firmware: tgl_guc_49.0.1.bin huc firmware: tgl_huc_7.5.0.bin

This problem only happen when there are both lowpower HEVC and AVC transcoding sessions. Only lowpower HEVC or AVC transcoding can work well.

I also tried latest Media-driver and still see the error:

$sample_multi_transcode -par enc.par
Multi Transcoding Sample Version 8.4.27.0

Par file is: enc.par

libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
Session 0:
Pipeline surfaces number (DecPool): 3
Input  video: AVC
Output video: HEVC

Session 1:
Pipeline surfaces number (DecPool): 3
Input  video: AVC
Output video: HEVC

Session 2:
[WARNING] Configuration changed on the Query() call
mfx.BRCParamMultiplier:0 changed to mfx.BRCParamMultiplier:1
ext.CDO2.BRefType:2 changed to ext.CDO2.BRefType:1
Pipeline surfaces number (DecPool): 3
Input  video: AVC
Output video: AVC

Session 0 was NOT joined with other sessions
Session 1 was NOT joined with other sessions
Session 2 was NOT joined with other sessions

Transcoding started
......
[ERROR], sts=MFX_ERR_DEVICE_FAILED(-17), PutBS, Encode: SyncOperation failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1950

[ERROR], sts=MFX_ERR_ABORTED(-12), Transcode, PutBS failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1912

[ERROR], sts=MFX_ERR_ABORTED(-12), Run, CTranscodingPipeline::Run::Transcode() [0x55c29adb05a0] failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:4584

 session 0 [0x55c29adb05a0] failed with status MFX_ERR_ABORTED shutting down the application...

session [0x55c29adb05a0] m_bForceStop is set
session [0x55c29b394290] m_bForceStop is set
session [0x55c29b7f3550] m_bForceStop is set
WARNING: m_bForceStop is set, returning NULL ptr from GetFreeSurface
WARNING: m_bForceStop is set, returning NULL ptr from GetFreeSurface

Transcoding finished

Common transcoding time is 0.994717 sec
-------------------------------------------------------------------------------
*** session 0 [0x55c29adb05a0] FAILED (MFX_ERR_ABORTED) 0.977266 sec, 177 frames, 181.118 fps
-i::h264 /home/work/video/2.h264 -o::h265 T00.h265 -qsv-ff -b 4000 -dist 1

*** session 1 [0x55c29b394290] PASSED (MFX_ERR_NONE) 0.994559 sec, 184 frames, 185.007 fps
-i::h264 /home/work/video/2.h264 -o::h265 T11.h265 -qsv-ff -b 4000 -dist 1

*** session 2 [0x55c29b7f3550] PASSED (MFX_ERR_NONE) 0.994178 sec, 185 frames, 186.083 fps
-i::h264 /home/work/video/3.h264 -o::h264 T22.h264 -qsv-ff -b 4000 -dist 1

-------------------------------------------------------------------------------

The test FAILED

XinfengZhang, 2021-08-23:
hi @dwang26 , is there hang or crash?  could you help to check the dmesg to show whether there are reset/hang?

dwang26, 2021-08-23:
Hi Xinfeng,
No gpu hang or crash in kernel log.  This issue can be reproduced every time I run the test.
I'm using a 1080p h264 clip as input. It has 1000 frames.

Xiaogangli-intel, 2021-08-30:
@dwang26 Could you try this PR #1256 to check whether it can help on this issue?

dwang26, 2021-08-30:
@Xiaogangli-intel I just tried this patch with latest Media-driver but I still can see the error. With this patch, the low-power HEVC encoding failed on 1st frame.  

Even without running together with low-power AVC encoding, the low-power HEVC encoding still failed. No error or GPU hang message in dmesg.

$sample_multi_transcode -par enc.par
Multi Transcoding Sample Version 8.4.27.0

Par file is: enc.par

libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
libva info: VA-API version 1.12.0
libva info: User environment variable requested driver 'iHD'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_12
libva info: va_openDriver() returns 0
Session 0:
Pipeline surfaces number (DecPool): 6
Input  video: AVC
Output video: HEVC

Session 1:
Pipeline surfaces number (DecPool): 6
Input  video: AVC
Output video: HEVC

Session 2:
[WARNING] Configuration changed on the Query() call
mfx.BRCParamMultiplier:0 changed to mfx.BRCParamMultiplier:1
mfx.GopRefDist:4 changed to mfx.GopRefDist:1
mfx.TimeStampCalc:4 changed to mfx.TimeStampCalc:1
mfx.JPEGColorFormat:4 changed to mfx.JPEGColorFormat:1
mfx.RestartInterval:4 changed to mfx.RestartInterval:1
ext.CDO2.BRefType:2 changed to ext.CDO2.BRefType:1
Pipeline surfaces number (DecPool): 3
Input  video: AVC
Output video: AVC

Session 0 was NOT joined with other sessions
Session 1 was NOT joined with other sessions
Session 2 was NOT joined with other sessions

Transcoding started
.
[ERROR], sts=MFX_ERR_GPU_HANG(-21), PutBS, Encode: SyncOperation failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1950

[ERROR], sts=MFX_ERR_ABORTED(-12), Transcode, PutBS failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1912

[ERROR], sts=MFX_ERR_ABORTED(-12), Run, CTranscodingPipeline::Run::Transcode() [0x55ac76730160] failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:4584

[ERROR], sts=MFX_ERR_GPU_HANG(-21), PutBS, Encode: SyncOperation failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1950

[ERROR], sts=MFX_ERR_ABORTED(-12), Transcode, PutBS failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1912

[ERROR], sts=MFX_ERR_ABORTED(-12), Run, CTranscodingPipeline::Run::Transcode() [0x55ac76140570] failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:4584

 session 0 [0x55ac76140570] failed with status MFX_ERR_ABORTED shutting down the application...

Transcoding started
.
[ERROR], sts=MFX_ERR_GPU_HANG(-21), PutBS, Encode: SyncOperation failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1950

[ERROR], sts=MFX_ERR_ABORTED(-12), Transcode, PutBS failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:1912

[ERROR], sts=MFX_ERR_ABORTED(-12), Run, CTranscodingPipeline::Run::Transcode() [0x55ac76730160] failed at /home/devkit/newmsdk/MediaSDK/samples/sample_multi_transcode/src/pipeline_transcode.cpp:4584

Xiaogangli-intel, 2021-08-31:
Got it, will do more investigation.

Xiaogangli-intel, 2021-11-03:
@elainewangprc Thank you, we already added some tests for this.
I will close this issue.

closed 2021-11-03 by Xiaogangli-intel