intel/media-driver #1257: Hardware video acceleration corrupts video

emaster987, 2021-08-30:
## System information
- CPU information: Processors: 8 × 11th Gen Intel® Core™ i7-1165G7 @ 2.80GHz
- GPU information: Mesa Intel® Xe Graphics
- Display server if rendering to display: X

## Issue behavior
I have tried to open videos with different codecs and found that H264, H265 and VP9 videos when hardware-accelerated are corrupted while AV1 videos aren't. 
How reproducible:
You can reproduce opening H264, H265 and VP9 videos on VLC using hardware accelerated decode by GPU.

Steps to Reproduce:
1. Open VLC (default settings) and open a H264, H265 or VP9 video

### Describe the current behavior
Actual results:
VLC playing a corrupted video.

### Describe the expected behavior
Expected results:
VLC playing the video stream right.

## Debug information
I have already reported this bug con bugzilla with number 1933143. See this link https://bugzilla.redhat.com/show_bug.cgi?id=1933143

## Debug information
- What's libva/libva-utils/gmmlib/media-driver version?
libva info: VA-API version 1.9.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_9
libva info: va_openDriver() returns 0

- Could you confirm whether GPU hardware exist or not by `ls /dev/dri`? 
by-path  card0  renderD128

- Could you provide vainfo log if possible by `vainfo >vainfo.log 2>&1`?
[vainfo.log](https://github.com/intel/media-driver/files/7075837/vainfo.log)

- Could you provide libva trace log if possible? Run cmd `export LIBVA_TRACE=/tmp/libva_trace.log` first then execute the case.
[libva_trace.log.093346.thd-0x000015d3.txt](https://github.com/intel/media-driver/files/7075825/libva_trace.log.093346.thd-0x000015d3.txt)
[libva_trace.log.094509.thd-0x000018aa.txt](https://github.com/intel/media-driver/files/7075826/libva_trace.log.094509.thd-0x000018aa.txt)
[libva_trace.log.094649.thd-0x000019c6.txt](https://github.com/intel/media-driver/files/7075827/libva_trace.log.094649.thd-0x000019c6.txt)
[libva_trace.log.094724.thd-0x00001a32.txt](https://github.com/intel/media-driver/files/7075828/libva_trace.log.094724.thd-0x00001a32.txt)
[libva_trace.log.094758.thd-0x00001a7c.txt](https://github.com/intel/media-driver/files/7075829/libva_trace.log.094758.thd-0x00001a7c.txt)
[libva_trace.log.094821.thd-0x00001a8d.txt](https://github.com/intel/media-driver/files/7075830/libva_trace.log.094821.thd-0x00001a8d.txt)

- Could you attach dmesg log if it's GPU hang by `dmesg >dmesg.log 2>&1`?
[dmesg.log](https://github.com/intel/media-driver/files/7075838/dmesg.log)

- Do you want to contribute a PR? (yes/no): 
What is PR? If its meaning is to give other information, I am here to be asked for.

XinfengZhang, 2021-08-31:
thanks for reporting, from libva_trace, there are no real function call, it just call initialization, query feature and termination , no Begin/EndPicture.  have you copy all of the log? it looks like the log of vainfo, not vlcplayer.

bbbear2002, 2021-09-08:
It seems that all Intel Xe cards have this issue.

Jexu, 2022-01-17:
Are you still seeing the issue? If not, we will close this issue and you can re-open again if have any other questions.

XinfengZhang, 2022-01-17:
@emaster987 suppose the issue was fixed by patch 8033de324f5eb19ad64199a43350f45d6dc0e6d2 and b66fee0c1d0a6d01b8b33adccc8a94f2b8238fdc ,
these two patch disable MMC in vaCreateSurfaces. @Jexu have you tried the vlc?

emaster987, 2022-01-21:
> Are you still seeing the issue? If not, we will close this issue and you can re-open again if have any other questions.

I am testing the software right now. I will report any news. As said here above, I am waiting for the patch to arrive to the major stream distribution and then I will try again

chemikadze, 2022-01-23:
Had this issue too (green lines in VLC) on i7-1185g7 with Ubuntu 20.04 and intel-media-va-driver-non-free 20.1.1+ds1-1build1.

[image]

Tried to build and run with latest master, that seemed to work:

[image]

However I later tried to rebuild without mentioned patches using 3b345cc (parent of 8033de3) and it did not show any artifacts either. What's even more weird, building same version as OS version from sources based on intel-media-20.1.1 / 54492c0d1 did not have that issue as well.

Jexu, 2022-01-24:
It is glade to see the issue is gone with latest libva/gmmlib. Perhaps, some fix in libva or gmmlib worked.

@emaster987 , can you also help try again with aboce experiment to see if it works for your issue?

emaster987, 2022-01-25:
> It is glade to see the issue is gone with latest libva/gmmlib. Perhaps, some fix in libva or gmmlib worked.
> 
> @emaster987 , can you also help try again with aboce experiment to see if it works for your issue?

The issue is gone with the latest update also for me. I just updated the system and then tried again the test done above (run intel_gpu_top and vlc choosing hardware acceleration) and the test was passed by every codec (AV1,H264,H265,VP8,VP9). Thanks to all of you

emaster987, 2022-01-25:
Forgot to say that I also updated libva/gmmlib because I updated all the system package from repository

Jexu, 2022-01-27:
Thanks all for your input, so if the the problem is gone, i will close this issue. Feel free to re-open it when you have any other problems.

tjyrinki, 2022-01-27:
Thank you for working on it, this is so far the first bug I've had with my new tiger lake laptop. The problem currently still exists in openSUSE Tumbleweed, I'll try to see which component should be patched/updated-

tjyrinki, 2022-01-28:
I've checked that with the ffmpeg video decoding the resulting YUV file is fine (tested with mpv rawvideo playback). And I've checked with intel_gpu_top that both ffmpeg and VLC seem to use the Video hw. Still the result with VLC regarding h264 file (or h265, or vp9) is the garbled green.

I'd like to reopen the issue if possible, as I think the problem might have just been hidden but not resolved.

Here are traces from playing back the same h264 file with VLC vs ffmpeg.

[vainfo.txt](https://github.com/intel/media-driver/files/7956625/vainfo.txt)

[vlc_libva_trace_log_070002_thd-0x000050e4.txt](https://github.com/intel/media-driver/files/7956589/vlc_libva_trace_log_070002_thd-0x000050e4.txt)
[vlc_libva_trace_log_070002_thd-0x000050e5.txt](https://github.com/intel/media-driver/files/7956590/vlc_libva_trace_log_070002_thd-0x000050e5.txt)
[vlc_libva_trace_log_070002_thd-0x000050e6.txt](https://github.com/intel/media-driver/files/7956591/vlc_libva_trace_log_070002_thd-0x000050e6.txt)
[vlc_libva_trace_log_070002_thd-0x000050e7.txt](https://github.com/intel/media-driver/files/7956593/vlc_libva_trace_log_070002_thd-0x000050e7.txt)
[vlc_libva_trace_log_070002_thd-0x000050e8.txt](https://github.com/intel/media-driver/files/7956594/vlc_libva_trace_log_070002_thd-0x000050e8.txt)
[vlc_libva_trace_log_070002_thd-0x000050e9.txt](https://github.com/intel/media-driver/files/7956595/vlc_libva_trace_log_070002_thd-0x000050e9.txt)
[vlc_libva_trace_log_070002_thd-0x000050ec.txt](https://github.com/intel/media-driver/files/7956596/vlc_libva_trace_log_070002_thd-0x000050ec.txt)
[vlc_libva_trace_log_070003_thd-0x000050e2.txt](https://github.com/intel/media-driver/files/7956598/vlc_libva_trace_log_070003_thd-0x000050e2.txt)
[ffmpeg_libva_trace_log_072110_thd-0x0000652a.txt](https://github.com/intel/media-driver/files/7956615/ffmpeg_libva_trace_log_072110_thd-0x0000652a.txt)
[ffmpeg_libva_trace_log_072110_thd-0x0000652b.txt](https://github.com/intel/media-driver/files/7956616/ffmpeg_libva_trace_log_072110_thd-0x0000652b.txt)
[ffmpeg_libva_trace_log_072110_thd-0x0000652c.txt](https://github.com/intel/media-driver/files/7956617/ffmpeg_libva_trace_log_072110_thd-0x0000652c.txt)
[ffmpeg_libva_trace_log_072110_thd-0x0000652d.txt](https://github.com/intel/media-driver/files/7956618/ffmpeg_libva_trace_log_072110_thd-0x0000652d.txt)
[ffmpeg_libva_trace_log_072110_thd-0x0000652e.txt](https://github.com/intel/media-driver/files/7956619/ffmpeg_libva_trace_log_072110_thd-0x0000652e.txt)
[ffmpeg_libva_trace_log_072110_thd-0x0000652f.txt](https://github.com/intel/media-driver/files/7956620/ffmpeg_libva_trace_log_072110_thd-0x0000652f.txt)
[ffmpeg_libva_trace_log_072110_thd-0x00006526.txt](https://github.com/intel/media-driver/files/7956621/ffmpeg_libva_trace_log_072110_thd-0x00006526.txt)
[ffmpeg_libva_trace_log_072110_thd-0x00006528.txt](https://github.com/intel/media-driver/files/7956622/ffmpeg_libva_trace_log_072110_thd-0x00006528.txt)
[ffmpeg_libva_trace_log_072110_thd-0x00006529.txt](https://github.com/intel/media-driver/files/7956623/ffmpeg_libva_trace_log_072110_thd-0x00006529.txt)
[ffmpeg_libva_trace_log_072110_thd-0x00006530.txt](https://github.com/intel/media-driver/files/7956624/ffmpeg_libva_trace_log_072110_thd-0x00006530.txt)

tjyrinki, 2022-01-28:
Here are some more after I switched from Wayland to X11 - mpv showing blue screen with the same file (hwdec=vaapi, gpu-context=x11egl), and finally a functional accelerated output with mpv -vo vaapi (which mpv says "Note that there is absolutely no reason to use this, other than compatibility.") - notably while all three other playbacks only show activity in "Video", the -vo vaapi shows considerable percentage also in "VideoEnhance" of intel_gpu_top.

[mpv_bluescreen_libva_trace_log_074333_thd-0x00007a6f.txt](https://github.com/intel/media-driver/files/7956743/mpv_bluescreen_libva_trace_log_074333_thd-0x00007a6f.txt)
[mpv_bluescreen_libva_trace_log_074333_thd-0x00007a60.txt](https://github.com/intel/media-driver/files/7956745/mpv_bluescreen_libva_trace_log_074333_thd-0x00007a60.txt)
[mpv_vo_vaapi_libva_trace_log_074434_thd-0x00007b72.txt](https://github.com/intel/media-driver/files/7956746/mpv_vo_vaapi_libva_trace_log_074434_thd-0x00007b72.txt)
[mpv_vo_vaapi_libva_trace_log_074434_thd-0x00007b82.txt](https://github.com/intel/media-driver/files/7956747/mpv_vo_vaapi_libva_trace_log_074434_thd-0x00007b82.txt)

XinfengZhang, 2022-02-09:
VLC just merged one patch https://github.com/videolan/vlc/commit/3633417edb6330c335c7d3602e255534979c05e7 which gl accept the modifiers from media.

tjyrinki, 2022-02-17:
Tried out it now, using vlc (4.0 snapshot) 20220215.e308b8c851 gives a solid green screen on h264 and vp9 codec videos, hw accel in use according to intel_gpu_top, and unaccelerated av1 decoding.

vlc 3.0.16 continues to give the garbled greenish screen that was traced above.

Firefox 97 added with "media.ffmpeg.vaapi.enabled" (the only change needed now in Firefox 97 to enable VA API) gives the same solid green screen as vlc 20220215 snapshot.

All of these in this post were tested both under X11 and Wayland, on up-to-date openSUSE Tumbleweed, with identical results in both environments.

pnegus, 2022-02-26:
> Tried out it now, using vlc (4.0 snapshot) 20220215.e308b8c851 gives a solid green screen on h264 and vp9 codec videos, hw accel in use according to intel_gpu_top, and unaccelerated av1 decoding. [...]

I have the same problem.

On fedora 35, latest 3.16.10 kernel, Intel TGL graphics (Iris Xe 96EU-- 11320H). Have intel-media-driver installed.

Vaapi is loaded successfully by firefox and gpu_top shows hardware decode but video shows up as a green screen.